Koenkk/zigbee2mqtt #4098: Update of CC2531

bizit524, 2020-08-13:
Quick question. on Updating CC2531. I was following the standard flash proceedure here. https://www.zigbee2mqtt.io/getting_started/flashing_the_cc2531.html

 I have an old version that has problems with some devicies pairing. Before I open up an issue, standard troubleshooting is to update to the latest version. 
My question is, I have read some places that if you flash it /update it it will get a different ID (think it's called PAN ID?) If I am to update it. How can I prevent that? or does updating it not affect it? the procedure it just erasing and reapplying latest firmware on it. The documentation doesn't really cover "updating per say"  mainly reflash 
Thanks :)

Koenkk, 2020-08-14:
Check https://www.zigbee2mqtt.io/information/FAQ.html#what-does-and-does-not-require-repairing-of-all-devices

bizit524, 2020-08-14:
ah ha! I did read that on your website. haha I couldn't find that again. Ok thank you :) Stay Awesome

closed 2020-08-14 by bizit524