Koenkk/zigbee2mqtt #4099: New dongle zigbee2mqtt compatible

papperone, 2020-08-14:
I don't know if this is the right place but cannot find any more suitable and on Discord I got adviced to open issue on GitHub.
I saw in the zigbee2matt.io there's no CC2538 dongle, I do have a tested one on my tindie store, based on a open source project, PCB has been redesigned, tested extesively and then put on my tindie. It works perfectly, it has the best performances in terms of coverage and due to the CC2538 can accept more than 150 devices.
Thanks to let me know if it's possible to mention it in this list: https://www.zigbee2mqtt.io/information/supported_adapters.html

Koenkk, 2020-08-14:
I'm fine adding this (also own the dongle myself now). There was a pr open some time ago: https://github.com/Koenkk/zigbee2mqtt.io/pull/208 , but it hasn't been merged because english instructions on how to flash the stick were missing.

papperone, 2020-08-14:
yes my dongle is based on that project https://www.tindie.com/products/GiovanniCas/cc2538-cc2592-zigbee-dongle-new-zb30/

if you woudl like I can make instruction in English how to flash the module with JTAG programmer, as well (with new PCB release I'm making now) via USB interface, just let me know...

Koenkk, 2020-08-17:
Yes, please make a pr.

closed 2020-08-17 by Koenkk

papperone, 2020-08-20:
> 
> 
> Yes, please make a pr.

Sorry to ask as I've never did it, do you have a template or instructions how to do it properly?

Koenkk, 2020-08-20:
@papperone you can click the pencil button here: https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/information/supported_adapters.md 

[image]