Koenkk/zigbee2mqtt #4102: USB CC2530+CC2591 device - any experiences?

Stoatwblr, 2020-08-14:
I stumbled across this beastie whilst looking for alternatives to CC2231 

https://www.aliexpress.com/item/4000060995530.html

It appears to be a CC2530+CC2591 with integrated USB interface and I'm wondering if this would be a better coordinator/router alternative to the el-cheapo CC2531 options (same supplier has CC2231 for a couple of dollars, or slightly more with external antenna)

Westcott1, 2020-08-14:
I got one of these a couple of months ago to 'replace' a CC2531 (version with a similar external aerial), which then became a remote router.
There doesn't seem to be much performance difference that I can see.

blakadder, 2020-08-14:
you're better off with one of these https://zigbee.blakadder.com/Ebyte_E18-TBH-01.html

closed 2020-09-30 by github-actions[bot]