Koenkk/zigbee2mqtt #4103: GledOpto GL-D-005ZS unsupported

markkNL, 2020-08-14:
In succession of issue 3746: https://github.com/Koenkk/zigbee2mqtt/issues/3746 I'm making this issue to add the GL-D-0005ZS light to z2mqtt, I've purchased a bunch of them and it would be nice if they would work :) Adding the other gledopto light seemed quite low effort so I'm hoping you can do the same with this one.
I've again added some logging below. Thank you :)

zigbee2mqtt:warn  2020-08-14 17:05:52: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:warn  2020-08-14 17:05:52: Received message from unsupported device with Zigbee model 'GL-D-005ZS'
zigbee2mqtt:warn  2020-08-14 17:05:52: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:info  2020-08-14 17:05:52: Successfully interviewed '0x00124b001f5dcc92', device has successfully been paired
zigbee2mqtt:warn  2020-08-14 17:05:52: Device '0x00124b001f5dcc92' with Zigbee model 'GL-D-005ZS' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
zigbee2mqtt:info  2020-08-14 17:05:52: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x00124b001f5dcc92","supported":false}}'

Koenkk, 2020-08-30:
sorry Mark, ive missed this. what kind of features does it support? (color, temperature)

markkNL, 2020-08-30:
Hi Koen, I'd assume it has the same features as the 004ZS that is already supported, they are all the same line, just the power is different (12W vs 9W).

Koenkk, 2020-08-30:
Added! 

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-08-30 by Koenkk

markkNL, 2020-08-30:
Great, thank you 👍 :)