Koenkk/zigbee2mqtt #4106: Support for Develco KEPZB-110 alarm keypad

alekslyse, 2020-08-14:
I got a Develco KEPZB-110 keypad (alarm keypad) and was hoping to fully support a proper alarm keypad in z2m. I got the regular arm to work, but the buttons (1234567890 and code 1234) all report as unknown. Is it an easy method to implement an alarm panel as I cant see many examples of code and keypad entry in other devices.

I made the device with the standard alarm and that works fine, and attached the pcap. 0x4383 is the alarm panel. Default z2m network code

`{ zigbeeModel: ['KEPZB-110'],
        model: 'KEPZB-110',
        vendor: 'Develco',
        description: 'Alarm Panel',
        supports: '',
        fromZigbee: [fz.ias_sos_alarm_2, fz.battery, fz.command_arm, fz.command_emergency],
        toZigbee: [],
    }`

[alarmpanel.pcapng.zip](https://github.com/Koenkk/zigbee2mqtt/files/5075948/alarmpanel.pcapng.zip)

alekslyse, 2020-08-14:
Also full specs here. Can this help? https://www.elektroimportoren.no/docs/lib/4514807-Manual-18.pdf

Koenkk, 2020-08-17:
You can probably use https://github.com/Koenkk/zigbee-herdsman-converters/pull/1301 as a starting point

closed 2020-09-24 by github-actions[bot]

ioull, 2021-09-29:
I do my first integration of the alarm panel / keypad KEPZB-110 from Develco/Frient :
Here is the code :
https://gist.github.com/ioull/25334cfcc78b920ecc1c8f8828cd3c90
Key and LEDs are working fine ! The only knowing 'bug' I have is that when there is a tamper detection, I'm not able to clear it ... Any Idea ?
As it is my first z2m integration, keep free to make remarks, I'm interrested to progress !!!
The last things not tested is the battery level ! I'm waiting the message from the device 😉

heinekmadsen, 2021-11-19:
@ioull I've got my hands on one of these keypads... i see using a code is working and i can see it in the Home Assistant events... but when scanning a rfid card it beeps, but i can't really see any unique code/identifier of which card was beeped in the events... Do you have it working when swiping a card?

ioull, 2021-11-19:
Very strange that it works with the code and not with rfid ... Indead it takes the same way, the only difference is that an rfid tag start with a '+' ... Are you sure you use a suppoorted RFID TAG ?
Just to be sure, you have to swipe the card (or enter a code), then push an action button in order to generate the event.

heinekmadsen, 2021-11-19:
@ioull sorry... guess it was just me and my eyes :)

How do you handle the events with HA? Do you use node red to catch the state_changed events? and how do you check which card is read/or code is read?

ioull, 2021-11-20:
I handle it throw a workflow in NodeRed. Here is how I manage it in order to set the led status 
```
[
    {
        "id": "6814538862850eec",
        "type": "tab",
        "label": "Flow 1",
        "disabled": true,
        "info": "",
        "env": []
    },
    {
        "id": "adde1e9caf7b8d23",
        "type": "mqtt in",
        "z": "6814538862850eec",
        "name": "ENT_TEL_keypad",
        "topic": "zigbee2mqtt/ENT_TEL_keypad",
        "qos": "1",
        "datatype": "json",
        "broker": "9b372184.f0e1a",
        "nl": false,
        "rap": false,
        "inputs": 0,
        "x": 130,
        "y": 200,
        "wires": [
            [
                "79919fbee76170db"
            ]
        ]
    },
    {
        "id": "68f2d041a7a0c77b",
        "type": "mqtt out",
        "z": "6814538862850eec",
        "name": "ENT_TEL_keypad",
        "topic": "zigbee2mqtt/ENT_TEL_keypad/set",
        "qos": "1",
        "retain": "false",
        "respTopic": "",
        "contentType": "",
        "userProps": "",
        "correl": "",
        "expiry": "",
        "broker": "9b372184.f0e1a",
        "x": 690,
        "y": 160,
        "wires": []
    },
    {
        "id": "c93878364358f080",
        "type": "function",
        "z": "6814538862850eec",
        "name": "Parse topic",
        "func": "const codes = [\n    '1234',\n    '+1122AABB'\n];\n\naction = msg.payload.action;\naction_code = msg.payload.action_code;\naction_transaction = msg.payload.action_transaction;\nif(action == 'disarm' && !codes.includes(action_code))\n{\n   action = 'invalid_code';\n}\nmsg.payload = '{\"arm_mode\":{\"mode\":\"'+action+'\",\"transaction\":'+action_transaction+'}}';\nmsg.alarmAction = action;\nmsg.code = action_code;\n\nreturn msg;\n",
        "outputs": 1,
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 490,
        "y": 200,
        "wires": [
            [
                "68f2d041a7a0c77b",
                "8e18454d16948f05"
            ]
        ]
    },
    {
        "id": "79919fbee76170db",
        "type": "switch",
        "z": "6814538862850eec",
        "name": "action ?",
        "property": "payload.action",
        "propertyType": "msg",
        "rules": [
            {
                "t": "nempty"
            }
        ],
        "checkall": "true",
        "repair": false,
        "outputs": 1,
        "x": 320,
        "y": 200,
        "wires": [
            [
                "c93878364358f080"
            ]
        ]
    },
    {
        "id": "8e18454d16948f05",
        "type": "switch",
        "z": "6814538862850eec",
        "name": "",
        "property": "alarmAction",
        "propertyType": "msg",
        "rules": [
            {
                "t": "eq",
                "v": "disarm",
                "vt": "str"
            },
            {
                "t": "eq",
                "v": "invalid_code",
                "vt": "str"
            },
            {
                "t": "eq",
                "v": "emergency",
                "vt": "str"
            },
            {
                "t": "else"
            }
        ],
        "checkall": "true",
        "repair": false,
        "outputs": 4,
        "x": 650,
        "y": 260,
        "wires": [
            [],
            [],
            [],
            []
        ]
    },
    {
        "id": "9b372184.f0e1a",
        "type": "mqtt-broker",
        "name": "mqttIoull",
        "broker": "192.168.10.9",
        "port": "1883",
        "clientid": "",
        "usetls": false,
        "compatmode": true,
        "keepalive": "60",
        "cleansession": true,
        "birthTopic": "",
        "birthQos": "0",
        "birthPayload": "",
        "closeTopic": "",
        "closeQos": "0",
        "closePayload": "",
        "willTopic": "",
        "willQos": "0",
        "willPayload": ""
    }
]
```

heinekmadsen, 2021-11-20:
So you use this to set the LED status’ on the keypad? How do you control them? Mine just stay turned off

ioull, 2021-11-21:
Did you try my flow ? It manage the keypad LEDs. The code 1234 will activate the alarm

heinekmadsen, 2021-11-22:
Well… I do see the yellow led flashing when using an invalid code…
Is it possible to also flash the green and red leds as pr. Technical manual?

According to the technical manual
***@***.***

ioull, 2021-11-22:
Oh sorry I forgot the part I update the status after alarm is armed. Here is the updated flow :

``` yaml
[
    {
        "id": "6814538862850eec",
        "type": "tab",
        "label": "Flow 1",
        "disabled": true,
        "info": "",
        "env": []
    },
    {
        "id": "adde1e9caf7b8d23",
        "type": "mqtt in",
        "z": "6814538862850eec",
        "name": "ENT_TEL_keypad",
        "topic": "zigbee2mqtt/ENT_TEL_keypad",
        "qos": "1",
        "datatype": "json",
        "broker": "9b372184.f0e1a",
        "nl": false,
        "rap": false,
        "inputs": 0,
        "x": 130,
        "y": 200,
        "wires": [
            [
                "79919fbee76170db"
            ]
        ]
    },
    {
        "id": "68f2d041a7a0c77b",
        "type": "mqtt out",
        "z": "6814538862850eec",
        "name": "ENT_TEL_keypad",
        "topic": "zigbee2mqtt/ENT_TEL_keypad/set",
        "qos": "1",
        "retain": "false",
        "respTopic": "",
        "contentType": "",
        "userProps": "",
        "correl": "",
        "expiry": "",
        "broker": "9b372184.f0e1a",
        "x": 690,
        "y": 160,
        "wires": []
    },
    {
        "id": "c93878364358f080",
        "type": "function",
        "z": "6814538862850eec",
        "name": "Parse topic",
        "func": "const codes = [\n    '1234',\n    '+1122AABB'\n];\n\naction = msg.payload.action;\naction_code = msg.payload.action_code;\naction_transaction = msg.payload.action_transaction;\nif(action == 'disarm' && !codes.includes(action_code))\n{\n   action = 'invalid_code';\n}\nmsg.payload = '{\"arm_mode\":{\"mode\":\"'+action+'\",\"transaction\":'+action_transaction+'}}';\nmsg.alarmAction = action;\nmsg.code = action_code;\n\nreturn msg;\n",
        "outputs": 1,
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 490,
        "y": 200,
        "wires": [
            [
                "68f2d041a7a0c77b",
                "8e18454d16948f05"
            ]
        ]
    },
[... 205 lines cut ...]
        "wires": []
    },
    {
        "id": "9b372184.f0e1a",
        "type": "mqtt-broker",
        "name": "mqttIoull",
        "broker": "192.168.10.9",
        "port": "1883",
        "clientid": "",
        "usetls": false,
        "compatmode": true,
        "keepalive": "60",
        "cleansession": true,
        "birthTopic": "",
        "birthQos": "0",
        "birthPayload": "",
        "closeTopic": "",
        "closeQos": "0",
        "closePayload": "",
        "willTopic": "",
        "willQos": "0",
        "willPayload": ""
    }
]
```

buche11, 2022-01-14:
Hi, just read your discussion as i searcehd for information to the Develco Keypad. I had managed it to work with the ZHA Integration for home assistant (no luck with z2mqtt or deconz). Keys and arming/disarming working fine. But as @heinekmadsen wrote: I cannot use the RFID chip. Keypad just blinks and there is no log etc. where i can see the chip number/identifier nore know where to put in the number. 
Do one of you managed to run the RFID Chips?
Best regards

heinekmadsen, 2022-01-14:
> Hi, just read your discussion as i searcehd for information to the Develco Keypad. I had managed it to work with the ZHA Integration for home assistant (no luck with z2mqtt or deconz). Keys and [...]

Hi...

Have it working by using the node-red examples from this thread.

hilnius, 2022-05-16:
Anyone looking for a very simple solution
Use mqtt publish with topic `zigbee2mqtt/keypad_name/set` 
and payload `{"arm_mode":{"mode":"arm_all_zones"}}`
the mode can be one of : `disarm,arm_day_zones,arm_night_zones,arm_all_zones,exit_delay,entry_delay,not_ready,in_alarm,arming_stay,arming_night,arming_away`

Works like a charm !

[image]

JordiPF, 2022-06-08:
> @ioullTengo en mis manos uno de estos teclados... Veo que usar un código funciona y puedo verlo en los eventos de Home Assistant... pero al escanear una tarjeta RFID suena un pitido, pero realmente no puedo ver algún código/identificador único de qué tarjeta se pitó en los eventos... ¿Lo tienes funcionando al deslizar una tarjeta?

I have it and working. you just jave to press disarm button adter you pass your rfid card

hemandk, 2022-06-08:
What mqtt payload should I use to get the LED blinking yellow, I.E. wrong code. I cannot seem to figure it out from the Node Red code?

I have the disarm, arm modes working via the payload arm_mode and that works like a charm.

hemandk, 2022-06-08:
The node red code from @ioull seems to somehow activate the yellow led for wrong code, but I cant figure what is sent to the panel to activate the led's.

ioull, 2022-06-08:
From my NR, I send the following, action_transaction is coming from the received message
msg.payload = '{"arm_mode":{"mode":"invalid_code","transaction":'+action_transaction+'}}';

From my old memory, you can set the LED only after an action on a keyboard ... but not really sure ...

JordiPF, 2022-06-08:
> El código rojo del nodo de@ioullparece activar de alguna manera el led amarillo para el código incorrecto, pero no puedo entender qué se envía al panel para activar los led.

No led on my flow

ioull, 2022-06-08:
Exactly !

hemandk, 2022-06-08:
> Exactly !

So just for testing i'm trying to send this in the developer console after putting in a code pressing disarm:

[image]

However it does not seem to work :(

ioull, 2022-06-08:
sorry if my response was not clear 
the payload should be {"arm_mode":{"mode":"invalid_code","transaction":XXXXXXXXX}}
XXXXXXXX coming from the field action_transaction from the arm request

ioull, 2022-06-08:
Perhaps it is better explain here :
https://www.zigbee2mqtt.io/devices/ZS130000078.html#arming-disarming-from-the-keypad

ramonleenders, 2022-08-24:
Does anyone know how to change the default "1234" code? @buche11 
Using this keypad with ZHA and seems to work perfectly. Only thing is I want to set my own code, but can't read anywhere on where to do this...

DennisJohnsen, 2023-01-23:
> Does anyone know how to change the default "1234" code? @buche11 Using this keypad with ZHA and seems to work perfectly. Only thing is I want to set my own code, but can't read anywhere on where to do this...

Did you figure it out? I see that through Deconz you can use post requests via Deconz rest api to change config on it. But when in Zigbee2MQTT i can't figure out a way.

hemandk, 2023-01-24:
But you can use any code you want. Just press a pin code and then either press open or close.

Then that code will be sent to HA or something else. It is up to you to validate the code in HA or what ever backend program your are using.

DennisJohnsen, 2023-01-24:
> But you can use any code you want. Just press a pin code and then either press open or close.
> 
> Then that code will be sent to HA or something else. It is up to you to validate the code in HA or what ever backend program your are using.

That’s kinda what I figured after working with it in NodeRed afterwards. I wondered why it was different in deconz, and also thought i needed to “pair” it with an RFID code , and a code on the device it self.

Should have responded to my own question the few hours later when i realized it 😅

ramonleenders, 2023-02-03:
> Did you figure it out? I see that through Deconz you can use post requests via Deconz rest api to change config on it. But when in Zigbee2MQTT i can't figure out a way.

Apparently Home Assistant has a "configuration" somewhere, which this is coupled with. I would have thought these would work standalone with their own code, but that wasn't the case. Not an issue, but just couldn't find out how this code was introduced.

There is an "Alarm Control Panel Options" section under the Network tab when configuring your integration. This can be found on the Settings - Integrations page. Scroll to the bottom and you will see "Master code for the alarm control panel(s)".

So it took me a while to figure this one out. Maybe that helps others too..

akanegally, 2024-04-01:
Hello, I developped a flow in Node Red that handle this keypad. I spent some time working on it, it works really well (handling rfid, led status, bip and so on), I just want to share it with you guys.

```JavaScript
[{"id":"7585c8cca575bbd1","type":"tab","label":"Keypad Frient","disabled":false,"info":"# Goal\r\nThis flow synchronize alarmo with a keypad frient KEYZB-110\r\nref : https://www.zigbee2mqtt.io/devices/\r\n\r\nYou need to adapt :\r\n- ","env":[]},{"id":"8ff49e42baa6e5a2","type":"function","z":"7585c8cca575bbd1","name":"ProcessKeypadEvents","func":"var p=msg.payload;\n\n//Use case : SOS button\nif ( p.action==\"emergency\" )\n{\n    return [msg,null,null];\n}\n\n\n//Filter void action\nif ( p.action == null || p.action_transaction==null)\n{\n    return [null,null,null];\n}\n//Store last valid keypad_action\nflow.set(\"keypad_last\",p);\n\nvar modes=\n{\n    \"arm_day_zones\": \"ARM_HOME\",\n    \"arm_night_zones\": \"ARM_NIGHT\",\n    \"arm_all_zones\": \"ARM_AWAY\",\n    \"disarm\": \"DISARM\"\n};\n\nif ( p.action in modes )\n{\n    var np={ \"command\": modes[p.action] };\n    np['code']=p.action_code;\n    msg.topic = flow.get(\"alarmo_command_topic\");\n    msg.payload=np;\n    msg['qos']=2;\n    msg['retain']=true; //According to alarmo documentation\n\n}\nelse\n    msg=null;\n\nreturn [null,msg,flow.get(\"keypad_last\")];","outputs":3,"timeout":0,"noerr":0,"initialize":"//Setup topic\nflow.set(\"keypad_event_topic\", \"zigbee2mqtt/en_alarm_keypad\");\nflow.set(\"keypad_command_topic\", flow.get(\"keypad_event_topic\")+\"/set\");\n\n//Default topic for alarmo, change it if needed\nflow.set(\"alarmo_event_topic\", \"alarmo/event\");\nflow.set(\"alarmo_command_topic\", \"alarmo/command\");\n\n//Last keypad action payload, DON'T TOUCH IT\nflow.set(\"keypad_last\", null); ","finalize":"","libs":[],"x":320,"y":160,"wires":[["5f07e2a2e80ffada"],["e6595d79046f9c72"],["9a25af79771647a9"]],"outputLabels":["SOS Button Clicked","Alarmo Command","[DEBUG] Last keypad action"],"info":"Open 'OnStart' tab, to edit the 4 following topics"},{"id":"a1636fec8e5262ce","type":"function","z":"7585c8cca575bbd1","name":"ProcessAlarmoEvents","func":"var keypad_last = flow.get(\"keypad_last\");\nif (keypad_last == null )\n    return [null,null];\n//Set topic\nmsg[\"topic\"] = flow.get(\"keypad_command_topic\");\n\nvar arm_events={\n    \"ARM_AWAY\":\"arm_all_zones\",\n    \"ARM_HOME\":\"arm_day_zones\",\n    \"ARM_NIGHT\":\"arm_night_zones\",\n    \"ARM_VACATION\":\"arm_all_zones\",\n    \"INVALID_CODE_PROVIDED\":\"invalid_code\",\n    \"COMMAND_NOT_ALLOWED\" : \"already_disarmed\",\n    \"FAILED_TO_ARM\": \"not_ready\"\n}\nvar event = msg.payload.event;\nif ( event in arm_events )\n{\n    var keypad_last_copy = JSON.parse(JSON.stringify(keypad_last));\n    var payload =\n    {\n        \"arm_mode\":\n        {\n            \"transaction\": keypad_last_copy.action_transaction,\n            \"mode\": arm_events[event]\n        }\n    };\n    msg[\"payload\"] = payload;\n    flow.set(\"keypad_last\",null);\n}\nelse\n    msg=null;\n\nreturn [msg, keypad_last];","outputs":2,"timeout":0,"noerr":0,"initialize":"","finalize":"","libs":[],"x":320,"y":40,"wires":[["24f7f5fd5dee8f32"],[]],"inputLabels":["Alarmo event"],"outputLabels":["Keypad responses","[DEBUG] Last keypad action"]},{"id":"e3297e7c1aae35f3","type":"mqtt in","z":"7585c8cca575bbd1","name":"OnKeypadEvents","topic":"zigbee2mqtt/en_alarm_keypad","qos":"0","datatype":"auto-detect","broker":"5914a38474afa853","nl":false,"rap":true,"rh":0,"inputs":0,"x":100,"y":160,"wires":[["8ff49e42baa6e5a2"]]},{"id":"18e2c8e68f700944","type":"mqtt in","z":"7585c8cca575bbd1","name":"OnAlarmoEvents","topic":"alarmo/event","qos":"0","datatype":"auto-detect","broker":"5914a38474afa853","nl":false,"rap":true,"rh":0,"inputs":0,"x":100,"y":40,"wires":[["a1636fec8e5262ce"]]},{"id":"24f7f5fd5dee8f32","type":"mqtt out","z":"7585c8cca575bbd1","name":"UpdateKeypad","topic":"","qos":"0","retain":"false","respTopic":"","contentType":"","userProps":"","correl":"","expiry":"","broker":"5914a38474afa853","x":580,"y":40,"wires":[]},{"id":"e6595d79046f9c72","type":"mqtt out","z":"7585c8cca575bbd1","name":"UpdateAlarmo","topic":"","qos":"2","retain":"true","respTopic":"","contentType":"","userProps":"","correl":"","expiry":"","broker":"5914a38474afa853","x":580,"y":160,"wires":[]},{"id":"e35b1984724bae27","type":"trigger-state","z":"7585c8cca575bbd1","name":"OnAlarmoState","server":"cc7e32c2.da4f1","version":4,"inputs":0,"outputs":2,"exposeAsEntityConfig":"","entityId":"alarm_control_panel.alarmo","entityIdType":"exact","debugEnabled":false,"constraints":[{"targetType":"this_entity","targetValue":"","propertyType":"current_state","propertyValue":"new_state.state","comparatorType":"is_not","comparatorValueDatatype":"prevEntity","comparatorValue":"state"}],"customOutputs":[],"outputInitially":false,"stateType":"str","enableInput":false,"x":100,"y":100,"wires":[["dfacf394f8cb268a"],[]]},{"id":"dfacf394f8cb268a","type":"function","z":"7585c8cca575bbd1","name":"ProcessAlarmoState","func":"var state=msg.payload;\nvar messages = []\nmsg[\"topic\"] = flow.get(\"keypad_command_topic\");\n\n//Hack : there is no event disarm\nif ( state == \"disarmed\") \n{\n    var keypad_last = flow.get(\"keypad_last\");\n    if (keypad_last != null )\n    {\n        var keypad_last_copy = JSON.parse(JSON.stringify(keypad_last));\n        var msg2=\n        {\n            \"topic\": msg.topic,\n            \"payload\ [... cut ...] load":"","birthMsg":{},"closeTopic":"","closeQos":"0","closeRetain":"false","closePayload":"","closeMsg":{},"willTopic":"","willQos":"0","willRetain":"false","willPayload":"","willMsg":{},"userProps":"","sessionExpiry":""},{"id":"cc7e32c2.da4f1","type":"server","name":"Home Assistant","addon":true}]
``` 
It's very easy to use. Just edit 'ProcessKeypadEvents' and OnKeypadEvents nodes to specify the topic of your keypad
[image]
If you have some trouble, consult the 'getting started' node for help.

zigbeefordomoticz, 2025-10-28:
> I do my first integration of the alarm panel / keypad KEPZB-110 from Develco/Frient : Here is the code : https://gist.github.com/ioull/25334cfcc78b920ecc1c8f8828cd3c90 Key and LEDs are working fine [...]

Have you been able to get the Tamper working ?