Koenkk/zigbee2mqtt #4107: IKEA Tradfri E1743 stop working with build 1.14.3

poberth, 2020-08-14:
# Bug Report
## What happened
Now getting this error when receiving message from device:
`
Zigbee2MQTT:error 2020-08-14 14:11:23: Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined
    at /zigbee2mqtt-1.14.3/lib/extension/receive.js:120:42
    at Array.filter (<anonymous>)
    at Receive.onZigbeeEvent (/zigbee2mqtt-1.14.3/lib/extension/receive.js:119:65)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.14.3/lib/controller.js:375:44)
    at Controller.onZigbeeEvent (/zigbee2mqtt-1.14.3/lib/controller.js:257:14)
    at Zigbee.emit (events.js:315:20)
    at Controller.<anonymous> (/zigbee2mqtt-1.14.3/lib/zigbee.js:67:52)
    at Controller.emit (events.js:315:20)
    at Controller.<anonymous> (/zigbee2mqtt-1.14.3/node_modules/zigbee-herdsman/dist/controller/controller.js:503:22)
    at Generator.next (<anonymous>))
`
## What did you expect to happen
Event should be sent as in 1.14.2

## How to reproduce it (minimal and precise)
Update to 1.14.3 from the home assistant plugin.

## Debug Info
Zigbee2MQTT version: 1.14.3
Adapter hardware: CC26X2R1
Adapter firmware version: 20200721

HELEMAALNIKS, 2020-08-14:
I have the same problem, I have two motion sensors that I want to pair. Zigbee2MQTT seems to recognize them, but it gives this error:
`Zigbee2MQTT:warn  2020-08-14 23:36:33: Received message from unsupported device with Zigbee model 'undefined'
Zigbee2MQTT:warn  2020-08-14 23:36:33: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.`

axwell, 2020-08-16:
Same issue here with ikea remote https://www.zigbee2mqtt.io/devices/E1524_E1810.html

Koenkk, 2020-08-17:
Is `zigbee_shepherd_devices` set to `true` in the configuration? If so try setting this to `false`.

HELEMAALNIKS, 2020-08-17:
Sorry if I don't get it, but do you mean the zigbee2mqtt configuration in supervisor? Because I don't even see `zigbee_shepherd_devices` in my configuration.

Koenkk, 2020-08-17:
@HELEMAALNIKS you have a different issue. In your case try **force** removing the device through https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove and repair it

poberth, 2020-08-24:
@Koenkk, `zigbee_shepherd_devices` is set to `true` because I had to add my Smart9 Temperature and Humidity Sensor (`{vendor: '_TYZB01_a476raq2', model: 'TS0201'}`).

When there is a new version of zigbee2mqtt is there a way to know that there has been some changes in the devices.js file? I read the change logs and didn't find anything.

Koenkk, 2020-08-24:
@poberth `devices.js` can break anytime, you should only have the changes for testing purposes and after that contribute the changes back to [zigbee-herdsman-converters](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js) to make the device work out-of-the-box (so that the device is supported without any changes).

closed 2020-10-01 by github-actions[bot]