Koenkk/zigbee2mqtt #4108: Pairing Climax 852PRO ULTRA - PhotoPIR

lolzor7, 2020-08-14:
Hello everyone,

I am trying to pair this photo pir sensor to my conbee 2
https://www.climax.com.tw/new/vst-852pro-ultra.php

I am getting this error on my raspberry pi console

```
zigbee2mqtt:info 2020-08-14 09:34:43: Device ‘0x00124b0008b4cf4b’ joined
> zigbee2mqtt:info 2020-08-14 09:34:43: Starting interview of ‘0x00124b0008b4cf4b’
> zigbee2mqtt:info 2020-08-14 09:34:43: MQTT publish: topic ‘zigbee2mqtt/bridge/log’, payload ‘{“type”:“device_connected”,“message”:{“friendly_name”:“0x00124b0008b4cf4b”}}’
> zigbee2mqtt:info 2020-08-14 09:34:43: MQTT publish: topic ‘zigbee2mqtt/bridge/log’, payload ‘{“type”:“pairing”,“message”:“interview_started”,“meta”:{“friendly_name”:“0x00124b0008b4cf4b”}}’
> zigbee2mqtt:error 2020-08-14 09:35:11: Failed to interview ‘0x00124b0008b4cf4b’, device has not successfully been paired
> zigbee2mqtt:info 2020-08-14 09:35:11: MQTT publish: topic ‘zigbee2mqtt/bridge/log’, payload ‘{“type”:“pairing”,“message”:“interview_failed”,“meta”:{“friendly_name”:“0x00124b0008b4cf4b”}}’
> zigbee2mqtt:warn 2020-08-14 09:39:43: Received message from unsupported device with Zigbee model ‘852_00.00.03.07TC’
> zigbee2mqtt:warn 2020-08-14 09:39:43: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
```

.

I added to configuration.yaml

```
homeassistant: false
permit_join: true
mqtt:
base_topic: zigbee2mqtt
server: ‘mqtt://192.168.1.58:1883’
serial:
port: /dev/ttyACM0
devices:
‘0x00124b0008b4cf4b’:
friendly_name: ‘Climax 852PRO-ULTRA’
advanced:
log_level: debug
```

and to devices.js

```
{
zigbeeModel: [‘852_00.00.03.07TC’], // The model ID from: Device with modelID ‘lumi.sens’ is not supported.
model: ‘VST-852P’, // Vendor model number, look on the device for a model number
vendor: ‘Climax’, // Vendor of the device (only used for documentation and startup logging)
description: ‘Climax Photo PIR Sensor’, // Description of the device, copy from vendor site. (only used for documentation and startup logging)
supports: ‘PIR and Photo’, // Actions this device supports (only used for documentation)
// fromZigbee: [], // We will add this later
toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
},
```

When i push the function button on the device I get this message in log level:
```

Zigbee2mqtt:error 2020-08-14 11:49:08: Failed to call ‘Receive’ ‘onZigbeeEvent’ (TypeError: Cannot read property ‘filter’ of undefined
at Receive.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/receive.js:119:65)
at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:360:44)
at Controller.onZigbeeEvent (/opt/zigbee2mqtt/lib/controller.js:246:14)
at Zigbee.emit (events.js:315:20)
at Controller. (/opt/zigbee2mqtt/lib/zigbee.js:67:52)
at Controller.emit (events.js:315:20)
at Controller. (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:503:22)
at Generator.next ()
at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:27:71
at new Promise ())
```

Could you help me out here?

I see there are several Climax devices manually paired to it so I think it can work…
Unforunately I don’t know what I can do more to try and get this to work, I’d be happy to get the PIR sensor readings already without the pictures

Thanks,
Kevin

Koenkk, 2020-08-17:
You are getting:

> Zigbee2mqtt:error 2020-08-14 11:49:08: Failed to call ‘Receive’ ‘onZigbeeEvent’ (TypeError: Cannot read property ‘filter’ of undefined

Because you commented `fromZigbee`, uncomment it and follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

lolzor7, 2020-08-17:
Hello, thanks for this I didn't notice

The following result:

No converter available for 'VST-852P' with cluster 'ssIasZone' and type 'readResponse' and data '{"iasCieAddr":"0x00212effff05d28d","zoneState":0}'Received Zigbee message from 'Climax 852PRO-ULTRA', type 'readResponse', cluster 'ssIasZone', data '{"iasCieAddr":"0x00212effff05d28d","zoneState":0}' from endpoint 1 with groupID null

How do I make a converter for  data '{"iasCieAddr":"0x00212effff05d28d","zoneState":0}' from endpoint 1 with groupID null

Met vriendelijke groeten,

Kevin Van den Bossche

GSM: 0490111029

lolzor7, 2020-08-17:
But still this is when I press the test button on the device.
When I activate the pir I dont receive a message.

Could it be possible that I have to send some paramaters to the device first?
It is used in alarm systems where you can set it as entry/exit or take a picture and view it remotely.

Koenkk, 2020-08-17:
Try using `fromZigbee: [fz.ias_occupancy_alarm_1]`, if that doesn't work: `fromZigbee: [fz.ias_occupancy_alarm_2]`

lolzor7, 2020-08-17:
No converter available for 'VST-852P' with cluster 'ssIasZone' and type 'readResponse' and data '{"iasCieAddr":"0x00212effff05d28d","zoneState":0}'

Still can"t get around this error

Koenkk, 2020-08-17:
Can you provide the herdsman debug logging when pairing this device?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

lolzor7, 2020-08-17:
This is the output:

`zigbee2mqtt:debug 2020-08-17 22:44:38: Received Zigbee message from 'Climax 852PRO-ULTRA', type 'readResponse', cluster 'ssIasZone', data '{"iasCieAddr":"0x00212effff05d28d","zoneState":0}' from endpoint 1 with groupID null
zigbee2mqtt:debug 2020-08-17 22:44:38: No converter available for 'VST-852P' with cluster 'ssIasZone' and type 'readResponse' and data '{"iasCieAddr":"0x00212effff05d28d","zoneState":0}'
  zigbee-herdsman:deconz:adapter response received +117ms
  zigbee-herdsman:controller:device:log Interview - IAS - after enrolling state (18): '{"iasCieAddr":"0x00212effff05d28d","zoneState":0}' +4s
  zigbee-herdsman:controller:endpoint Read 0x00124b0008b4cf4b/1 ssIasZone(["iasCieAddr","zoneState"], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) +4s
  zigbee-herdsman:deconz:adapter zclFrame.payload: +505ms
  zigbee-herdsman:deconz:adapter [ { attrId: 16 }, { attrId: 0 } ] +1ms
  zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0x46fe EP:1 SeqNr. 166 request id: 31 +553ms
  zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 31 status: 0 +552ms
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110 +3s
  zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 167 +3s
  zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0x46fe request id: 31 confirm status: 0 +12ms
  zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send +3s
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010 +5ms
  zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 168 +51ms
  zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 168 srcAddr: 0x46fe destAddr: 0x0 profile id: 0x104 cluster id: 0x500 +46ms
  zigbee-herdsman:deconz:frameParser response payload: 24,29,1,16,0,0,240,141,210,5,255,255,46,33,0,0,0,0,48,0 +0ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":29,"manufacturerCode":null,"commandIdentifier":1},"Payload":[{"attrId":16,"status":0,"dataType":240,"attrData":"0x00212effff05d28d"},{"attrId":0,"status":0,"dataType":48,"attrData":0}]},"address":18174,"endpoint":1,"linkquality":164,"groupID":null}' +4s
`

lolzor7, 2020-08-18:
This is the output when adding another (but the same) device.

```
zigbee-herdsman:controller:log Device '0x00124b0008b5ca65' joined +1m
  zigbee-herdsman:controller:log Device '0x00124b0008b5ca65' accepted by handler +2ms
  zigbee-herdsman:controller:log New device '0x00124b0008b5ca65' joined +1ms
  zigbee-herdsman:controller:log Creating device '0x00124b0008b5ca65' +0ms
zigbee2mqtt:info  2020-08-18 09:43:34: Device '0x00124b0008b5ca65' joined
  zigbee-herdsman:controller:log Interview '0x00124b0008b5ca65' start +104ms
zigbee2mqtt:info  2020-08-18 09:43:34: Starting interview of '0x00124b0008b5ca65'
  zigbee-herdsman:controller:device:log Interview - start device '0x00124b0008b5ca65' +0ms
zigbee2mqtt:info  2020-08-18 09:43:34: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00124b0008b5ca65"}}'
zigbee2mqtt:info  2020-08-18 09:43:34: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00124b0008b5ca65"}}'
  zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xc7fd EP:0 SeqNr. 65 request id: 4 +157ms
  zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 4 status: 0 +150ms
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110 +390ms
  zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 66 +402ms
  zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xc7fd request id: 4 confirm status: 0 +13ms
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010 +3ms
  zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 67 +50ms
  zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 67 srcAddr: 0xc7fd destAddr: 0x0 profile id: 0x0 cluster id: 0x8002 +47ms
  zigbee-herdsman:deconz:frameParser response payload: 4,0,253,199,2,64,128,0,0,80,160,0,0,0,160,0,0 +1ms
  zigbee-herdsman:deconz:adapter RECEIVING NODE_DESCRIPTOR - addr: 0xc7fd type: EndDevice manufacturer: 0x0 +1m
  zigbee-herdsman:controller:device:log Interview - got node descriptor for device '0x00124b0008b5ca65' +491ms
  zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xc7fd EP:0 SeqNr. 68 request id: 5 +50ms
  zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 5 status: 0 +49ms
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110 +14ms
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101110 +16ms
  zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 69 +51ms
  zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xc7fd request id: 5 confirm status: 0 +20ms
  zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 70 +51ms
  zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - seqNr.: 70 status: 5 +52ms
  zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 71 +49ms
  zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 71 srcAddr: 0xc7fd destAddr: 0x0 profile id: 0x0 cluster id: 0x8005 +49ms
  zigbee-herdsman:deconz:frameParser response payload: 5,0,253,199,1,1 +1ms
  zigbee-herdsman:deconz:adapter ACTIVE_ENDPOINTS - addr: 0xc7fd EP list: 1 +201ms
  zigbee-herdsman:controller:device:log Interview - got active endpoints for device '0x00124b0008b5ca65' +204ms
  zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xc7fd EP:0 SeqNr. 72 request id: 6 +51ms
  zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 6 status: 0 +49ms
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110 +444ms
  zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 73 +453ms
  zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xc7fd request id: 6 confirm status: 0 +10ms
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010 +9ms
  zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 74 +51ms
  zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 74 srcAddr: 0xc7fd destAddr: 0x0 profile id: 0x0 cluster id: 0x8004 +42ms
  zigbee-herdsman:deconz:frameParser response payload: 6,0,253,199,20,1,4,1,4,4,0,4,0,0,3,0,0,5,3,5,2,0,0,3,5 +1ms
  zigbee-herdsman:deconz:adapter RECEIVING SIMPLE_DESCRIPTOR - addr: 0xc7fd EP:1 inClusters: 0,3,1280,1283 outClusters: 0,1283 +555ms
  zigbee-herdsman:controller:device:log Interview - got simple descriptor for endpoint '1' device '0x00124b0008b5ca65' +549ms
  zigbee-herdsman:controller:endpoint Read 0x00124b0008b5ca65/1 genBasic(["modelId"], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) +0ms
  zigbee-herdsman:deconz:adapter zclFrame.payload: +29ms
  zigbee-herdsman:deconz:adapter [ { attrId: 5 } ] +0ms
  zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xc7fd EP:1 SeqNr. 75 request id: 7 +60ms
  zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 7 status: 0 +60ms
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010 +127ms
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010 +6ms
  zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 76 +151ms
[... cut ...]
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":5,"manufacturerCode":null,"commandIdentifier":1},"Payload":[{"attrId":5,"status":0,"dataType":66,"attrData":"852_00.00.03.05TC"}]},"address":51197,"endpoint":1,"linkquality":187,"groupID":null}' +199ms
zigbee2mqtt:debug 2020-08-18 09:43:36: Received Zigbee message from '0x00124b0008b5ca65', type 'readResponse', cluster 'genBasic', data '{"modelId":"852_00.00.03.05TC"}' from endpoint 1 with groupID null

```

lolzor7, 2020-08-18:
Maybe I need any of these parameters? 
[image]

Koenkk, 2020-08-18:
The issue is:

```
  zigbee-herdsman:controller:device:log Interview - failed for device '0x00124b0008b5ca65' with error 'Error: Interview failed because of failed IAS enroll (zoneState didn't change ('0x00124b0008b5ca65')
```

For some reason the devices refuses to enroll, therefore you are not getting updates on occupancy. 

Do you have the original gateway to sniff the traffic with? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

Would be interesting to see how they managed to get it working.

lolzor7, 2020-08-18:
Unfortunately I haven't got the original one.

It comes from companies like securitas or alarm.com

Does this mean it won't work ? Or what can I do more to make them work?

Koenkk, 2020-08-19:
An alternative would be to sniff the traffic of the join, maybe it will reveal something (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

closed 2020-09-26 by github-actions[bot]