Koenkk/zigbee2mqtt #4109: BUFFER_FULL during reporting setup results in no reporting

sjorge, 2020-08-15:
# Bug Report
## What happened
I had to do some work on the electrics and had to turn off power to a room. When I turned back on the bulbs I noticed they were not reporting correctly.

After some back and forth I think I figured out why, it's a set of 4 LED panels + a switch controller.

```
Zigbee2MQTT:info  2020-08-15 08:37:36: MQTT publish: topic 'zigbee2mqtt/office/lights/switch', payload '{"linkquality":31,"power":4,"state":"ON","update_available":false}'
Zigbee2MQTT:info  2020-08-15 08:37:39: Successfully setup reporting for '0x14b457fffe301910' - 1 - genOnOff
Zigbee2MQTT:info  2020-08-15 08:37:39: Successfully setup reporting for '0x14b457fffe2fee7a' - 1 - genOnOff
Zigbee2MQTT:info  2020-08-15 08:37:39: Successfully setup reporting for '0xec1bbdfffe9f5f8f' - 1 - genOnOff
Zigbee2MQTT:info  2020-08-15 08:37:39: Successfully setup reporting for '0xec1bbdfffe189fde' - 1 - genOnOff
Zigbee2MQTT:error 2020-08-15 08:37:39: Failed to setup reporting for '0xec1bbdfffe189fde' - Error: Bind 0xec1bbdfffe189fde/1 genLevelCtrl from '0x00124b001938a7e5/1' failed (SREQ '--> ZDO - bindReq - {"dstaddr":34989,"srcaddr":"0xec1bbdfffe189fde","srcendpoint":1,"clusterid":8,"dstaddrmode":3,"dstaddress":"0x00124b001938a7e5","dstendpoint":1}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)'))
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:291:27)
    at Generator.next (<anonymous>)
    at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:24:58)
Zigbee2MQTT:info  2020-08-15 08:37:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"office/lights/panel4"},"type":"device_announced"}'
Zigbee2MQTT:error 2020-08-15 08:37:39: Failed to setup reporting for '0x14b457fffe301910' - Error: ConfigureReporting 0x14b457fffe301910/1 genLevelCtrl([{"attribute":"currentLevel","minimumReportInterval":3,"maximumReportInterval":300,"reportableChange":1}], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (SREQ '--> AF - dataRequest - {"dstaddr":57134,"destendpoint":1,"srcendpoint":1,"clusterid":8,"transid":25,"options":0,"radius":30,"len":12,"data":{"type":"Buffer","data":[16,11,6,0,0,0,32,3,0,44,1,1]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)'))
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:291:27)
    at Generator.next (<anonymous>)
    at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:24:58)
Zigbee2MQTT:info  2020-08-15 08:37:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"office/lights/panel1"},"type":"device_announced"}'
Zigbee2MQTT:error 2020-08-15 08:37:40: Failed to setup reporting for '0x14b457fffe2fee7a' - Error: ConfigureReporting 0x14b457fffe2fee7a/1 genLevelCtrl([{"attribute":"currentLevel","minimumReportInterval":3,"maximumReportInterval":300,"reportableChange":1}], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (SREQ '--> AF - dataRequest - {"dstaddr":17611,"destendpoint":1,"srcendpoint":1,"clusterid":8,"transid":26,"options":0,"radius":30,"len":12,"data":{"type":"Buffer","data":[16,12,6,0,0,0,32,3,0,44,1,1]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)'))
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:291:27)
    at Generator.next (<anonymous>)
    at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:24:58)
Zigbee2MQTT:info  2020-08-15 08:37:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"office/lights/panel2"},"type":"device_announced"}'
Zigbee2MQTT:info  2020-08-15 08:37:40: Successfully setup reporting for '0xec1bbdfffe9f5f8f' - 1 - genLevelCtrl
Zigbee2MQTT:info  2020-08-15 08:37:40: Successfully setup reporting for '0xec1bbdfffe9f5f8f' - 1 - lightingColorCtrl
Zigbee2MQTT:info  2020-08-15 08:37:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"office/lights/panel3"},"type":"device_announced"}'
Zigbee2MQTT:info  2020-08-15 08:37:43: MQTT publish: topic 'zigbee2mqtt/office/lights/switch', payload '{"linkquality":44,"power":75,"state":"ON","update_available":false}'
```

As you can see reporting fails for some because the buffer is full. I can't toggle them individually either to get them to work.

## What did you expect to happen
Reporting to back off and retry to configure if it hits a BUFFER_FULL.

## How to reproduce it (minimal and precise)
Not sure, it always happens for me when I 4 or more bulbs get restored to power at once.

## Debug Info
Zigbee2MQTT version: 1.14.3-dev (commit #410f6a0)
Adapter hardware: CC2531,
Adapter firmware version: `{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20200728,"transportrev":2},"type":"zStack12"}`

sjorge, 2020-08-15:
I also tried running to reconfigure the panels, but that did not do anything (it has no configure block anyway), it would be nice that at least that also triggered reporting too (when enabled)

sjorge, 2020-08-15:
Also this probably explains the intermittened 0 brightness issues I am seeing with bulbs, usually after a guest used the physical switch.

sjorge, 2020-08-17:
Worked fine after a few hours when there was less network traffic. So maybe rate limiting the configuration commands that happen at once will be useful.

Koenkk, 2020-08-17:
Should probably be fixed now. Zigbee2MQTT will now retry 5 times with 2 seconds of sleep in between (should be enough for the buffers to empty). 

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

sjorge, 2020-08-17:
Nice, will pull in the changes once the dep is update in z2m.

sjorge, 2020-08-27:
Forgot to report, this does seem to work for me. (Unless I do like all my lamps at ones, but then the coordinator is out of memory so that is unrelated... pending zzh-p or zzh driver for upgrade on that one)

closed 2020-09-06 by sjorge

mremedi2023, 2024-08-30:
Same here, could you solve it?