Koenkk/zigbee2mqtt #4110: Communication with CC2531 stops

vdkeybus, 2020-08-15:
# Bug Report
## What happened
Communication with CC2531 stops after several hours of correct operation.

## What did you expect to happen
Worked well for months. Problem seems to have started after addition of about 10 new devices.

## Steps taken
A reset of the CC2531 (pressing button next to green LED) **has no effect**. LED does not even turn off. CC2531 must be fully powered down and up again (unplugging and plugging USB), before zigbee2mqtt will communicate with the CC2531 again.

## Debug Info
Zigbee2MQTT version: 1.14.3-dev
Adapter hardware: CC2531
Adapter firmware version: latest source routing

`Aug 15 13:06:10 zacate systemd[1]: Stopping zigbee2mqtt...
Aug 15 13:06:10 zacate npm[47180]: Zigbee2MQTT:debug 2020-08-15 13:06:10: Saving state to file /opt/zigbee2mqtt/data/state.json
Aug 15 13:06:10 zacate npm[47180]: Zigbee2MQTT:info  2020-08-15 13:06:10: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
Aug 15 13:06:10 zacate npm[47180]: Zigbee2MQTT:info  2020-08-15 13:06:10: Disconnecting from MQTT server
Aug 15 13:06:16 zacate npm[47180]: Zigbee2MQTT:error 2020-08-15 13:06:16: Failed to stop zigbee
Aug 15 13:06:17 zacate systemd[1]: zigbee2mqtt.service: Succeeded.
Aug 15 13:06:17 zacate systemd[1]: Stopped zigbee2mqtt.
Aug 15 13:06:17 zacate systemd[1]: Started zigbee2mqtt.
Aug 15 13:06:21 zacate npm[128745]: > zigbee2mqtt@1.14.3-dev start /opt/zigbee2mqtt
Aug 15 13:06:21 zacate npm[128745]: > node index.js
Aug 15 13:06:25 zacate npm[128786]: Zigbee2MQTT:info  2020-08-15 13:06:25: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-08-15.13-06-25' filename: log.txt
Aug 15 13:06:25 zacate npm[128786]: Zigbee2MQTT:debug 2020-08-15 13:06:25: Removing old log directory '/opt/zigbee2mqtt/data/log/2020-08-14.15-13-10'
Aug 15 13:06:26 zacate npm[128786]: Zigbee2MQTT:debug 2020-08-15 13:06:26: Loaded state from file /opt/zigbee2mqtt/data/state.json
Aug 15 13:06:27 zacate npm[128786]: Zigbee2MQTT:info  2020-08-15 13:06:27: Starting Zigbee2MQTT version 1.14.3-dev (commit #410f6a0)
Aug 15 13:06:27 zacate npm[128786]: Zigbee2MQTT:info  2020-08-15 13:06:27: Starting zigbee-herdsman...
Aug 15 13:06:27 zacate npm[128786]: Zigbee2MQTT:debug 2020-08-15 13:06:27: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
Aug 15 13:06:47 zacate npm[128786]: Zigbee2MQTT:error 2020-08-15 13:06:47: Error while starting zigbee-herdsman
Aug 15 13:06:47 zacate npm[128786]: Zigbee2MQTT:error 2020-08-15 13:06:47: Failed to start zigbee
Aug 15 13:06:47 zacate npm[128786]: Zigbee2MQTT:error 2020-08-15 13:06:47: Exiting...
Aug 15 13:06:47 zacate npm[128786]: Zigbee2MQTT:error 2020-08-15 13:06:47: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
Aug 15 13:06:47 zacate npm[128786]:     at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:89:31)
Aug 15 13:06:47 zacate npm[128786]:     at Generator.throw (<anonymous>)
Aug 15 13:06:47 zacate npm[128786]:     at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
Aug 15 13:06:47 zacate npm[128745]: npm ERR! code ELIFECYCLE
Aug 15 13:06:47 zacate npm[128745]: npm ERR! errno 1
Aug 15 13:06:47 zacate npm[128745]: npm ERR! zigbee2mqtt@1.14.3-dev start: `node index.js`
Aug 15 13:06:47 zacate npm[128745]: npm ERR! Exit status 1
Aug 15 13:06:47 zacate npm[128745]: npm ERR!
Aug 15 13:06:47 zacate npm[128745]: npm ERR! Failed at the zigbee2mqtt@1.14.3-dev start script.
Aug 15 13:06:47 zacate npm[128745]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
Aug 15 13:06:47 zacate npm[128745]: npm ERR! A complete log of this run can be found in:
Aug 15 13:06:47 zacate npm[128745]: npm ERR!     /home/zigbee2mqtt/.npm/_logs/2020-08-15T11_06_47_858Z-debug.log
Aug 15 13:06:48 zacate systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Aug 15 13:06:48 zacate systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Aug 15 13:06:48 zacate systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 1.
Aug 15 13:06:48 zacate systemd[1]: Stopped zigbee2mqtt.`

sjorge, 2020-08-15:
Are you using the default or source routing firmware, with more than 15ish devices I’d go source routing.

Currently stableish for me with 44 devices (30ish routers)

vdkeybus, 2020-08-15:
source routing

I am now using another CC2531 instead, but the results so far are disastrous. The network that worked relatively well (about 1 in 5 times a command is executed with 2 to 5 s delay) is now super-unreliable.

I am counting on the self-organizing abilities of the Zigbee network, but it seems my network is now stuck in a state where nothing can communicate anymore with anything. About 1/3 of the commands now executes, and 1/2 of that with 2..5 s delay.

vdkeybus, 2020-08-15:
I am in the same state again (button S1 does not reset anymore) 2 observations I made so far:

1. I get messages like:

Aug 15 23:20:07 zacate npm[182330]: Zigbee2MQTT:error 2020-08-15 23:20:07: Publish 'set' 'state' to '0x000b57fffe2ad0a9' failed: 'Error: Command 0x000b57fffe2ad0a9/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (SREQ '--> AF - dataRequest - {"dstaddr":40912,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":60,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,143,1]}}' failed with status '(0x10: MEM_ERROR)' (expected '(0x00: SUCCESS)'))'

I suspect the CC2531 may be running out of RAM. Is it possible to _reduce_ the number of children ? I have plenty of routers.

2. The CC2531, with its 39 devices, effectively manages to create a mess instead of a mesh. At one moment I have an important link from ground floor to first floor of lqi 44. 10 mins. later, this is lqi 0, creating a situation in which the rest of the network is rendered totally inaccessible. Are there any parameters to influence the network mesh optimization ?

Thanks.

vdkeybus, 2020-08-16:
I reflashed the CC2531 and re-paired every node in the network, progressing from closest to farthest nodes. At the same time, I moved the Zigbee channel from 11 to 15, as I am suspecting interference from 2.4GHz WiFi on channel 1 (neighbours). LQI was initially good (42..55), but rapidly settled to 2..5 for most connections.

Again I ran into the MEM_ERROR problem, but this was during the use of zigbee2mqttassistant. I think the network topology scan, which is always initiated, may trigger the problem.

Another case where MEM_ERROR happened is when 4+ commands were given simultaneously to members of a group.

After MEM_ERROR, it seems the CC2531 is no longer stable and must be power-cycled. I tried to do an OTA update and it errored out on 'undefined' until I power-cycled the coordinator. It is now performing the OTA update without error.

Koenkk, 2020-08-17:
39 devices is probably to much for the CC2531, make sure that you don't use the availability (https://www.zigbee2mqtt.io/information/availability.html) feature. If you don't I recommend switching to a more powerful adapter: https://www.zigbee2mqtt.io/information/supported_adapters.html#electrolama-zig-a-zig-ah-zzh

vdkeybus, 2020-08-17:
Hi Koen,

39 devices is probably too much for the CC2531, make sure that you don't

> use the availability (
> https://www.zigbee2mqtt.io/information/availability.html) feature.
I have:
availability_timeout: 0
in my configuration.yaml. I believe this disabled it, right ?

> If you don't I recommend switching to a more powerful adapter:
> https://www.zigbee2mqtt.io/information/supported_adapters.html#electrolama-zig-a-zig-ah-zzh
I figured as much. I have ordered a couple of alternatives already.

Thanks.

vdkeybus, 2020-08-21:
Yesterday, I got a Conbee II. After replacing the CC2531 with it (and modifying the adapter setting) the network comes up, but many routes are unreliable (low lqi, no response from device).

Does the stick (a CC2531, maybe also Conbee) keep persistent information about the network ?

Is it allowable to do a drop-in replacement, or should I rather re-pair ?

Similarly, am I risking bricking the Conbee II (as has happened with the CC2531) ? I cannot reprogram that one, I'm afraid.

Koenkk, 2020-08-23:
@vdkeybus didn't expect that, but thinking about this more, it can indeed work. However this can cause several side effects. Since the ieeeAddress of the coordinator changes, some functionality which depends on binding might not be working anymore (since that use the old ieeeAddr). I would for sure recommend repairing. Just remove `database.db` and `state.json`. Then change the network key by putting the following in the `configuration.yaml`.

```yaml
advanced:
 network_key: GENERATE
```

vdkeybus, 2020-08-23:
Ok, I didn't realize about the network key option. That's useful, thanks.

maruiz81, 2020-09-10:
Hi, I am having the same MEM_ERROR issue in a CC2531. But I only have 11 devices connected. Is it normal? It was working fine with the same device for months

Koenkk, 2020-09-10:
@maruiz81 could you provide the herdsman debug logging of this?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

ALaDoffe, 2020-09-10:
@Koenkk 

Hello
It seems that we are several that experience CC2531 overload since z2m 1.14.x like #4296 
First error occures with Zigbee2MQTT version 1.14.3-dev (commit #9b4ef70) but I can not find anymore was the last good version for me.
My network (20 Devices) didn't change and it was pretty stable before.

Regards.

Koenkk, 2020-09-10:
@ALaDoffe could you also provide the logging like requested in https://github.com/Koenkk/zigbee2mqtt/issues/4110#issuecomment-690075450 ?

maruiz81, 2020-09-10:
[zigbeelogs.txt](https://github.com/Koenkk/zigbee2mqtt/files/5200789/zigbeelogs.txt)
Here is my logs, I hope it help.

Koenkk, 2020-09-10:
@maruiz81 the herdsman debug logging is logged to stdout, not the log files (like I think you posted now)

maruiz81, 2020-09-10:
> @maruiz81 the herdsman debug logging is logged to stdout, not the log files (like I think you posted now)

that's true, I could not find how to store in the file. I have added this but it doesn't work for hersman
log_output:
    - console
    - file

Koenkk, 2020-09-10:
@maruiz81 I don't know how you are running z2m, but you can copy it from the STDOUT.

maruiz81, 2020-09-10:
Ok, it is not complete but here it is: 

`2020-09-10T11:50:35.963Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,33,0,69]
2020-09-10T11:50:35.964Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,33,0,69]
2020-09-10T11:50:35.964Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 33 - [0] - 69
2020-09-10T11:50:35.965Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - bindReq - {"status":0}
2020-09-10T11:50:35.965Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-09-10T11:50:37.434Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x842e14fffe139658:18847/1 (0,4)
2020-09-10T11:50:37.437Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":18847,"destendpoint":1,"srcendpoint":1,"clusterid":2820,"transid":7,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[16,3,0,5,5,8,5,11,5]}}
2020-09-10T11:50:37.438Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,19,36,1,159,73,1,1,4,11,7,0,30,9,16,3,0,5,5,8,5,11,5,239]
2020-09-10T11:50:37.449Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,16,116]
2020-09-10T11:50:37.449Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,16,116]
2020-09-10T11:50:37.450Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [16] - 116
2020-09-10T11:50:37.450Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":16}
2020-09-10T11:50:37.451Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-09-10T11:50:37.452Z zigbee-herdsman:controller:endpoint Read 0x842e14fffe139658/1 haElectricalMeasurement(["rmsVoltage","rmsCurrent","activePower"], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (SREQ '--> AF - dataRequest - {"dstaddr":18847,"destendpoint":1,"srcendpoint":1,"clusterid":2820,"transid":7,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[16,3,0,5,5,8,5,11,5]}}' failed with status '(0x10: MEM_ERROR)' (expected '(0x00: SUCCESS)'))
2020-09-10T11:50:39.301Z zigbee-herdsman:controller:endpoint Read 0x842e14fffe139658/1 haElectricalMeasurement(["rmsVoltage","rmsCurrent","activePower"], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})
2020-09-10T11:50:39.302Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x842e14fffe139658:18847/1 (0,0)
2020-09-10T11:50:39.303Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":18847,"destendpoint":1,"srcendpoint":1,"clusterid":2820,"transid":8,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[16,4,0,5,5,8,5,11,5]}}
2020-09-10T11:50:39.304Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,19,36,1,159,73,1,1,4,11,8,0,30,9,16,4,0,5,5,8,5,11,5,231]
2020-09-10T11:50:39.313Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,16,116]
2020-09-10T11:50:39.313Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,16,116]
2020-09-10T11:50:39.313Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [16] - 116
2020-09-10T11:50:39.314Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":16}
2020-09-10T11:50:39.314Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-09-10T11:50:39.315Z zigbee-herdsman:controller:endpoint Read 0x842e14fffe139658/1 haElectricalMeasurement(["rmsVoltage","rmsCurrent","activePower"], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (SREQ '--> AF - dataRequest - {"dstaddr":18847,"destendpoint":1,"srcendpoint":1,"clusterid":2820,"transid":8,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[16,4,0,5,5,8,5,11,5]}}' failed with status '(0x10: MEM_ERROR)' (expected '(0x00: SUCCESS)'))
2020-09-10T11:50:45.969Z zigbee-herdsman:controller:endpoint Bind 0x04cf8cdf3c786bbd/1 genPowerCfg from '0x00124b0018e85c70/1' failed (AREQ - ZDO - bindRsp after 10000ms)
Zigbee2MQTT:error 2020-09-10 13:50:45: Failed to configure '0x04cf8cdf3c786bbd', attempt 2 (Error: Bind 0x04cf8cdf3c786bbd/1 genPowerCfg from '0x00124b0018e85c70/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (/zigbee2mqtt-1.14.4/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7))
2020-09-10T11:50:49.301Z zigbee-herdsman:controller:endpoint Read 0x842e14fffe139658/1 haElectricalMeasurement(["rmsVoltage","rmsCurrent","activePower"], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})
2020-09-10T11:50:49.303Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x842e14fffe139658:18847/1 (0,0)
2020-09-10T11:50:49.305Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":18847,"destendpoint":1,"srcendpoint":1,"clusterid":2820,"transid":9,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[16,5,0,5,5,8,5,11,5]}}
[... cut ...]
2020-09-10T11:50:57.472Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,26,1,13,209]
2020-09-10T11:50:57.472Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [26,1,13] - 209
2020-09-10T11:50:57.473Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":26,"endpoint":1,"transid":13}
2020-09-10T11:50:57.474Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-09-10T11:50:57.474Z zigbee-herdsman:adapter:zStack:adapter Data confirm error (0x842e14fffe139658:18847,26,4)`

Koenkk, 2020-09-10:
This issue might be caused by TS0121_plug, since this plug does not support reporting of the power values, we read it every 10 seconds. This might be heavy for the coordinator and cause the issues you are experiencing.

To find out whether this  is the cause: the latest dev branch added an option to control the poll interval, first update to the latest dev branch and set the `measurement_poll_interval` to a very high value (e.g. 999999999). Docs: https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/devices/TS0121_plug.md#device-type-specific-configuration

vdkeybus, 2020-09-10:
Is zigbee2mqtt assistant used ? It increased traffic on my network and I
had a subjective feeling that I ran into the memory issues much quicker
when this tool was running.

maruiz81, 2020-09-10:
> This issue might be caused by TS0121_plug, since this plug does not support reporting of the power values, we read it every 10 seconds. This might be heavy for the coordinator and cause the issues [...]

I have set up `measurement_poll_interval` to 999999999 and It seems to work much better now. It reacts quickly when I enable or disable something. Before it didn't react or took many seconds until turn on/off. I will tell you in case it starts to fail again. Let me know if you want to check something or another fix for the issue. Maybe I could set the poll interval to a 1 minute?

Koenkk, 2020-09-10:
@maruiz81 yes, first monitor for some time if everything is OK, then try e.g. 60 seconds for the poll interval.

ALaDoffe, 2020-09-10:
@Koenkk 

Koen, here joined my  herdsman debug log
My network failed again today

Still KO after z2m stop and start with debug
[Debug_before_unplug_plug_Z2M_KO.txt](https://github.com/Koenkk/zigbee2mqtt/files/5203217/Debug_before_unplug_plug_Z2M_KO.txt)

OK after z2m stop, unplug/plug CC2531 and start with debug
[Debug_After_unplug_plugCC_Z2M_OK.txt](https://github.com/Koenkk/zigbee2mqtt/files/5203218/Debug_After_unplug_plugCC_Z2M_OK.txt)

I just noticed on reading it that Z2M started with rtscts: false that is now default behavior.
I have a CC2531 as far I remember, rtscts must be set to true with it (that was default before)
I don't know if it is related.

Regards

Koenkk, 2020-09-10:
Yes it seems that there is something wrong in the recent releases. Issue has also been reported in @aBaDDoNNL in #4307.

I've checked your log, it seems that somehow the memory gets corrupted in the adapter (which can indeed be solved by replugging). Now you provided the logging of an already crashed adapter, are you able to provide a more complete log, so from the point where everything still works OK until it stops working?

ALaDoffe, 2020-09-11:
@Koenkk 

The only log that I have is Z2M log reporting Info.

Issue start with 
error 2020-09-10 10:46:05: Failed to ping '0x01124b001baff38b'
error 2020-09-10 10:46:52: Failed to ping '0x0017880104c0b0aa'
error 2020-09-10 10:47:41: Failed to ping '0x01124b001bb0054d'
error 2020-09-10 10:49:06: Failed to ping '0x01124b001bb00e16'
error 2020-09-10 10:49:55: Failed to ping '0x01124b001bafeceb'
error 2020-09-10 10:55:50: Failed to ping '0x7cb03eaa0a007777'
error 2020-09-10 10:55:56: Failed to ping '0x7cb03eaa0a038b23'
error 2020-09-10 10:56:30: Failed to ping '0x7cb03eaa0a04cd29'
error 2020-09-10 10:56:35: Failed to ping '0x7cb03eaa0a045f94'
error 2020-09-10 10:59:41: Failed to ping '0x00158d00042ac001'
error 2020-09-10 11:05:46: Failed to read state of '0x7cb03eaa0a038b23' after reconnect
error 2020-09-10 11:20:57: Failed to ping '0x7cb03eaa0a04cd29'
error 2020-09-10 11:25:12: Failed to ping '0x7cb03eaa0a038b23'
error 2020-09-10 11:39:55: Failed to read state of '0x01124b001bafeceb' after reconnect
error 2020-09-10 11:42:01: Failed to ping '0x7cb03eaa0a04593b'
error 2020-09-10 11:44:53: Failed to ping '0x01124b001bafeceb'
error 2020-09-10 12:17:15: Failed to read state of '0x01124b001bafeceb' after reconnect
error 2020-09-10 12:22:15: Failed to ping '0x01124b001bafeceb'
error 2020-09-10 12:39:53: Failed to read state of '0x01124b001bafeceb' after reconnect
error 2020-09-10 12:44:53: Failed to ping '0x01124b001bafeceb'
error 2020-09-10 13:39:53: Failed to read state of '0x01124b001bafeceb' after reconnect
error 2020-09-10 13:44:55: Failed to ping '0x01124b001bafeceb'
error 2020-09-10 13:50:32: Failed to read state of '0x01124b001bafeceb' after reconnect
error 2020-09-10 13:55:32: Failed to ping '0x01124b001bafeceb'
error 2020-09-10 14:40:08: Failed to read state of '0x01124b001bafeceb' after reconnect
error 2020-09-10 14:44:53: Failed to ping '0x01124b001bafeceb'
error 2020-09-10 15:39:54: Failed to read state of '0x01124b001bafeceb' after reconnect
error 2020-09-10 15:44:54: Failed to ping '0x01124b001bafeceb'
error 2020-09-10 16:02:35: Failed to read state of '0x01124b001baff38b' after reconnect
error 2020-09-10 16:07:35: Failed to ping '0x01124b001baff38b'
error 2020-09-10 16:40:06: Failed to read state of '0x01124b001bafeceb' after reconnect
error 2020-09-10 16:44:54: Failed to ping '0x01124b001bafeceb'
error 2020-09-10 17:39:54: Failed to read state of '0x01124b001bafeceb' after reconnect
error 2020-09-10 17:41:06: Publish 'set' 'state' to '0x01124b001bafeceb' failed: 'Error: Command 0x01124b001bafeceb/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 9042 - 11 - 86 - 6 - 11 after 10000ms)'
info  2020-09-10 17:41:06: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x01124b001bafeceb' failed: 'Error: Command 0x01124b001bafeceb/11 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Timeout - 9042 - 11 - 86 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"0x01124b001bafeceb"},"type":"zigbee_publish_error"}'

Regarding "rtscts" I forced it to true with the last restart.

Koenkk, 2020-09-11:
@ALaDoffe can you disable the availability feature and see if it keeps stable? 

Remove the following from your `configuration.yaml`:

```
advanced:
  availability_timeout: 60
```

Let me know if it keeps working after that.

ALaDoffe, 2020-09-11:
@Koenkk 

I have

advanced:
  availability_timeout: 300

Just removed it for testing
But it was really stable before V1.14.3-dev

What about downgrading ?

Koenkk, 2020-09-11:
@ALaDoffe lets see if removing availability fixes it, if that is the case I know where the issue comes from and how to fix it.

ALaDoffe, 2020-09-11:
@Koenkk 
Ok. So wait and see.

Koenkk, 2020-09-11:
yes, please do to have a clean start

ALaDoffe, 2020-09-11:
Done

maruiz81, 2020-09-11:
I have checked this morning with `measurement_poll_interval: 60` because the previous fix was performing very well and I have two issues:
- It slows down the sets very quickly
- I don't have the measurement in Tuya anyway.

So I set to the maximum again at the moment. Let me know if you need more feedback.

Koenkk, 2020-09-11:
@maruiz81 can you try changing the definition to https://github.com/Koenkk/zigbee-herdsman-converters/blob/fe4cbde9218fd2ccf9e5303d75d2cc0f1738c92b/devices.js#L1414 in devices.js ? See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

maruiz81, 2020-09-12:
> @maruiz81 can you try changing the definition to https://github.com/Koenkk/zigbee-herdsman-converters/blob/fe4cbde9218fd2ccf9e5303d75d2cc0f1738c92b/devices.js#L1414 in devices.js ? See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

@Koenkk I am going to try when I'll be back at home.

ALaDoffe, 2020-09-14:
@Koenkk 
Hi 
Unfortunately issue still persist with availability feature disabled.
However devices are still available, I can control most of them but with long delay between command and response.
Network map shows only 2 connections 
[image]

Logs
[Z2M_errors.txt](https://github.com/Koenkk/zigbee2mqtt/files/5216286/Z2M_errors.txt)

Regards

Koenkk, 2020-09-14:
@ALaDoffe I've did some fixes in the latest dev branch (https://github.com/Koenkk/zigbee2mqtt/issues/4307#issuecomment-691682736) please test those. Let's also close the issue in favour of #4307 to avoid having 2 threads of the same issue.

closed 2020-09-14 by Koenkk

maruiz81, 2020-09-14:
> @maruiz81 can you try changing the definition to https://github.com/Koenkk/zigbee-herdsman-converters/blob/fe4cbde9218fd2ccf9e5303d75d2cc0f1738c92b/devices.js#L1414 in devices.js ? See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

@Koenkk does this test make sense still? or is it already integrated in the edge version?

Koenkk, 2020-09-14:
@maruiz81 that is not integrated yet, let me know if power reading still work after that change.

maruiz81, 2020-09-15:
> @maruiz81 that is not integrated yet, let me know if power reading still work after that change.

I have made the changes and unfortunately, it can read the power and energy values. 

Maybe is not related but I can see in the trace the next MEM_ERROR in a Xiaomi light sensor:

```
zigbee2mqtt:error 2020-09-15 10:31:00: Failed to configure '0x04cf8cdf3c786bbd', attempt 1 (Error: Bind 0x04cf8cdf3c786bbd/1 genPowerCfg from '0x00124b0018e85c70/1' failed (Error: SREQ '--> ZDO - bindReq - {"dstaddr":1669,"srcaddr":"0x04cf8cdf3c786bbd","srcendpoint":1,"clusterid":1,"dstaddrmode":3,"dstaddress":"0x00124b0018e85c70","dstendpoint":1}' failed with status '(0x10: MEM_ERROR)' (expected '(0x00: SUCCESS)'))
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:255:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
```

Koenkk, 2020-09-15:
@maruiz81 it seems that polling this device is too heavy in your case. As you don't need to power values anyway, just disable this.

maruiz81, 2020-09-17:
@Koenkk I have restarted home assistant and finally it seems to be reading properly the power now I am going to give it some days og testing. Thank you