Koenkk/zigbee2mqtt #4112: Failed to disable reporting

to4ko, 2020-08-15:
# Bug Report
## What happened
Messages like one shown below pops up in z2m log. These errors are for the devices (specifically Xiaomi plugs and Aqara sockets) that are doing self-announcement form time time with relay restart (leading to connected device toggle).

error 2020-08-15 15:29:35: Failed to disable reporting for '0x00158d00039bd56d' - Error: Unbind 0x00158d00039bd56d/1 genOnOff from '0x00124b001940c183/1' failed (AREQ - ZDO - unbindRsp after 10000ms)
    at Timeout._onTimeout (/zigbee2mqtt-1.14.3/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)

## What did you expect to happen
have reporting disabled
## How to reproduce it (minimal and precise)
turn reporting on for big network of Xiaomi devices

## Debug Info
Zigbee2MQTT version:11.14.3
Adapter hardware: CC2538
Adapter firmware version: 'revision': 20200211

sjorge, 2020-08-15:
Is 0x00158d00039bd56d a battery device perhaps? If so, try waking it up and restarting z2m.

to4ko, 2020-08-15:
as it's stated in description - they are xiaomi plugs and aqara sockets

[image]

Koenkk, 2020-08-17:
Could you sniff the traffic of this?

to4ko, 2020-08-17:
will do asap

closed 2020-09-24 by github-actions[bot]