Koenkk/zigbee2mqtt #4115: Aqara Wireless Switches stopped working after plugin update

SVhuMMer, 2020-08-16:
# Bug Report
## After the latest update my 3 aqara wireless wall switches stopped working (two 86sw2 WXKG02LM and one remote.b286acn01 WXKG02LM)
## Expected unchanged behavior

## How to reproduce it (minimal and precise)
Update Zigbee2MQTT package to 1.14.3
## Debug Info
Zigbee2MQTT version: 1.14.3
Adapter hardware: CC2538 
Adapter firmware version:  zStack30x

papinist, 2020-08-20:
Same issue here.
After many tries, I solved it changing in my automation "left click" with "single left action".
I don't remember if this entry was present before, but for sure my automation was working before..

Koenkk, 2020-08-20:
@papinist can you share your `configuration.yaml`?

papinist, 2020-08-20:
Sure.
Are you referring to the main configuration.yaml (I can't see any references to zigbee2mqtt there) or the zigbee2mqtt.yaml in \packages? I'm kind of a newbie here..

Koenkk, 2020-08-20:
the zigbee2mqtt one (i guess `zigbee2mqtt.yaml` in your case?)

papinist, 2020-08-20:
/config/packages/zigbee2mqtt.yaml
`# Input select for Zigbee2mqtt debug level
input_select:
  zigbee2mqtt_log_level:
    name: Zigbee2mqtt Log Level
    options:
      - debug
      - info
      - warn
      - error
    initial: info
    icon: mdi:format-list-bulleted

# Input text to input Zigbee2mqtt friendly_name for scripts
input_text:
  zigbee2mqtt_old_name:
    name: Zigbee2mqtt Old Name
  zigbee2mqtt_new_name:
    name: Zigbee2mqtt New Name
  zigbee2mqtt_remove:
    name: Zigbee2mqtt Remove

# Scripts for renaming & removing devices
script:
  zigbee2mqtt_rename:
    alias: Zigbee2mqtt Rename
    sequence:
      service: mqtt.publish
      data_template:
        topic: zigbee2mqtt/bridge/config/rename
        payload_template: >-
          {
            "old": "{{ states.input_text.zigbee2mqtt_old_name.state | string }}",
            "new": "{{ states.input_text.zigbee2mqtt_new_name.state | string }}"
          }
  zigbee2mqtt_remove:
    alias: Zigbee2mqtt Remove
    sequence:
      service: mqtt.publish
      data_template:
        topic: zigbee2mqtt/bridge/config/remove
        payload_template: "{{ states.input_text.zigbee2mqtt_remove.state | string }}"

# Timer for joining time remaining (120 sec = 2 min)
timer:
  zigbee_permit_join:
    name: Time remaining
    duration: 120

sensor:
  # Sensor for monitoring the bridge state
  - platform: mqtt
    name: Zigbee2mqtt Bridge state
    state_topic: "zigbee2mqtt/bridge/state"
    icon: mdi:router-wireless
  # Sensor for Showing the Zigbee2mqtt Version
  - platform: mqtt
    name: Zigbee2mqtt Version
    state_topic: "zigbee2mqtt/bridge/config"
    value_template: "{{ value_json.version }}"
    icon: mdi:zigbee
  # Sensor for Showing the Coordinator Version
  - platform: mqtt
    name: Coordinator Version
    state_topic: "zigbee2mqtt/bridge/config"
    value_template: "{{ value_json.coordinator }}"
    icon: mdi:chip

# Switch for enabling joining
switch:
  - platform: mqtt
    name: "Zigbee2mqtt Main join"
    state_topic: "zigbee2mqtt/bridge/config/permit_join"
    command_topic: "zigbee2mqtt/bridge/config/permit_join"
    payload_on: "true"
    payload_off: "false"

automation:
  # Automation for sending MQTT message on input select change
  - alias: Zigbee2mqtt Log Level
    initial_state: "on"
    trigger:
      platform: state
      entity_id: input_select.zigbee2mqtt_log_level
    action:
      - service: mqtt.publish
        data:
          payload_template: "{{ states('input_select.zigbee2mqtt_log_level') }}"
          topic: zigbee2mqtt/bridge/config/log_level
  # Automation to start timer when enable join is turned on
  - id: zigbee_join_enabled
    alias: Zigbee Join Enabled
  #  hide_entity: true
    trigger:
      platform: state
      entity_id: switch.zigbee2mqtt_main_join
      to: "on"
    action:
      service: timer.start
      entity_id: timer.zigbee_permit_join
  # Automation to stop timer when switch turned off and turn off switch when timer finished
  - id: zigbee_join_disabled
    alias: Zigbee Join Disabled
  #  hide_entity: true
    trigger:
      - platform: event
        event_type: timer.finished
        event_data:
          entity_id: timer.zigbee_permit_join
      - platform: state
        entity_id: switch.zigbee2mqtt_main_join
        to: "off"
    action:
      - service: timer.cancel
        data:
          entity_id: timer.zigbee_permit_join
      - service: switch.turn_off
        entity_id: switch.zigbee2mqtt_main_join
        `

Koenkk, 2020-08-23:
Can you also share your `devices.yaml`?

papinist, 2020-08-23:
Where should I find it? i can't seem to have any config files with this name.

Koenkk, 2020-08-24:
I found the issue, (its the same as https://github.com/Koenkk/zigbee2mqtt/issues/4168 ). Should be fixed in the latest dev branch, see https://github.com/Koenkk/zigbee2mqtt/issues/4168#issuecomment-678824223

closed 2020-10-01 by github-actions[bot]