Koenkk/zigbee2mqtt #4116: Entity '...' is unknown - Can't be deleted!

milobt, 2020-08-17:
I previously created a group called 'lounge_spot_lights' as a test and then tried to remove it, however, I have not been able to.
Every time I start up Z2M it reports:

```
info  2020-08-17 13:53:31: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"entity_not_found","message":{"friendly_name":"lounge_spot_lights"}}'
error 2020-08-17 13:53:31: Entity 'lounge_spot_lights' is unknown
```

I have tried using the command to remove a group and device but neither work and instead produce errors:

```
error 2020-08-17 13:57:59: Failed to call 'BridgeLegacy' 'onMQTTMessage' (AssertionError [ERR_ASSERTION]: Group 'lounge_spot_lights' does not exist
    at BridgeLegacy.removeGroup (/zigbee2mqtt-1.14.3/lib/extension/legacy/bridgeLegacy.js:275:9)
    at BridgeLegacy.onMQTTMessage (/zigbee2mqtt-1.14.3/lib/extension/legacy/bridgeLegacy.js:368:44)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.14.3/lib/controller.js:375:44)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:97:5))
```

```
error 2020-08-17 14:01:24: Cannot remove, device 'lounge_spot_lights' does not exist
info  2020-08-17 14:01:24: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed_failed","message":"lounge_spot_lights"}'
```

At present the group is not listed in my groups.yaml file and I have replaced it in the past to try and delete the "entity", but nothing seems to work.

Any suggestions please?

## Debug Info
Zigbee2MQTT version: 1.14.3
Adapter hardware: CC26X2R1 
Firmware version: 20191106

Koenkk, 2020-08-17:
```
info  2020-08-17 13:53:31: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"entity_not_found","message":{"friendly_name":"lounge_spot_lights"}}'
error 2020-08-17 13:53:31: Entity 'lounge_spot_lights' is unknown
``` 

Is probably logged because Zigbee2MQTT receives an MQTT message on `zigbee2mqtt/lounge_spot_lights/set`, can you post the debug logging above this error?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

milobt, 2020-08-18:
Debug log level gives the same error on startup:
```
Zigbee2MQTT:info  2020-08-18 08:19:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"entity_not_found","message":{"friendly_name":"lounge_spot_lights"}}'
Zigbee2MQTT:error 2020-08-18 08:19:37: Entity 'lounge_spot_lights' is unknown

```

Remove device:
```
Zigbee2MQTT:debug 2020-08-18 08:20:51: Received MQTT message on 'zigbee2mqtt/bridge/config/remove' with data 'lounge_spot_lights'
Zigbee2MQTT:error 2020-08-18 08:20:51: Cannot remove, device 'lounge_spot_lights' does not exist
Zigbee2MQTT:info  2020-08-18 08:20:51: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed_failed","message":"lounge_spot_lights"}'
```

Force remove device:
```
Zigbee2MQTT:debug 2020-08-18 08:21:57: Received MQTT message on 'zigbee2mqtt/bridge/config/force_remove' with data 'lounge_spot_lights'
Zigbee2MQTT:error 2020-08-18 08:21:57: Cannot force remove, device 'lounge_spot_lights' does not exist
Zigbee2MQTT:info  2020-08-18 08:21:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_force_removed_failed","message":"lounge_spot_lights"}'
```

Koenkk, 2020-08-18:
Please post the log what happens before:

```
Zigbee2MQTT:info  2020-08-18 08:19:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"entity_not_found","message":{"friendly_name":"lounge_spot_lights"}}'
Zigbee2MQTT:error 2020-08-18 08:19:37: Entity 'lounge_spot_lights' is unknown
```

milobt, 2020-08-19:
https://pastebin.com/1EdBHE8U

Search for 'lounge_spot_lights'

Thanks.

Koenkk, 2020-08-19:
The paste is private, I cannot access it.

milobt, 2020-08-20:
Made public, please check again.
Thanks.

Koenkk, 2020-08-20:
This is not with debug logging enabled.

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

milobt, 2020-08-20:
Sorry, this should be with debug log level now: https://pastebin.com/rLTbC7n2

Koenkk, 2020-08-20:
As can be seen in the sniff, Zigbee2MQTT receives a request to turn off this group, but because the group does not exist this fails:

```
debug 2020-08-20 20:20:03: Received MQTT message on 'zigbee2mqtt/lounge_spot_lights/set' with data '{"state": "ON"}'
info  2020-08-20 20:20:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"entity_not_found","message":{"friendly_name":"lounge_spot_lights"}}'
```

In case you are not sending this message yourself, it might be an retained MQTT message.

milobt, 2020-09-15:
I can confirm that uninstalling and re-installing MQTT add-on purged the phantom entity.

closed 2020-09-15 by milobt