Koenkk/zigbee2mqtt #4117: Zigbee2MQTT:error 2020-08-17 17:28:03: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)

Mikepoplas, 2020-08-17:
# Bug Report
## What happened
zigbee2mqtt failed to start

## What did you expect to happen
zigbee2mqtt starts
## How to reproduce it (minimal and precise)
Happened after update, can noot be more precise sorry
## Debug Info
Zigbee2MQTT version:
Adapter hardware:  CC26X2R1 
Adapter firmware version: 3.0.0.12 20200805 hex and rev E 2.1

[https://hastebin.com/yivirekutu.sql] Logs

Also tried  pan_id: 6754 and also reflashed CC26X2R1 with no issues but same error.
Saw this issue but no corruption on device or plug/unplug/reboot worked as reported in other issues.

Regards

Mikepoplas, 2020-08-17:
Restoring a 1 week old proxmox backup does not solve the issue and that is strange, it was working for sure 1 week ago, same error.

olili, 2020-08-18:
Had the same issue. 
In my my case it was an interference with an existing zigbee network with same pan id. 
Simple change of pan id and channel solved the issue.

Mikepoplas, 2020-08-18:
Already tried that but no luck.

Koenkk, 2020-08-18:
When updating the firmware of the adapter, did you execute step 5 of https://www.zigbee2mqtt.io/information/flashing_via_uniflash.html

Mikepoplas, 2020-08-19:
Nope, but I reflashed and did it now, same result.

Koenkk, 2020-08-20:
Have you also tried `/dev/ttyACM1`?

Mikepoplas, 2020-08-20:
Same error:
Zigbee2MQTT:debug 2020-08-20 09:26:58: Using zigbee-herdsman with settings: '{"network":{"panID":6755,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[26],"networkKey":"HIDDEN"},"databasePath":"/share/zigbee2mqtt/database.db","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","backupPath":"/share/zigbee2mqtt/coordinator_backup.json","serialPort":{"path":"/dev/ttyACM1"},"adapter":{"concurrent":null}}'
Zigbee2MQTT:error 2020-08-20 09:27:18: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-08-20 09:27:18: Failed to start zigbee
Zigbee2MQTT:error 2020-08-20 09:27:18: Exiting...
Zigbee2MQTT:error 2020-08-20 09:27:18: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/zigbee2mqtt-1.14.3/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:87:31)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.14.3/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm 
ERR! zigbee2mqtt@1.14.3 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.14.3 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-08-20T07_27_18_559Z-debug.log
2020-08-20T09:27:18: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
Gonna try a clean install in a new nvme that arrives today to discard weird fs corruption issues even if nvme and smartctl info seems ok and there is no error logged. 

Regards

R91g, 2020-08-20:
I have the same issue after upgrade my CC2531 with a CC26X2R1.

I flashed with the firmware CC26X2R1_20200805.hex and with all the steps described in:
https://www.zigbee2mqtt.io/information/flashing_via_uniflash.html

**zigbee2mqtt version 1.14.3**

[Configuration](https://hastebin.com/amumejotuk.makefile)
[Logs](https://hastebin.com/hevobuheye.sql)

I tried change the pain_id, channel and network_key

waltersweb, 2020-08-20:
Same here, tried with a Raspberry Pi 3b iObroker, Raspberry 4 and also with CC2531 and conbee II stick. Nothing works

zigbee.0 | 2020-08-20 18:14:39.827 | error | (4238) Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'     at Znp.<anonymous> (/opt/iobroker/node_modules/iobroker.zigbee/node_modules/zigbee-her
-- | -- | -- | --
zigbee.0 | 2020-08-20 18:14:39.825 | error | (4238) Failed to start Zigbee
zigbee.0 | 2020-08-20 18:14:39.542 | info | (4238) Starting Zigbee...
zigbee.0 | 2020-08-20 18:14:39.541 | info | (4238) Try to reconnect. 1 attempts left
zigbee.0 | 2020-08-20 18:14:29.538 | error | (4238) Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)     at ZStackAdapter.<anonymous> (/opt/iobroker/node_modules/iobroker.zigbee/node_modules/zigbee-herdsman/dist/ad
zigbee.0 | 2020-08-20 18:14:29.536 | error | (4238) Failed to start Zigbee
zigbee.0 | 2020-08-20 18:14:09.179 | info | (4238) Starting Zigbee...

Koenkk, 2020-08-20:
@Mikepoplas to figure out wether the adapter is the issue, can you try on the most simple setup as possible? (e.g. raspberry pi by following: https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html#running-zigbee2mqtt)

@R91g did it ever work with the CC2652? If not make sure you have revision E, can be checked in uniflash:

[image]

@waltersweb in your case the adpater is already locked up by another application. see https://www.zigbee2mqtt.io/information/FAQ.html#error-resource-temporarily-unavailable-cannot-lock-port

waltersweb, 2020-08-21:
@Koenkk That is actually the second message I get (lowest line of the log is the oldest). How I read the errors:

- Adapter is starting
- Adapter tries to ping the Stick but says it's unreachable after 6 Seconds
- trying a second time
- now the port can't be locked (I guess because it is still bussy with whatever is going on with the initial ping?)

I ran your ls -l /proc/[0-9]/fd/ |grep /dev/ttyACM0 and first it told me that I can't open that folder since I dont have the rights (Pi User). When I run the same command with sudo in front, it doesn't come back with anything. same when I log in with root or to sudo su. does that mean that there is no running application?
I ran that line both before starting adapter, in between errors one and two and after the second error. Always empty

Koenkk, 2020-08-23:
@waltersweb can you provide the herdsman debug logging of starting zigbee2mqtt?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

waltersweb, 2020-08-24:
okay this is strange... I used deconz setup for the last days since I had a conbee II stick at hand now anyways. That worked fine without any issues. 
I replaced the direct connected conbee II and instead put a usb extension cable in between the stick and the raspberry since I found somewhere that it might improve the meshing function of deconz.

Now I still wanted to help with this issue and deactivated the deconz setup, went to the iobroker zigbee instance and activated the herdsman logging, started the instance and... it went to green right away... what the... I rebooted the raspberry and still it is working.
Now I am not sure the USB cable was the fix... But it might be?

Mikepoplas, 2020-08-24:
@Koenkk same in bare metal:

added 774 packages in 15.147s
root@kasa:/opt/zigbee2mqtt# nano /opt/zigbee2mqtt/data/configuration.yaml
root@kasa:/opt/zigbee2mqtt# cd /opt/zigbee2mqtt
root@kasa:/opt/zigbee2mqtt# 
root@kasa:/opt/zigbee2mqtt# npm start

> zigbee2mqtt@1.14.3 start /opt/zigbee2mqtt
> node index.js

Zigbee2MQTT:info  2020-08-25 00:20:43: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-08-25.00-20-43' filename: log.txt
Zigbee2MQTT:debug 2020-08-25 00:20:43: Can't load state from file /opt/zigbee2mqtt/data/state.json (doesn't exist)
Zigbee2MQTT:info  2020-08-25 00:20:43: Starting Zigbee2MQTT version 1.14.3 (commit #0db3f3c)
Zigbee2MQTT:info  2020-08-25 00:20:43: Starting zigbee-herdsman...
Zigbee2MQTT:debug 2020-08-25 00:20:43: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[25],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"path":"/dev/ttyACM0"},"adapter":{"concurrent":null}}'
Zigbee2MQTT:error 2020-08-25 00:21:03: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-08-25 00:21:03: Failed to start zigbee
Zigbee2MQTT:error 2020-08-25 00:21:03: Exiting...
Zigbee2MQTT:error 2020-08-25 00:21:03: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:87:31)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.14.3 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.14.3 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-08-24T22_21_03_482Z-debug.log
root@kasa:/opt/zigbee2mqtt# ls -l /dev/serial/by-id
total 0
lrwxrwxrwx 1 root root 13 Aug 24 12:03 usb-Texas_Instruments_XDS110__03.00.00.07__Embed_with_CMSIS-DAP_L1100F1X-if00 -> ../../ttyACM0
lrwxrwxrwx 1 root root 13 Aug 24 12:03 usb-Texas_Instruments_XDS110__03.00.00.07__Embed_with_CMSIS-DAP_L1100F1X-if03 -> ../../ttyACM1
root@kasa:/opt/zigbee2mqtt# test -w /dev/ttyACM0 && echo success || echo failure 
success

Regards

Koenkk, 2020-08-25:
@Mikepoplas strange! In your previous post you wrote:

 > with a fresh new CC26X2R1 never flashed before with same results.

I don't understand this, do you mean you didn't flash a firmware? Also what revision is this: https://github.com/Koenkk/zigbee2mqtt/issues/4117#issuecomment-677842270 ?

Mikepoplas, 2020-08-25:
@Koenkk I mean I also tried a new CC26X2R1 board to discard weird errors, I had a spare one. Revision is E 2.1. Indeed, it's strange.

Regards

Mikepoplas, 2020-08-26:
@Koenkk Unplug CC26X2R1 and plugin CC2531 in same vm or bare metal solves the problem, so issue it's at/related to CC26X2R1, any idea to advance/debug on this?

Regards

DrTron, 2020-08-26:
Do you guys by any chance happen to run openhab2 with the zwave-binding?
That's what got me with a CC2652RB-stick:
https://community.openhab.org/t/apparently-the-zwave-binding-blocks-the-dev-ttyusb0-port-in-combination-with-a-cc2652rb-zigbee2mqtt-dongle/103245

Koenkk, 2020-08-26:
@Mikepoplas I assume that nothing else is running on the bare metal installation? (to rule out problems like @DrTron).

Can you try pressing the buttons of the board, one of them should reset the board, try to start zigbee2mqtt after this.

oldfart101, 2020-08-26:
Operating environment (HassOS):
Raspberry PI 4 (4 GB model)
HassOS 4.12 / Supervisor version 235
Home Assistant 0.114.3

flashed CC2531 Zigbee Stick [Lumi]

serial:
/dev/ttyACM0
/dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0014D956FC-if00
/dev/ttyAMA0

Zigbee working in homeassistant with /dev/serial/by-id/usb-Texas_Instruments...

**zigbee2mqtt config:**

data_path: /share/zigbee2mqtt
devices: devices.yaml
groups: groups.yaml
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: '192.168.1.10:1885' # mqtt broker running on Ubuntu VM on Unraid
  user: username
  password: password
serial:
  port: >-
    /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0014D956FC-if00
advanced:
  pan_id: 6754
  channel: 11
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  availability_blacklist: []
ban: []
whitelist: []
queue: {}
socat:
  enabled: false
  master: 'pty,raw,echo=0,link=/dev/ttyZ2M,mode=777'
  slave: 'tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5'
  restartdelay: 1
  initialdelay: 1
  options: '-d -d'
  log: false

**zigbee2mqtt error log:**

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
./run.sh: line 17: [Info] Configuration backup found in /share/zigbee2mqtt/.configuration.yaml.bk. Skipping config backup.: No such file or directory
[Info] Socat is DISABLED and not started
2020-08-26T09:35:54: PM2 log: Launching in no daemon mode
2020-08-26T09:35:54: PM2 log: App [npm:0] starting in -fork mode-
2020-08-26T09:35:54: PM2 log: App [npm:0] online
> zigbee2mqtt@1.14.3 start /zigbee2mqtt-1.14.3
> node index.js
Zigbee2MQTT:info  2020-08-26 09:35:57: Logging to console and directory: '/share/zigbee2mqtt/log/2020-08-26.09-35-57' filename: log.txt
Zigbee2MQTT:info  2020-08-26 09:35:58: Starting Zigbee2MQTT version 1.14.3 (commit #unknown)
Zigbee2MQTT:info  2020-08-26 09:35:58: Starting zigbee-herdsman...
Zigbee2MQTT:error 2020-08-26 09:36:17: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-08-26 09:36:17: Failed to start zigbee
Zigbee2MQTT:error 2020-08-26 09:36:17: Exiting...
Zigbee2MQTT:error 2020-08-26 09:36:17: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/zigbee2mqtt-1.14.3/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:87:31)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.14.3/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm
 ERR! zigbee2mqtt@1.14.3 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.14.3 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-08-26T08_36_17_880Z-debug.log
2020-08-26T09:36:18: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2020-08-26T09:36:18: PM2 log: App [npm:0] starting in -fork mode-
2020-08-26T09:36:18: PM2 log: App [npm:0] online
> zigbee2mqtt@1.14.3 start /zigbee2mqtt-1.14.3
> node index.js
Zigbee2MQTT:info  2020-08-26 09:36:21: Logging to console and directory: '/share/zigbee2mqtt/log/2020-08-26.09-36-21' filename: log.txt
Zigbee2MQTT:info  2020-08-26 09:36:22: Starting Zigbee2MQTT version 1.14.3 (commit #unknown)
Zigbee2MQTT:info  2020-08-26 09:36:22: Starting zigbee-herdsman...
Zigbee2MQTT:error 2020-08-26 09:36:42: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-08-26 09:36:42: Failed to start zigbee
Zigbee2MQTT:error 2020-08-26 09:36:42: Exiting...
Zigbee2MQTT:error 2020-08-26 09:36:42: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/zigbee2mqtt-1.14.3/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:87:31)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.14.3/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm
 ERR! zigbee2mqtt@1.14.3 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.14.3 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-08-26T08_36_42_264Z-debug.log
2020-08-26T09:36:43: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2020-08-26T09:36:43: PM2 log: App [npm:0] starting in -fork mode-
2020-08-26T09:36:43: PM2 log: App [npm:0] online
> zigbee2mqtt@1.14.3 start /zigbee2mqtt-1.14.3
> node index.js
Zigbee2MQTT:info  2020-08-26 09:36:45: Logging to console and directory: '/share/zigbee2mqtt/log/2020-08-26.09-36-45' filename: log.txt
Zigbee2MQTT:info  2020-08-26 09:36:46: Starting Zigbee2MQTT version 1.14.3 (commit #unknown)
Zigbee2MQTT:info  2020-08-26 09:36:46: Starting zigbee-herdsman...

Mikepoplas, 2020-08-28:
It seams all was caused by uniflash, strange:
[image]

Regards

Koenkk, 2020-08-28:
@Mikepoplas you found the issue? It works now?

Mikepoplas, 2020-08-28:
@Koenkk yes, no idea exactly what was going on, for sure at least a couple of things at same time, pan_id + flash error. Tried a lot of things so I cannot be sure, sorry about it.

Regards

closed 2020-08-28 by Mikepoplas