Koenkk/zigbee2mqtt #4119: Kwikset 910 (99100-045) unsupported

zombielinux, 2020-08-17:
# Bug Report
## What happened
Kwikset 910 lock did not fully integrate itself with Zigbee2mqtt

## What did you expect to happen

Lock to pair and be usable. 

## How to reproduce it (minimal and precise)

Put zigbee2mqtt into permit_join mode. Follow Kwikset instructions to pair device to network.

## Debug Info
Zigbee2MQTT version: Latest docker pull
Adapter hardware: Conbee II

Without any changes to devices.js, device comes in as "unsupported" with a model of "SMARTCODE_DEADBOLT_5"

Copying the "SMARTCODE_DEADBOLT_5_L" entry yeilds a pairing with lock/unlock events as well as keypad lock unlock events. 

It also yeilds the occasional message: 

```
Failed to configure '0x002446fffd0070eb', attempt 1 (TypeError: Cannot read property 'message' of undefined
at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:266:57)
at Generator.throw (<anonymous>)
at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:25:65))
```

database.db shows:

```
{"id":2,"type":"EndDevice","ieeeAddr":"0x002446fffd0070eb","nwkAddr":28674,"manufId":4242,"manufName":"Kwikset","powerSource":"Battery","modelId":"SMARTCODE_DEADBOLT_5","epList":[2],"endpoints":{"2":{"profId":260,"epId":2,"devId":10,"inClusterList":[0,1,3,4,5,9,32,257,1026,2821,64957],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"SMARTCODE_DEADBOLT_5","manufacturerName":"Kwikset","powerSource":3,"zclVersion":1,"appVersion":28,"stackVersion":2,"hwVersion":3}}},"binds":[]}},"appVersion":28,"stackVersion":2,"hwVersion":3,"zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1597683096551}
```

zombielinux, 2020-09-17:
Update: 

I've actually got it working with the following config:

```
    {
        zigbeeModel: ['SMARTCODE_DEADBOLT_5'],
        model: '99100-045',
        vendor: 'Kwikset',
        description: '910 SmartCode traditional electronic deadbolt',
        supports: 'lock/unlock, battery',
        fromZigbee: [fz.lock, fz.lock_operation_event, fz.battery, fz.lock_programming_event, fz.lock_pin_code_rep],
        toZigbee: [tz.generic_lock, , tz.pincode_lock],
        meta: {configureKey: 4, pinCodeCount: 30},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(2);
            console.log(device);
            console.log(endpoint.clusters);
            await bind(endpoint, coordinatorEndpoint, ['closuresDoorLock', 'genPowerCfg']);
            await configureReporting.lockState(endpoint);
            await configureReporting.batteryPercentageRemaining(endpoint);
        },
    },
```

closed 2020-10-25 by github-actions[bot]

Langley74, 2023-02-20:
> Update:
>
> I've actually got it working with the following config:
>
> ```
>     {
>         zigbeeModel: ['SMARTCODE_DEADBOLT_5'],
>         model: '99100-045',
>         vendor: 'Kwikset',
>         description: [...]

Where do you place this code?