Koenkk/zigbee2mqtt #4122: lutron connected remote Paired but doesn't work

surshis, 2020-08-18:
Hassio on raspberry 3. Latest zigbee2mqtt.
Cc2531 flashed with 2018 firmware.

Everything works fine  Except Lutron connected bulb remote LCBR. Successfully paired, log shows all steps successful. Creates 2 sensors- remote action and linkquality. But both sensors show status - Unknown. 
Pressing any button on the remote doesn't result in any further messages. 
Read zigbee2mqtt documentation. Says LCBR is supported and should generate the button press galues 
Looking to use in automations based on button press.
What am I doing wrong?

Koenkk, 2020-08-18:
Can you try changing the definition in `devices.js` to:

```js
{
    zigbeeModel: ['LZL4BWHL01 Remote'],
    model: 'LZL4BWHL01',
    vendor: 'Lutron',
    description: 'Connected bulb remote control',
    supports: 'on/off, brightness',
    fromZigbee: [fz.insta_down_hold, fz.insta_up_hold, fz.LZL4B_onoff, fz.insta_stop],
    toZigbee: [],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
    },
},
```

Make sure that the configure is successfully, you can trigger it manually via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure . Probably you need wakeup the device by clicking a button on it before sending the command.

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html how to edit `devices.js`.

surshis, 2020-08-19:
Thanks did that, getting this error message 
Zigbee2MQTT:error 2020-08-18 22:13:10: Failed to configure 'lutron remote', attempt 1 (Error: Bind 0xffff000fe7ff****/1 genOnOff from '0x00124b001938****/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (/zigbee2mqtt-1.14.3/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7))

Last 4 digits masked.

Koenkk, 2020-08-19:
Can you provide the herdsman debug logging when pressing buttons on the device?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

surshis, 2020-08-20:
Thanks for the response.
Here's the log- (mostly there is no response at button press on the remote. i get this response sometimes only). thanks.
Zigbee2MQTT:debug 2020-08-19 21:59:29: Device 'lutron' announced itself
Zigbee2MQTT:info  2020-08-19 21:59:29: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"lutron"}}'
2020-08-20T04:59:30.870Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,127,35,90,9,255,231,15,0,255,255,65,232,50]
2020-08-20T04:59:30.871Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,127,35,90,9,255,231,15,0,255,255,65,232,50]
2020-08-20T04:59:30.871Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [127,35,90,9,255,231,15,0,255,255,65,232] - 50
2020-08-20T04:59:30.872Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":9087,"extaddr":"0xffff000fe7ff095a","parentaddr":59457}
2020-08-20T04:59:30.873Z zigbee-herdsman:controller:log Device '0xffff000fe7ff095a' joined
2020-08-20T04:59:30.873Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-08-20T04:59:30.874Z zigbee-herdsman:controller:log Device '0xffff000fe7ff095a' accepted by handler
2020-08-20T04:59:30.874Z zigbee-herdsman:controller:log Not interviewing '0xffff000fe7ff095a', completed 'true', in progress 'false'
2020-08-20T04:59:30.994Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,13,69,193,127,35,127,35,90,9,255,231,15,0,255,255,128,77]
2020-08-20T04:59:30.995Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,193,127,35,127,35,90,9,255,231,15,0,255,255,128,77]
2020-08-20T04:59:30.995Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 13 - 2 - 5 - 193 - [127,35,127,35,90,9,255,231,15,0,255,255,128] - 77
2020-08-20T04:59:30.996Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - endDeviceAnnceInd - {"srcaddr":9087,"nwkaddr":9087,"ieeeaddr":"0xffff000fe7ff095a","capabilities":128}
2020-08-20T04:59:30.997Z zigbee-herdsman:controller:log Device announce '0xffff000fe7ff095a'
Zigbee2MQTT:debug 2020-08-19 21:59:30: Device 'lutron' announced itself
Zigbee2MQTT:info  2020-08-19 21:59:31: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"lutron"}}'
2020-08-20T04:59:31.002Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []

surshis, 2020-08-21:
Thanks. Since thete was nothing wrong in the logs, but it was still not working, decided to flash the latest firmware on the cc2531. Can confirm the Lutron LCBR remote is now working properly. Thanks.

closed 2020-08-21 by surshis