Koenkk/zigbee2mqtt #4123: Invalid JSON 'LOCK', skipping...

qosmio, 2020-08-18:
Between c66e65e..b740929 something changed either in the herdsman code or z2m code. It is not longer recognizing bare word "unlock" or "lock" in the payload, instead it's requiring it be in a complete json object: "{"state": "LOCK"}" or "{"state": "UNLOCK"}"

# Bug Report
## What happened
When using an entity button to toggle the locks open/close an error is being thrown: Invalid JSON 'LOCK', skipping...
## What did you expect to happen
Allow the bare word in the payload like previous behavior
## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2MQTT version:
Adapter hardware: CC2538
Adapter firmware version:

sjorge, 2020-08-18:
I'm pretty sure it is this commit:

https://github.com/Koenkk/zigbee2mqtt/commit/8c5fae900eb5e7ef2a41005c584a41612ddfe308

qosmio, 2020-08-18:
> 8c5fae9

Yup, this most definitely was the culprit. Adding unlock, lock to the list on the latest branch allows it to work properly.

Koenkk, 2020-08-18:
Fixed! thanks for reporting.

closed 2020-08-18 by Koenkk