Koenkk/zigbee2mqtt #4124: 2018 Smasung SmartThings MultiPurpose Sensor not reporting battery

liamstears, 2020-08-18:
With my quick tests I'm sure this will be a quick fix for you

It is correctly identified as the correct device which you have listed as:

```
    {
        zigbeeModel: ['motion'],
        model: 'IM6001-MTP01',
        vendor: 'SmartThings',
        description: 'Motion sensor (2018 model)',
        supports: 'occupancy and temperature',
        fromZigbee: [
            fz.temperature,
            fz.ignore_iaszone_report,
            fz.ias_occupancy_alarm_1, fz.battery_3V,
        ],
        toZigbee: [],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['msTemperatureMeasurement', 'genPowerCfg']);
            await configureReporting.temperature(endpoint);
            await configureReporting.batteryVoltage(endpoint);
        },
    },

```

However battery does not work but all other functions do.

For a quick test I ran it with the configuration for the 2015 model here:

```
    {
        zigbeeModel: ['3321-S'],
        model: '3321-S',
        vendor: 'SmartThings',
        description: 'Multi Sensor (2015 model)',
        supports: 'contact and temperature',
        fromZigbee: [fz.temperature, fz.smartsense_multi, fz.ias_contact_alarm_1, fz.battery_cr2450],
        toZigbee: [],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['msTemperatureMeasurement']);
            await endpoint.read('genPowerCfg', ['batteryVoltage']);
            await configureReporting.temperature(endpoint);
            await configureReporting.batteryVoltage(endpoint);
        },
    },

```

With this the battery is correctly reported but the temperature is not

I do not know enough about how this code works to combine the 2 to get this working 100% but I am sure you will know how and can quickly resolve this problem for me...

Koenkk, 2020-08-18:
Can you share the `data/database.db` entry of this device?

liamstears, 2020-08-18:
> Can you share the `data/database.db` entry of this device?

I would love to but for some reason when I try to add the device again now it's not adding. Tried the edge version and literally nothing in the logs. What's weird is it pairs to smartthings no problem.

Any ideas?

liamstears, 2020-08-18:
> Can you share the `data/database.db` entry of this device?

Nevermind, got it, needed to restart my whole pi - odd?

Here it is:

```
{"id":5,"type":"EndDevice","ieeeAddr":"0x286d970001009fc3","nwkAddr":55356,"manufId":4673,"manufName":"Samjin","powerSource":"Battery","modelId":"multi","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,32,1026,1280,2821,64514],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"multi","manufacturerName":"Samjin","powerSource":3,"zclVersion":2,"appVersion":17,"hwVersion":0,"swBuildId":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0019389d70","zoneState":0}}},"binds":[{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b0019389d70","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0019389d70","endpointID":1}]}},"appVersion":17,"hwVersion":0,"swBuildId":"","zclVersion":2,"interviewCompleted":false,"meta":{"configured":1},"lastSeen":1597784817991}

```

Koenkk, 2020-08-19:
Can you try with:

```js
{
    zigbeeModel: ['motion'],
    model: 'IM6001-MTP01',
    vendor: 'SmartThings',
    description: 'Motion sensor (2018 model)',
    supports: 'occupancy and temperature',
    fromZigbee: [
        fz.temperature,
        fz.ignore_iaszone_report,
        fz.ias_occupancy_alarm_1, fz.battery,
    ],
    toZigbee: [],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['msTemperatureMeasurement', 'genPowerCfg']);
        await configureReporting.temperature(endpoint);
        await configureReporting.batteryPercentageRemaining(endpoint);
    },
},
```

Force a manual configure via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure and check the log if its ok (might need to wakeup the device before executing the command). If it's OK I expect the battery to be reported within 30 hours.

liamstears, 2020-08-19:
> Can you try with:
>
> ```js
> {
>     zigbeeModel: ['motion'],
>     model: 'IM6001-MTP01',
>     vendor: 'SmartThings',
>     description: 'Motion sensor (2018 model)',
>     supports: 'occupancy and [...]

I am so sorry I actually quoted the wrong device at the beginning of this thread, it should have been:

```
    {
        zigbeeModel: ['multi'],
        model: 'IM6001-MPP01',
        vendor: 'SmartThings',
        description: 'Multipurpose sensor (2018 model)',
        supports: 'contact',
        fromZigbee: [fz.temperature, fz.ias_contact_alarm_1, fz.battery_3V, fz.ignore_iaszone_attrepo
        toZigbee: [],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['msTemperatureMeasurement', 'genPowerCfg']);
            await configureReporting.temperature(endpoint);
            await configureReporting.batteryVoltage(endpoint);
        },
    },

```

Does the same still apply anyway with the 2 changes? And 30 hours? When I changed to the older sensor config it showed battery level instantly once adding the device...

liamstears, 2020-08-19:
I have been playing around with the code and the following seems to be working including battery and temperature although it did take 10mins before the temperature started showing and working:

```
    {
        zigbeeModel: ['multi'],
        model: 'IM6001-MPP01',
        vendor: 'SmartThings',
        description: 'Multipurpose sensor (2018 model)',
        supports: 'contact',
        fromZigbee: [fz.temperature, fz.ias_contact_alarm_1, fz.battery_cr2450, fz.ignore_iaszone_att
        toZigbee: [],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['msTemperatureMeasurement', 'genPowerCfg']);
            await endpoint.read('genPowerCfg', ['batteryVoltage']);
            await configureReporting.temperature(endpoint);
            await configureReporting.batteryPercentageRemaining(endpoint);
        },
    },

```

As per your post/suggestion above I'm going to leave the following code going and see how it goes:

```
    {
        zigbeeModel: ['multi'],
        model: 'IM6001-MPP01',
        vendor: 'SmartThings',
        description: 'Multipurpose sensor (2018 model)',
        supports: 'contact',
        fromZigbee: [fz.temperature, fz.ias_contact_alarm_1, fz.battery, fz.ignore_iaszone_attreport]
        toZigbee: [],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['msTemperatureMeasurement', 'genPowerCfg']);
            await configureReporting.temperature(endpoint);
            await configureReporting.batteryPercentageRemaining(endpoint);
        },
    },

```

liamstears, 2020-08-20:
Can confirm battery is now showing after around 20 hours with the following:

```
    {
        zigbeeModel: ['multi'],
        model: 'IM6001-MPP01',
        vendor: 'SmartThings',
        description: 'Multipurpose sensor (2018 model)',
        supports: 'contact',
        fromZigbee: [fz.temperature, fz.ias_contact_alarm_1, fz.battery, fz.ignore_iaszone_attreport]
        toZigbee: [],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['msTemperatureMeasurement', 'genPowerCfg']);
            await configureReporting.temperature(endpoint);
            await configureReporting.batteryPercentageRemaining(endpoint);
        },
    },

```

Koenkk, 2020-08-23:
@liamstears this will be in the next zibee2mqtt release (1.5 weeks from now). Reporting of battery is limited since this value to save battery :)