Koenkk/zigbee2mqtt #4125: OSRAM Gardenpole Mini RGBW colours do not match settings

iFloris, 2020-08-18:
# Bug Report
## What happened
A light displayed a different colour to what I selected in Home Assistant.
For instance, any type of whiteish colour = off. Blue = green. Any warm white tint = red.
Would it help if I made a video?

## What did you expect to happen
The light to display the colour I selected.

## How to reproduce it (minimal and precise)
Open up Home Assistant. Select any colour. The colour is never what was selected.

## Debug Info
Zigbee2MQTT version:
Adapter hardware: CC1352P-2
Adapter firmware version: 20200805
Version of Zigbee2Mqtt: 1.14.3 
Model is Gardenpole Mini RGBW OSRAM by OSRAM
Firmware version is V1.05.10

Koenkk, 2020-08-18:
Could you provide the debug log when setting the color and the color you expect it to be?

iFloris, 2020-08-19:
```Zigbee2MQTT:debug 2020-08-19 11:04:30: Received MQTT message on 'zigbee2mqtt/VH tuin padverlichting/set' with data '{"state": "ON"}'
Zigbee2MQTT:debug 2020-08-19 11:04:30: Publishing 'set' 'state' to 'VH tuin padverlichting'
Zigbee2MQTT:info  2020-08-19 11:04:30: MQTT publish: topic 'zigbee2mqtt/VH tuin padverlichting', payload '{"state":"ON","last_seen":"2020-08-19T11:04:30+02:00","update_available":false,"color":{"x":0.146,"y":0.238},"color_temp":153,"brightness":254,"linkquality":75}'
Zigbee2MQTT:debug 2020-08-19 11:04:30: Received Zigbee message from 'VH tuin padverlichting', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":254}' from endpoint 3 with groupID 0
Zigbee2MQTT:info  2020-08-19 11:04:30: MQTT publish: topic 'zigbee2mqtt/VH tuin padverlichting', payload '{"state":"ON","last_seen":"2020-08-19T11:04:30+02:00","update_available":false,"color":{"x":0.146,"y":0.238},"color_temp":153,"brightness":254,"linkquality":75}'
Zigbee2MQTT:debug 2020-08-19 11:04:30: Received Zigbee message from 'VH tuin padverlichting', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 3 with groupID 0
Zigbee2MQTT:info  2020-08-19 11:04:30: MQTT publish: topic 'zigbee2mqtt/VH tuin padverlichting', payload '{"state":"ON","last_seen":"2020-08-19T11:04:30+02:00","update_available":false,"color":{"x":0.146,"y":0.238},"color_temp":153,"brightness":254,"linkquality":75}'
Zigbee2MQTT:debug 2020-08-19 11:04:31: Received Zigbee message from 'AH archief lamp', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-08-19 11:04:31: Successfully pinged 'AH archief lamp'
Zigbee2MQTT:debug 2020-08-19 11:04:31: Received Zigbee message from 'VH achterdeur keuken', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-08-19 11:04:31: Successfully pinged 'VH achterdeur keuken'
Zigbee2MQTT:debug 2020-08-19 11:04:32: Received Zigbee message from 'VH keuken repeater', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-08-19 11:04:32: MQTT publish: topic 'zigbee2mqtt/VH keuken repeater', payload '{"linkquality":75,"last_seen":"2020-08-19T11:04:32+02:00","update_available":false}'
Zigbee2MQTT:debug 2020-08-19 11:04:32: Successfully pinged 'VH keuken repeater'
Zigbee2MQTT:debug 2020-08-19 11:04:34: Received Zigbee message from 'VH werkkamer Eline repeater', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-08-19 11:04:34: MQTT publish: topic 'zigbee2mqtt/VH werkkamer Eline repeater', payload '{"linkquality":75,"last_seen":"2020-08-19T11:04:34+02:00","update_available":false}'
Zigbee2MQTT:debug 2020-08-19 11:04:34: Successfully pinged 'VH werkkamer Eline repeater'
Zigbee2MQTT:debug 2020-08-19 11:04:35: Received MQTT message on 'zigbee2mqtt/VH tuin padverlichting/set' with data '{"state": "ON", "color": {"x": 0.701, "y": 0.299}}'
Zigbee2MQTT:debug 2020-08-19 11:04:35: Skipping state because of Home Assistant
Zigbee2MQTT:debug 2020-08-19 11:04:35: Publishing 'set' 'color' to 'VH tuin padverlichting'
Zigbee2MQTT:info  2020-08-19 11:04:35: MQTT publish: topic 'zigbee2mqtt/VH tuin padverlichting', payload '{"state":"ON","last_seen":"2020-08-19T11:04:35+02:00","update_available":false,"color":{"x":0.701,"y":0.299},"color_temp":37,"brightness":254,"linkquality":75}'
Zigbee2MQTT:debug 2020-08-19 11:04:35: Received Zigbee message from 'VH tuin padverlichting', type 'attributeReport', cluster 'lightingColorCtrl', data '{"currentX":45940}' from endpoint 3 with groupID 0
Zigbee2MQTT:info  2020-08-19 11:04:35: MQTT publish: topic 'zigbee2mqtt/VH tuin padverlichting', payload '{"state":"ON","last_seen":"2020-08-19T11:04:35+02:00","update_available":false,"color":{"x":0.701,"y":0.299},"color_temp":37,"brightness":254,"linkquality":75}'
Zigbee2MQTT:debug 2020-08-19 11:04:35: Received Zigbee message from 'VH tuin padverlichting', type 'attributeReport', cluster 'lightingColorCtrl', data '{"currentY":19077}' from endpoint 3 with groupID 0
Zigbee2MQTT:info  2020-08-19 11:04:35: MQTT publish: topic 'zigbee2mqtt/VH tuin padverlichting', payload '{"state":"ON","last_seen":"2020-08-19T11:04:35+02:00","update_available":false,"color":{"x":0.701,"y":0.2911},"color_temp":37,"brightness":254,"linkquality":78}'
Zigbee2MQTT:debug 2020-08-19 11:04:36: Received MQTT message on 'zigbee2mqtt/VH tuin padverlichting/set' with data '{"state": "ON", "color": {"x": 0.146, "y": 0.238}}'
Zigbee2MQTT:debug 2020-08-19 11:04:36: Skipping state because of Home Assistant
Zigbee2MQTT:debug 2020-08-19 11:04:36: Publishing 'set' 'color' to 'VH tuin padverlichting'
Zigbee2MQTT:info  2020-08-19 11:04:36: MQTT publish: topic 'zigbee2mqtt/VH tuin padverlichting', payload '{"state":"ON","last_seen":"2020-08-19T11:04:36+02:00","update_available":false,"color":{"x":0.146,"y":0.238},"color_temp":60,"brightness":254,"linkquality":78}'
Zigbee2MQTT:debug 2020-08-19 11:04:37: Received MQTT message on 'zigbee2mqtt/VH tuin padverlichting/set' with data '{"state": "ON", "color": {"x": 0.259, "y": 0.122}}'
Zigbee2MQTT:debug 2020-08-19 11:04:37: Skipping state because of Home Assistant
Zigbee2MQTT:debug 2020-08-19 11:04:37: Publishing 'set' 'color' to 'VH tuin padverlichting'
[... cut ...]
Zigbee2MQTT:debug 2020-08-19 11:04:49: Received Zigbee message from 'VH tuin padverlichting', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":254}' from endpoint 3 with groupID 0
Zigbee2MQTT:info  2020-08-19 11:04:49: MQTT publish: topic 'zigbee2mqtt/VH tuin padverlichting', payload '{"state":"OFF","last_seen":"2020-08-19T11:04:49+02:00","update_available":false,"color":{"x":0.146,"y":0.238},"color_temp":60,"brightness":254,"linkquality":75}'
``` 
Hello @Koenk 
Above is the log for quickly changing colors between blue and red, as well as sliding the temperature slider from blue to orange.
Is this what you meant or did you mean to select a very specific color?

Koenkk, 2020-08-19:
If we take e.g. 

```
Zigbee2MQTT:debug 2020-08-19 11:04:35: Received MQTT message on 'zigbee2mqtt/VH tuin padverlichting/set' with data '{"state": "ON", "color": {"x": 0.701, "y": 0.299}}'
```

- Can you provide the herdsman debug logging of this?
- What color is shown on the bulb? (this should be red)

closed 2020-09-26 by github-actions[bot]