Koenkk/zigbee2mqtt #4127: Homeassistant integration of devices consisting of multiple endpoints

vdkeybus, 2020-08-18:
# Bug Report
## What happened
Trying to add support for the iCasa RM11S remote control.

It consists of 4 light control sensors, a master on-off switch and 2 scene selectors. The light control sensors are a pretty common on/off (short presses) or dim-up/down (long presses) affair. The device emits 'action' and 'brightness' properties for that, from endpoints 1 to 4.

Pressing the master on- or off-switch acts as if the 4 channel on- or off-switches are pressed simultaneously. 4 messages are sent per press. Same for the scene recall switches.

## What did you expect to happen
The problem is that this device is represented by a single device in HA, whereas it should actually be represented by 4 devices.
E.g. now it is device 0xccccccfffe26a33a, but it should be 0xccccccfffe26a33a/1, 0xccccccfffe26a33a/2, ... 0xccccccfffe26a33a/4 or something similar. I need a hint where I would best implement this. It seems that the endpoint is not accessible in lib/extensions/homeassistant.js.

## How to reproduce it (minimal and precise)
N/A

## Debug Info
Zigbee2MQTT version: latest dev
Adapter hardware: N/A
Adapter firmware version: N/A

vdkeybus, 2020-08-18:
Hold the horses. It seems that pressing the master on- or off-buttons all originate from endpoint 1, also that ones the relate to channels 2 to 4.

The problem redefines itself to: how to access the group id in homeassistant.js. There is an action_group property though, but it is cleaner to refer to the actual group number I think.

And then: is it possible to redefine the group ids on this remote with the bind command (considering the endpoints).

Koenkk, 2020-08-19:
Such device are currently integrated as one action sensor, the action itself should contain the endpoint (e.g. `single_l1`, `hold_left`, etc.)

vdkeybus, 2020-08-19:
Ok, but (how) can I access the 'endpoint' or 'groupID' to form an action in
homeassistant.js ?

Koenkk, 2020-08-19:
I think you need to grab the attributes from the action sensor in your automation https://www.home-assistant.io/docs/configuration/templating/#attributes

vdkeybus, 2020-08-19:
Thanks for the hint.

I did not realize that homeassistant.js runs on behalf of HA.

However this device sends the following type of message:
payload
'{"action":"on","action_group":26370,"battery":43.5,"brightness":13,"click":"on","linkquality":49,"rate":50}'

Unless I fix the action_group numbers (not even sure how I would do that),
I do not see how I could announce the entities to HA.
'ICZB-RM11S': [cfg.sensor_battery, cfg.sensor_click, cfg.sensor_action,
cfg.sensor_brightness],

I don't see how I would extract the value of e.g. channel 2 into the value
using a template. However I look at this, I think the best option would be
to create devices per endpoint. Is there another device that is like 4
dimmers in one ?

The whole reason I'm trying to do this (I could bind each group to a Zigbee
light), is that the remote should rather steer 4 lights that are *not* on
Zigbee.

Koenkk, 2020-08-20:
Can you share me the current `devices.js` config that you have?

vdkeybus, 2020-08-20:
> Can you share me the current devices.js config that you have? (for this

This is what I came up with (based it on the ICZB-KDP14S).

...
    {
        zigbeeModel: ['ICZB-RM11S'],
        model: 'ICZB-RM11S',
        vendor: 'iCasa',
        description: 'Zigbee 3.0 Remote Control',
        supports: 'click, action, brightness, scenes',
        fromZigbee: [
            fz.command_recall, fz.legacy_scenes_recall_click,
fz.command_on, fz.legacy_genOnOff_cmdOn, fz.command_off,
            fz.legacy_genOnOff_cmdOff, fz.battery,
fz.legacy_cmd_move_with_onoff, fz.legacy_cmd_stop_with_onoff,
        ],
        toZigbee: [],
    },
...

Koenkk, 2020-08-20:
I would recommend not to use the `legacy` converters, use the converters that start with `command_*`.

vdkeybus, 2020-08-20:
Ok, I now used the non-legacy command_s, but the brightness is no longer
variable. Does this mean that the brightness calculation is done in z2m ?
Somehow I believed that this value was managed in the endpoint (where
latency effects are minimal).

Is it the intention that brightness should now be calculated in the
automations ?

Also still trying to figure out how to deal with the command/endpoint pair.

...
    {
        zigbeeModel: ['ICZB-RM11S'],
        model: 'ICZB-RM11S',
        vendor: 'iCasa',
        description: 'Zigbee 3.0 Remote Control',
        supports: 'click, action, brightness, scenes',
        fromZigbee: [
            fz.command_recall, fz.command_on, fz.command_off,
fz.command_move, fz.command_stop, fz.battery,
        ],
        toZigbee: [],
    },
...

Koenkk, 2020-08-20:
`brightness` was indeed made up by zigbee2mqtt. You should clear your `data/state.json` entry of this device to get rid of it.

Can you provide some logging of the problematic part?

vdkeybus, 2020-08-20:
Hi Koen,

I got rid of brightness by removing and rediscovering the device.

See below for logs (see attached image for button layout). Note that the
endpoint number for the all-on, all-off, S1 and S2 buttons depends on the
last endpoint that transmitted, which is probably a bug in the remote
control. I also did not try to bind any channel with another group yet. As
it looks right now, the groupID will somehow have to be used.

Repeating message (every few minutes, when idle):
Received Zigbee message from '0xccccccfffe26a33a', type
'commandQueryNextImageRequest', cluster 'genOta', data
'{"fieldControl":1,"fileVersion":20,"imageType":1000,"manufacturerCode":4644}'
from endpoint 1 with groupID 0

Ch. 1: 0
short: Received Zigbee message from '0xccccccfffe26a33a', type
'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID
26369
long: Received Zigbee message from '0xccccccfffe26a33a', type
'commandMoveWithOnOff', cluster 'genLevelCtrl', data
'{"movemode":1,"rate":50}' from endpoint 1 with groupID 26369
release: Received Zigbee message from '0xccccccfffe26a33a', type
'commandStopWithOnOff', cluster 'genLevelCtrl', data '{}' from endpoint 1
with groupID 26369

Ch. 1: 1
short: Received Zigbee message from '0xccccccfffe26a33a', type 'commandOn',
cluster 'genOnOff', data '{}' from endpoint 1 with groupID 26369
long: Received Zigbee message from '0xccccccfffe26a33a', type
'commandMoveWithOnOff', cluster 'genLevelCtrl', data
'{"movemode":0,"rate":50}' from endpoint 1 with groupID 26369
release: Received Zigbee message from '0xccccccfffe26a33a', type
'commandStopWithOnOff', cluster 'genLevelCtrl', data '{}' from endpoint 1
with groupID 26369

Ch. 2: 0
short: Received Zigbee message from '0xccccccfffe26a33a', type
'commandOff', cluster 'genOnOff', data '{}' from endpoint 2 with groupID
26370
long: Received Zigbee message from '0xccccccfffe26a33a', type
'commandMoveWithOnOff', cluster 'genLevelCtrl', data
'{"movemode":1,"rate":50}' from endpoint 2 with groupID 26370
release: Received Zigbee message from '0xccccccfffe26a33a', type
'commandStopWithOnOff', cluster 'genLevelCtrl', data '{}' from endpoint 2
with groupID 26370

Ch. 2: 1
short: Received Zigbee message from '0xccccccfffe26a33a', type 'commandOn',
cluster 'genOnOff', data '{}' from endpoint 2 with groupID 26370
long: Received Zigbee message from '0xccccccfffe26a33a', type
'commandMoveWithOnOff', cluster 'genLevelCtrl', data
'{"movemode":0,"rate":50}' from endpoint 2 with groupID 26370
release: Received Zigbee message from '0xccccccfffe26a33a', type
'commandStopWithOnOff', cluster 'genLevelCtrl', data '{}' from endpoint 2
with groupID 26370

Ch. 3: 0
short: Received Zigbee message from '0xccccccfffe26a33a', type
'commandOff', cluster 'genOnOff', data '{}' from endpoint 3 with groupID
26371
long: Received Zigbee message from '0xccccccfffe26a33a', type
'commandMoveWithOnOff', cluster 'genLevelCtrl', data
'{"movemode":1,"rate":50}' from endpoint 3 with groupID 26371
release: Received Zigbee message from '0xccccccfffe26a33a', type
'commandStopWithOnOff', cluster 'genLevelCtrl', data '{}' from endpoint 3
with groupID 26371

Ch. 3: 1
[... 108 lines cut ...]
Received Zigbee message from '0xccccccfffe26a33a', type
'commandMoveWithOnOff', cluster 'genLevelCtrl', data
'{"movemode":1,"rate":50}' from endpoint 3 with groupID 26370
Received Zigbee message from '0xccccccfffe26a33a', type
'commandMoveWithOnOff', cluster 'genLevelCtrl', data
'{"movemode":1,"rate":50}' from endpoint 3 with groupID 26369
Received Zigbee message from '0xccccccfffe26a33a', type
'commandMoveWithOnOff', cluster 'genLevelCtrl', data
'{"movemode":1,"rate":50}' from endpoint 3 with groupID 26371
Received Zigbee message from '0xccccccfffe26a33a', type
'commandMoveWithOnOff', cluster 'genLevelCtrl', data
'{"movemode":1,"rate":50}' from endpoint 3 with groupID 26372
release:
Received Zigbee message from '0xccccccfffe26a33a', type
'commandStopWithOnOff', cluster 'genLevelCtrl', data '{}' from endpoint 3
with groupID 26369
Received Zigbee message from '0xccccccfffe26a33a', type
'commandStopWithOnOff', cluster 'genLevelCtrl', data '{}' from endpoint 3
with groupID 26370
Received Zigbee message from '0xccccccfffe26a33a', type
'commandStopWithOnOff', cluster 'genLevelCtrl', data '{}' from endpoint 3
with groupID 26371
Received Zigbee message from '0xccccccfffe26a33a', type
'commandStopWithOnOff', cluster 'genLevelCtrl', data '{}' from endpoint 3
with groupID 26372

Koenkk, 2020-08-23:
1. I've never used app daemon, but the docs also mention MQTT https://appdaemon.readthedocs.io/en/latest/. I expect it will be possible to respond to an MQTT message (as that will contain both `action` and `action_group` which should be similar as https://www.zigbee2mqtt.io/integration/home_assistant.html#via-mqtt).
2. Why?

vdkeybus, 2020-08-23:
1. So far, all my automations have been agnostic of the transport
mechanism. It is the HA MQTT integration that simply provides me with
events. Maybe I can create an event on the combination of action and action
group in homeassistant.js. I'll try that first.
2. I have no idea how the remote ends up with the group numbers. I'd rather
give the remote group numbers than figuring out which numbers it is using
and updating the apps.yaml. Also allows me to have multiple remotes.

Koenkk, 2020-08-24:
1. I understand, I've tried with the action sensor and it works fine:

[image]

JonOve, 2020-09-18:
Hello,

Thanks a lot for this great effort. I am using this remote for quite some time, but only with touchlink. Works fine, but not all devices report state properly. So sometimes HA is not aligned with the actual state. Will the ICZB-RM11S now become available in Zigbee2Mqtt?

JonOve, 2020-10-19:
Hello,

After all this effort, will the ICZB RM11S become available in Zigbee2MQTT?? I think it will make quite some people happy.

ratenfanger, 2020-10-19:
I have created a pull request in zigbee-herdsman-converters with the configuration of @vdkeybus. When it is merged support will be added in the next zigbee2mqtt release.

If you need it now you can add the configuration to devices.js yourself @JonOve. It is very easy. See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

JonOve, 2020-10-20:
@ratenfager Many thanks for your support, really appreciated. Honestly
speaking I read the instructions on
https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html over and
over again, but I do not seem to be able to properly understand. Maybe
because I run HA on a Synology using the Zigbee2MQTT add on, or maybe I am
just lacking knowledge. It would be great to see some screenprints of what
the process looks like. Thanks again!!

ratenfanger, 2020-11-02:
Hi @JonOve. Support for the iCasa RM11S is included in the latest release of Zigbee2MQTT (1.16.0). So you don't have to mess with editing the zigbee herdsman converters yourself anymore :)

As the remote is now supported this issue may be closed.

JonOve, 2020-11-03:
Yes and it works great! Thanks a lot for your support!

closed 2020-11-03 by Koenkk