Koenkk/zigbee2mqtt #4131: Device not working anymore because of wrong endpoint

TilmanK, 2020-08-19:
I upgraded from 1.14.1 to 1.14.3. This broke my living room lamp. I switched between both git tags and noticed that this is reproducible. The reason can be seen when sniffing traffic.

When sending a {state: "on"} to the device, 1.14.1 sends it to the correct endpoint:

ZigBee Application Support Layer Data, Dst Endpt: 2, Src Endpt: 1
    Frame Control Field: Data (0x00)
    Destination Endpoint: 2
    Cluster: On/Off (0x0006)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 213

Doing the same with 1.14.3 uses 1 as destination endpoint which is wrong: 

ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x00)
    Destination Endpoint: 1
    Cluster: On/Off (0x0006)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 25

The root cause lies in this bug: https://github.com/Koenkk/zigbee2mqtt/issues/4000
With this change https://github.com/Koenkk/zigbee2mqtt/issues/4000

The device is not the same. You just captured my device and changed the identifier by removing the space. :)

TilmanK, 2020-08-19:
I'll submit a PR for this...

Koenkk, 2020-08-19:
Merged, sorry for breaking! Assuming this can be closed now.

closed 2020-08-19 by Koenkk