Koenkk/zigbee2mqtt #4134: GLEDOPTO is reporting "Unsupported device" - Support was dropped after update of container

exetico, 2020-08-19:
I've used my GLEDOPTO RGBCCT for long time - with no issues, what so ever.

Today i updated my zigbee2mqtt container to the latest version, and it's sadly reporting that my LED driver is unsupported? (I removed the device, and started a new paring, just to be 100% sure...)

```
Zigbee2MQTT:info  2020-08-19 19:39:27: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2020-08-19 19:39:27: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.14.3","commit":"f8066e8","coordinator":{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20200721}},"network":{"panID":6755,"extendedPanID":"0xdddddddddddddddd","channel":15},"log_level":"info","permit_join":true}'
Zigbee2MQTT:info  2020-08-19 19:40:57: Device '0x00124b0019c796e8' joined
Zigbee2MQTT:info  2020-08-19 19:40:57: Starting interview of '0x00124b0019c796e8'
Zigbee2MQTT:info  2020-08-19 19:40:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00124b0019c796e8"}}'
Zigbee2MQTT:info  2020-08-19 19:40:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00124b0019c796e8"}}'
Zigbee2MQTT:info  2020-08-19 19:40:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00124b0019c796e8"}}'
Zigbee2MQTT:info  2020-08-19 19:40:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00124b0019c796e8"}}'
Zigbee2MQTT:info  2020-08-19 19:40:58: Successfully interviewed '0x00124b0019c796e8', device has successfully been paired
Zigbee2MQTT:warn  2020-08-19 19:40:58: Device '0x00124b0019c796e8' with Zigbee model 'GLEDOPTO' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Zigbee2MQTT:info  2020-08-19 19:40:58: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x00124b0019c796e8","supported":false}}'
Zigbee2MQTT:info  2020-08-19 19:41:07: MQTT publish: topic 'zigbee2mqtt/aqara1', payload '{"battery":100,"voltage":3145,"linkquality":90}'

```
It was working just fine, added through Home Assistant auto-discovery and so, with no issues.

Now I can't use it. I've spotted a few issues where GLEDOPTO is mentioned, but is it anything related to the latest changes to fix #3987, and maybe other, too?

Koenkk, 2020-08-19:
Should be fixed.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

exetico, 2020-08-20:
Great, thanks!
[image]

Confirmed OK on latest-dev.

exetico, 2020-10-06:
Hi @Koenkk 

I just updated my controllers firmware, and updated to the latest stable branch, too. Not I can't control my LED Controller again.

All other devices are working like they should.

Koenkk, 2020-10-07:
@exetico what error do you get?

exetico, 2020-10-07:
Hi, 

Yesterday I had some issues, so I couldn't provide the parring logs. Here's a fresh example:
```
info  2020-10-07 15:51:04: Device '0x00124b0019c796e8' joined
info  2020-10-07 15:51:04: Starting interview of '0x00124b0019c796e8'
info  2020-10-07 15:51:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00124b0019c796e8"}}'
info  2020-10-07 15:51:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00124b0019c796e8"}}'
info  2020-10-07 15:51:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00124b0019c796e8"}}'
info  2020-10-07 15:51:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00124b0019c796e8"}}'
info  2020-10-07 15:51:06: Successfully interviewed '0x00124b0019c796e8', device has successfully been paired
warn  2020-10-07 15:51:06: Device '0x00124b0019c796e8' with Zigbee model 'GLEDOPTO' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
info  2020-10-07 15:51:06: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x00124b0019c796e8","supported":false}}'

```

As you can see, it's reporting that GLEDOPTO is not supported again :-( As mentioned, i switched from dev to the latest stable.

Shouldn't it be supported in the stable again, or? I've tested in the `latest-dev` branch again, and there it's working just fine :-) Maybe changes from `Koenkk/zigbee-herdsman-converters/` arn't pushed to the `latest` branch, or? :-)

Koenkk, 2020-10-07:
Can you provide the `data/database.db` entry again of this device? Maybe something changed there. The entry of your OP should work both in latest-dev and latest. Also make sure you don't have any local changes in `devices.js` (I assume you don't)

exetico, 2020-10-07:
Sure, here's the dump of the device, after I've paired it again (on the `latest-dev` branch).
```
{
    "id": 31,
    "type": "Router",
    "ieeeAddr": "0x00124b0019c796e8",
    "nwkAddr": 35335,
    "manufId": 0,
    "manufName": "GLEDOPTO",
    "powerSource": "Mains (single phase)",
    "modelId": "GLEDOPTO",
    "epList": [11, 13],
    "endpoints": {
        "11": {
            "profId": 49246,
            "epId": 11,
            "devId": 528,
            "inClusterList": [0, 3, 4, 5, 6, 8, 768],
            "outClusterList": [],
            "clusters": {
                "genBasic": {
                    "attributes": {
                        "modelId": "GLEDOPTO",
                        "manufacturerName": "GLEDOPTO",
                        "powerSource": 1,
                        "zclVersion": 1,
                        "appVersion": 1,
                        "stackVersion": 2,
                        "hwVersion": 1,
                        "dateCode": "20131206",
                        "swBuildId": "1.0.2"
                    }
                }
            },
            "binds": []
        },
        "13": {
            "profId": 49246,
            "epId": 13,
            "devId": 57694,
            "inClusterList": [4096],
            "outClusterList": [4096],
            "clusters": {},
            "binds": []
        }
    },
    "appVersion": 1,
    "stackVersion": 2,
    "hwVersion": 1,
    "dateCode": "20131206",
    "swBuildId": "1.0.2",
    "zclVersion": 1,
    "interviewCompleted": true,
    "meta": {},
    "lastSeen": 1602085866875
}
```
I've not changed a thing in `devices` and it's not part of my machine, if i take a look at the volume I've allocated the container:
```
~/z/a/data> ls -alh
total 76K
drwxr-xr-x 4 root root 4.0K Oct  7 17:56 ./
drwxr-xr-x 3 root root 4.0K Jul 31 11:46 ../
-rw-r--r-- 1 root root 2.0K Oct  7 17:58 configuration.yaml
-rw-r--r-- 1 root root 4.6K Oct  7 17:56 coordinator_backup.json
-rw-r--r-- 1 root root  27K Oct  7 17:56 database.db
drwxr-xr-x 2 root root  16K Oct  7 17:56 log/
drwxr-xr-x 2 root root 4.0K Jul 31 23:57 old_log/
-rw-r--r-- 1 root root  101 Jul 31 11:47 secret.yaml
-rw-r--r-- 1 root root 3.0K Oct  7 18:36 state.json
```

Here's the compose:
```
  zigbee2mqtt:
    image: koenkk/zigbee2mqtt:latest-dev
    container_name: zigbee2mqtt
    labels:
      com.centurylinklabs.watchtower.enable: "false"
    volumes:
      - /home/${usernamegoeshere}/zigbee2mqtt/app/data:/app/data
      - /etc/localtime:/etc/localtime:ro
    restart: unless-stopped
    devices:
      - /dev/cc2652r:/dev/cc2652r
```

_I don't think it's relevant, cause everything else works as expected :-)_

exetico, 2020-10-07:
Too damn much work in the last couple of days... I forgot to pull the latest stable image, cause (as you can see), watchtower arn't pulling the latest version, too.

I'm sorry for wasting your time :-)

closed 2020-10-07 by exetico

Koenkk, 2020-10-07:
no problem, good to hear!

xdmore70, 2024-10-08:
I have the same problem, Gledopto device works fine but the frontend report unsopported 

[image]
Frontend release is:
[image]

I find this old post  i hope some can help me

EIBHomeControl, 2024-11-13:
> I have the same problem, Gledopto device works fine but the frontend report unsopported
> 
> ![light](https://private-user-images.githubusercontent.com/184242746/374569090-13962b83-0a8b-465e-ab47-a2dcc9904de8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE0OTUyMTUsIm5iZiI6MTczMTQ5NDkxNSwicGF0aCI6Ii8xODQyNDI3NDYvMzc0NTY5MDkwLTEzOTYyYjgzLTBhOGItNDY1ZS1hYjQ3LWEyZGNjOTkwNGRlOC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTEzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTExM1QxMDQ4MzVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04MWFiMzVlYmZhODcwNThkNzlkZWQ5MDRlMTYyNmU5NGIyMTIyOWFjYjYxNTA3ZjRiNWFjYzRjZGJiYWY0YzA5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.TJjkNXNNnjiimSUQMycpAithAKSkpUDG9_DITO0TnkU) Frontend release is: ![Screenshot 2024-10-08 150559](https://private-user-images.githubusercontent.com/184242746/374569311-4d498023-17ed-4364-8952-f55f59607f6a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE0OTUyMTUsIm5iZiI6MTczMTQ5NDkxNSwicGF0aCI6Ii8xODQyNDI3NDYvMzc0NTY5MzExLTRkNDk4MDIzLTE3ZWQtNDM2NC04OTUyLWY1NWY1OTYwN2Y2YS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTEzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTExM1QxMDQ4MzVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zNjM4MmY5Zjc4MDBhN2NmNjNhNGZhYTUzMzMxM2RjNmQ0NDhjMjczZWQwYjQ5NDhhYzg1YjE2OTI1OGY3MmE2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.NlncT6mATAtB2CyW4crZnqzz3fIUSbcQRha-vyDG9Gg)
> 
> I find this old post i hope some can help me

me too