Koenkk/zigbee2mqtt #4135: No network route error when sending to group

algirdasc, 2020-08-19:
# Bug Report
## What happened
After changing coordinator to slaesh’s CC2652RB stick I'm unable to control lights by its group. I'm getting `Data request failed with error: 'No network route' (205))'` error.

## What did you expect to happen
I expect to control all the lights in particular group.

## How to reproduce it (minimal and precise)
Buy few IKEA zigbee lamps, add them to group, try sending command to that group. In my case, here is the log:
```
info  2020-08-19 23:11:28: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"groups","message":[{"optimistic":true,"devices":["0x000b57fffebaf77c","0x000b57fffebb9bf9","0x000b57fffee9abcb"],"friendly_name":"livingroom_lamps","ID":1},{"optimistic":true,"devices":["0x000d6ffffe2ccb06/1","0x086bd7fffe2e7b4d/1"],"friendly_name":"kitchen_lamps","ID":3}]}'
info  2020-08-19 23:11:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"devices","message":[{"ieeeAddr":"0x00124b0021a8ec69","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack3x0","dateCode":"20200715","lastSeen":1597867892406},{"ieeeAddr":"0x086bd7fffe915a5a","type":"EndDevice","networkAddress":55511,"model":"2AJZ4KPFT","vendor":"Konke","description":"Temperature and humidity sensor","friendly_name":"bedroom_th_sensor","manufacturerID":4098,"manufacturerName":"Konke","powerSource":"Unknown","modelID":"3AFE140103020000","hardwareVersion":1,"dateCode":"","lastSeen":1597866233426},{"ieeeAddr":"0x00158d000423afc6","type":"EndDevice","networkAddress":39657,"model":"WSDCGQ01LM","vendor":"Xiaomi","description":"MiJia temperature & humidity sensor","friendly_name":"boilerroom_th_sensor","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","modelID":"lumi.sensor_ht","hardwareVersion":30,"softwareBuildID":"3000-0001","dateCode":"20160516","lastSeen":1597867298759},{"ieeeAddr":"0x00158d00041dfc73","type":"EndDevice","networkAddress":33590,"model":"WSDCGQ01LM","vendor":"Xiaomi","description":"MiJia temperature & humidity sensor","friendly_name":"hallway_floor_th_sensor","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","modelID":"lumi.sensor_ht","hardwareVersion":30,"softwareBuildID":"3000-0001","dateCode":"20160516","lastSeen":1597865303281},{"ieeeAddr":"0x00158d0003cc4ae7","type":"EndDevice","networkAddress":28814,"model":"WSDCGQ01LM","vendor":"Xiaomi","description":"MiJia temperature & humidity sensor","friendly_name":"bathroom_heater_th_sensor","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","modelID":"lumi.sensor_ht","hardwareVersion":30,"softwareBuildID":"3000-0001","dateCode":"20160516","lastSeen":1597867848146},{"ieeeAddr":"0x086bd7fffe915fba","type":"EndDevice","networkAddress":5046,"model":"2AJZ4KPFT","vendor":"Konke","description":"Temperature and humidity sensor","friendly_name":"bathroom_floor_th_sensor","manufacturerID":4712,"manufacturerName":"Konke","powerSource":"Unknown","modelID":"3AFE220103020000","hardwareVersion":1,"dateCode":"","lastSeen":1597866330553},{"ieeeAddr":"0x086bd7fffe2e7b4d","type":"Router","networkAddress":17052,"model":"LED1733G7","vendor":"IKEA","description":"TRADFRI LED bulb E14 600 lumen, dimmable, white spectrum, opal white","friendly_name":"kitchen2_lamp","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelID":"TRADFRI bulb E14 WS opal 600lm","hardwareVersion":1,"softwareBuildID":"2.0.029","dateCode":"20200312","lastSeen":1597602604930},{"ieeeAddr":"0x000d6ffffe2ccb06","type":"Router","networkAddress":3979,"model":"LED1733G7","vendor":"IKEA","description":"TRADFRI LED bulb E14 600 lumen, dimmable, white spectrum, opal white","friendly_name":"kitchen1_lamp","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelID":"TRADFRI bulb E14 WS opal 600lm","hardwareVersion":1,"softwareBuildID":"2.0.029","dateCode":"20200312","lastSeen":1597602613892},{"ieeeAddr":"0x000b57fffebb9bf9","type":"Router","networkAddress":27051,"model":"LED1545G12","vendor":"IKEA","description":"TRADFRI LED bulb E26/E27 980 lumen, dimmable, white spectrum, opal white","friendly_name":"livingroom2_lamp","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelID":"TRADFRI bulb E27 WS opal 980lm","hardwareVersion":1,"softwareBuildID":"2.3.050","dateCode":"20200311","lastSeen":1597867877915},{"ieeeAddr":"0x000b57fffebaf77c","type":"Router","networkAddress":64385,"model":"LED1545G12","vendor":"IKEA","description":"TRADFRI LED bulb E26/E27 980 lumen, dimmable, white spectrum, opal white","friendly_name":"livingroom1_lamp","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelID":"TRADFRI bulb E27 WS opal 980lm","hardwareVersion":1,"softwareBuildID":"2.3.050","dateCode":"20200311","lastSeen":1597867886025},{"ieeeAddr":"0x000b57fffee9abcb","type":"Router","networkAddress":62488,"model":"LED1545G12","vendor":"IKEA","description":"TRADFRI LED bulb E26/E27 980 lumen, dimmable, white spectrum, opal white","friendly_name":"livingroom3_lamp","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelID":"TRADFRI bulb E27 WS opal 980lm","hardwareVersion":1,"softwareBuildID":"2.3.050","dateCode":"20200311","lastSeen":1597867881920},{"ieeeAddr":"0x086bd7fffe5304a8","type":"Router","networkAddress":31440,"model":"LED1537R6/LED1739R5","vendor":"IKEA","description":"TRADFRI LED bulb GU10 400 lumen, dimmable, white spectrum","friendly_name":"bedroom_lamp","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelID":"TRADFRI bulb GU10 WS 400lm","hardwareVersion":1,"softwareBuildID":"2.0.029","dateCode":"20200312","lastSeen":1597867863996},{"ieeeAddr":"0x14b457fffe479356","type":"Router","networkAddress":5102,"model":"LED1842G3","vendor":"IKEA","description":"TRADFRI LED bulb E27 WW clear 250 lumen, dimmable","friendly_name":"tv_lamp","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelID":"TRADFRI bulb E27 WW clear 250lm","hardwareVersion":2,"softwareBuildID":"2.3.050","dateCode":"20181203","lastSeen":1597079322718},{"ieeeAddr":"0xccccccfffe6b4faf","type":"EndDevice","networkAddress":18726,"model":"E1525/E1745","vendor":"IKEA","description":"TRADFRI motion sensor","friendly_name":"bedroom_motion_sensor","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Battery","modelID":"TRADFRI motion sensor","hardwareVersion":1,"softwareBuildID":"2.0.022","dateCode":"20190308","lastSeen":1597867826448},{"ieeeAddr":"0x680ae2fffe378d2f","type":"EndDevice","networkAddress":61227,"model":"E1525/E1745","vendor":"IKEA","description":"TRADFRI motion sensor","friendly_name":"kidsroom_motion_sensor","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Battery","modelID":"TRADFRI motion sensor","hardwareVersion":1,"softwareBuildID":"2.0.022","dateCode":"20190308","lastSeen":1597864780743},{"ieeeAddr":"0x680ae2fffe34a484","type":"EndDevice","networkAddress":21652,"model":"E1525/E1745","vendor":"IKEA","description":"TRADFRI motion sensor","friendly_name":"bathroom_motion_sensor","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Battery","modelID":"TRADFRI motion sensor","hardwareVersion":1,"softwareBuildID":"2.0.022","dateCode":"20190308","lastSeen":1597867787735},{"ieeeAddr":"0x000b57fffec317f4","type":"Router","networkAddress":3947,"model":"LED1623G12","vendor":"IKEA","description":"TRADFRI LED bulb E27 1000 lumen, dimmable, opal white","friendly_name":"fishingpole_lamp","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelID":"TRADFRI bulb E27 W opal 1000lm","hardwareVersion":1,"softwareBuildID":"1.2.214","dateCode":"20170302","lastSeen":1597080007606},{"ieeeAddr":"0xccccccfffe29515e","type":"EndDevice","networkAddress":12592,"model":"E1743","vendor":"IKEA","description":"TRADFRI ON/OFF switch","friendly_name":"bedroom_lamp_switch","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Battery","modelID":"TRADFRI on/off switch","hardwareVersion":1,"softwareBuildID":"2.2.010","dateCode":"20190723","lastSeen":1597081883967},{"ieeeAddr":"0x00158d0004288e0f","type":"Router","networkAddress":17665,"model":"LLKZMK11LM","vendor":"Xiaomi","description":"Aqara wireless relay controller","friendly_name":"bathroom_wa [... cut ...]
info  2020-08-19 23:13:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'livingroom_lamps' failed: 'Error: Command 1 genOnOff.off({}) failed (Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"livingroom_lamps"}}'
```

## Debug Info
Zigbee2MQTT version: 1.14.3 (commit #1213fd6)
Adapter hardware: slaesh’s CC2652RB stick
Adapter firmware version: `{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20200715}}`

Koenkk, 2020-08-19:
- Can you try the latest firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin
- If that doens't work please provide the herdsman debug logging of this issue

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

algirdasc, 2020-08-19:
Thanks for quick response. 

Updated latest firmware: `{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20200805}}`

But that did not help. Here are the debug log:

```
Zigbee2MQTT:debug 2020-08-19 23:33:49: Received MQTT message on 'zigbee2mqtt/livingroom_lamps/set' with data '{"state": "OFF"}'
Zigbee2MQTT:debug 2020-08-19 23:33:49: Publishing 'set' 'state' to 'livingroom_lamps'
  zigbee-herdsman:controller:group Command 1 genOnOff.off({}) +37s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":1,"dstaddr":"0x0000000000000001","destendpoint":255,"dstpanid":0,"srcendpoint":1,"clusterid":6,"transid":40,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[17,24,0]}} +5s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,23,36,2,1,1,0,0,0,0,0,0,0,255,0,0,1,6,0,40,0,30,3,0,17,24,0,245] +5s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,0,103] +5s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,0,103] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 2 - [0] - 103 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequestExt - {"status":0} +5s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,205,1,40,35] +784ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,205,1,40,35] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [205,1,40] - 35 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":205,"endpoint":1,"transid":40} +6s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:controller:group Command 1 genOnOff.off({}) failed (Data request failed with error: 'No network route' (205)) +37s
Zigbee2MQTT:error 2020-08-19 23:33:50: Publish 'set' 'state' to 'livingroom_lamps' failed: 'Error: Command 1 genOnOff.off({}) failed (Data request failed with error: 'No network route' (205))'
Zigbee2MQTT:debug 2020-08-19 23:33:50: Error: Command 1 genOnOff.off({}) failed (Data request failed with error: 'No network route' (205))
    at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:694:31)
    at Generator.next (<anonymous>)
    at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:24:58)
Zigbee2MQTT:info  2020-08-19 23:33:50: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'livingroom_lamps' failed: 'Error: Command 1 genOnOff.off({}) failed (Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"livingroom_lamps"}}'
```

Koenkk, 2020-08-20:
Can you try changing the group ID from `1` to `2` (just change it in `configuration.yaml`)

algirdasc, 2020-08-20:
I've tried setting ID to `2`, to `20`, also tried creating separate group for Aqara/Xiaomi wall switch to make sure that the problem is not in IKEA lamps, but the result is always the same - No network route. Devices can be controlled individually just fine, but not when trying to control whole group. Is there something more I could test?

slaesh, 2020-08-26:
I am not quite familiar with zigbee groups at all, but I can't imagine that it's a HW problem.. Its just the CPU connected to an antenna :)
we may need to check the coordinator-firmware itself.. @Koenkk can you reproduce it with any other devices in a group?
maybe it's something device-specific..

i am the next two weeks in home-office, we will move to another flat and need to paint all the walls etc.. maybe I during the next days i can grep some devices from my office, but not really sure.
will keep you updated if i found something new! =)

Koenkk, 2020-08-27:
I've tried to reproduce the issue:
- Clean zigbee2mqtt 1.14.3 (empty `data` directory except `configuration.yaml`), clean flashed `CC2652RB_20200805.zip` firmware

```yaml
homeassistant: false
permit_join: true
serial:
  port: /dev/tty.SLAB_USBtoUART
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
groups:
  '1':
    friendly_name: group_1
```
- Started zigbee2mqtt and toggled group 1 through `zigbee2mqtt/group_1/set` `{"state": "TOGGLE"}` -> works
- Next I joined a bulb and added this to the group, also did a toggle and the bulb responds
- Note that it doesn't matter which and if there are any devices in the group since groups commands are broadcasted over the whole network (the device itself is responsible for checking if its in a group: https://www.zigbee2mqtt.io/information/groups.html#how-do-groups-work). The coordinator has no knowledge which devices are in a group.

slaesh, 2020-08-28:
Thanks for that! 

Let's dive into it.. I want to understand the "error" ..
What does the message "Data request failed with error: 'No network route' (205)" mean?
Does it occur on any falsy response?
Or may is it a faulty configuration (device config or coordinator config)?

Koenkk, 2020-08-28:
It means that the coordinator is not able at all to transmit the message on the network. But I believe that in this case a reset may fix it (like I described in https://github.com/Koenkk/zigbee2mqtt/issues/4135#issuecomment-682092658)

algirdasc, 2020-09-01:
Should I try to reset USB dongle somehow?

Koenkk, 2020-09-01:
@algirdasc yes, I recommend following: https://github.com/Koenkk/zigbee2mqtt/issues/4135#issuecomment-682092658

slaesh, 2020-09-02:
@algirdasc some good news? :)

algirdasc, 2020-09-02:
Not yet, I'm going to try it this week.

closed 2020-10-11 by github-actions[bot]

sabo, 2020-11-18:
Sorry to reopen a stale issue, but I'm also running into this exact problem, and have some more information that might be useful. For me, the issue does seem to be device-specific.

- Sending messages to a group consisting of only Hue lightbulbs works as expected.
- Sending messages to a group of Tradfri bulbs or Sengled bulbs returns the 'no network route (205)' error.
- Sending messages to a group with at least one Hue bulb works as expected, for all devices in that group. That is, if there is at least one Hue bulb in a group with Tradfri or Sengled bulbs, every bulb in that group responds to messages as expected.

I'm using slaesh's stick with a fresh flash of the 20201026 coordinator firmware and version 1.16.1 of zigbee2mqtt in a docker container. Changing the group IDs and re-flashing the stick did not change anything.

Config file:
 ```yaml
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://10.0.1.1'
  include_device_information: true
serial:
  port: /dev/zigbee
advanced:
  log_level: debug
  report: true
  channel: 20
  pan_id: 7011
  availability_timeout: 60
  availability_blacklist:
    - bedroom_button
    - backdoor_button
    - kitchen_button
devices:
  '0xb0ce18140302bd97':
    friendly_name: kitchen_ceil_1
  '0xb0ce18140302c11e':
    friendly_name: kitchen_ceil_2
  '0x000b57fffe8ae5f0':
    friendly_name: livingroom_ceil_1
  '0x000b57fffe8af193':
    friendly_name: livingroom_ceil_2
  '0x000b57fffe8ff7f9':
    friendly_name: livingroom_dimmer
  '0x0017880103b6e669':
    friendly_name: bedroom_ceil_1
  '0x00178801034c7bdc':
    friendly_name: bedroom_ceil_3
  '0x00178801034cab17':
    friendly_name: bedroom_ceil_2
  '0x000b57fffed73ff8':
    friendly_name: office_lamp_top
  '0x000b57fffecc19d4':
    friendly_name: office_lamp_bottom
  '0xb0ce18140302b88b':
    friendly_name: mudroom_1
  '0x286d970001042899':
    friendly_name: bedroom_button
  '0x286d9700010427e1':
    friendly_name: backdoor_button
  '0x286d970001085a24':
    friendly_name: kitchen_button
  '0xd0cf5efffe22d1da':
    friendly_name: bedroom_nightstand
  '0x000b57fffe9d0a45':
    friendly_name: '0x000b57fffe9d0a45'
  '0x000b57fffe9a5ad1':
    friendly_name: '0x000b57fffe9a5ad1'
  '0x000b57fffe4aaa06':
    friendly_name: '0x000b57fffe4aaa06'
  '0x0017880103b6e023':
    friendly_name: office_ceiling_light
groups:
  '20':
    friendly_name: all_hue
[... 122 lines cut ...]
Zigbee2MQTT:info  2020-11-18 07:51:48: MQTT publish: topic 'zigbee2mqtt/mixed_hue_tradfri', payload '{"brightness":254,"state":"ON"}'
Zigbee2MQTT:info  2020-11-18 07:51:48: MQTT publish: topic 'zigbee2mqtt/livingroom_ceil_2', payload '{"brightness":254,"device":{"applicationVersion":17,"dateCode":"20170302","friendlyName":"livingroom_ceil_2","hardwareVersion":1,"ieeeAddr":"0x000b57fffe8af193","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"LED1622G12","networkAddress":14522,"powerSource":"Mains (single phase)","softwareBuildID":"1.2.214","stackVersion":87,"type":"Router","zclVersion":1},"linkquality":99,"state":"ON","update_available":false}'
```

- Mixed group of Hue and Sengled bulbs
```
Zigbee2MQTT:debug 2020-11-18 08:05:37: Received MQTT message on 'zigbee2mqtt/mixed_hue_sengled/set' with data '{"state": "OFF"}'
Zigbee2MQTT:debug 2020-11-18 08:05:37: Publishing 'set' 'state' to 'mixed_hue_sengled'
2020-11-18T13:05:37.083Z zigbee-herdsman:controller:group Command 70 genOnOff.off({})
2020-11-18T13:05:37.083Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":1,"dstaddr":"0x0000000000000046","destendpoint":255,"dstpanid":0,"srcendpoint":1,"clusterid":6,"transid":224,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[17,201,0]}}
2020-11-18T13:05:37.083Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,23,36,2,1,70,0,0,0,0,0,0,0,255,0,0,1,6,0,224,0,30,3,0,17,201,0,171]
2020-11-18T13:05:37.091Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,0,103]
2020-11-18T13:05:37.091Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,0,103]
2020-11-18T13:05:37.091Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 2 - [0] - 103
2020-11-18T13:05:37.091Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequestExt - {"status":0}
2020-11-18T13:05:37.092Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-11-18T13:05:37.096Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,224,38]
2020-11-18T13:05:37.096Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,224,38]
2020-11-18T13:05:37.096Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,224] - 38
2020-11-18T13:05:37.096Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":224}
2020-11-18T13:05:37.096Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:info  2020-11-18 08:05:37: MQTT publish: topic 'zigbee2mqtt/bedroom_ceil_1', payload '{"brightness":117,"device":{"applicationVersion":2,"dateCode":"20170311","friendlyName":"bedroom_ceil_1","hardwareVersion":1,"ieeeAddr":"0x0017880103b6e669","manufacturerID":4107,"manufacturerName":"Philips","model":"9290011370","networkAddress":35729,"powerSource":"Mains (single phase)","softwareBuildID":"1.23.0_r20156","stackVersion":1,"type":"Router","zclVersion":1},"linkquality":69,"state":"OFF","update_available":false}'
[... cut ...] tness":117,"state":"OFF"}'
Zigbee2MQTT:info  2020-11-18 08:05:37: MQTT publish: topic 'zigbee2mqtt/mudroom_1', payload '{"brightness":117,"color_temp":370,"device":{"applicationVersion":1,"friendlyName":"mudroom_1","hardwareVersion":1,"ieeeAddr":"0xb0ce18140302b88b","manufacturerID":4448,"manufacturerName":"sengled","model":"Z01-A19NAE26","networkAddress":3298,"powerSource":"Unknown","stackVersion":0,"type":"EndDevice","zclVersion":1},"linkquality":66,"state":"OFF","update_available":false}'
```

tim-brand, 2020-11-23:
I'm having the same issue as well. My groups are not working at all. I only have Hue bulbs. So it doesn't seem related to the brand.

My problem started after working on a solution for #4473 . I moved the stick a couple of times and have setup a new network with it once, just to test (not running anymore).
At some point I had issues starting my original config again. It kept saying that I had to select another pan_id. Haven't done that, just did the trick with removing the antenna, and blocking the signal. Some retries later it started my original config again.
But somewhere from this point the groups aren't working anymore. I also noticed that there were many errors during a network scan.  Many of my routers (Hue bulbs) were reporting LQI and routing table issues. I got them solved by pulling and reconnecting the power for most of them. But even after solving them all, the groups still don't work.
I have removed all group configurations and rejoined 2 bulbs and created a new group, this still doesn't work.

It has worked perfectly before, I'm out of ideas now.

tim-brand, 2020-11-23:
I just created a complete new setup. A new config file in an empty data directory,  with different network settings than my existing network.
Then I added a single hue bulb. Created a new group and added that bulp to it. When sending the mqtt event, I still get the `'No network route' (205)`.
I'm starting to think this is might be a hardware issue. As I said, it started after I have been testing with the stick in different situations where I plugged it in and out a couple of times. (cover is removed to be able to access the reset button).

@slaesh Do you think it's even possible this might be a hardware issue?

The log from just before the device joined, untill the error:
```
 zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,194,49,195,143,5,9,1,136,23,0,0,0,174] +2m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,194,49,195,143,5,9,1,136,23,0,0,0,174] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [194,49,195,143,5,9,1,136,23,0,0,0] - 174 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":12738,"extaddr":"0x0017880109058fc3","parentaddr":0} +2m
  zigbee-herdsman:controller:log Device '0x0017880109058fc3' joined +2m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
  zigbee-herdsman:controller:log Device '0x0017880109058fc3' accepted by handler +1ms
  zigbee-herdsman:controller:log New device '0x0017880109058fc3' joined +0ms
  zigbee-herdsman:controller:log Creating device '0x0017880109058fc3' +0ms
Zigbee2MQTT:info  2020-11-23 21:16:18: Device '0x0017880109058fc3' joined
  zigbee-herdsman:controller:log Interview '0x0017880109058fc3' start +13ms
Zigbee2MQTT:info  2020-11-23 21:16:18: Starting interview of '0x0017880109058fc3'
  zigbee-herdsman:controller:device:log Interview - start device '0x0017880109058fc3' +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":12738,"nwkaddrofinterest":12738} +2m
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,194,49,194,49,35] +2m
Zigbee2MQTT:info  2020-11-23 21:16:18: MQTT publish: topic 'zigbee2mqtt-timdev2/bridge/event', payload '{"data":{"friendly_name":"0x0017880109058fc3","ieee_address":"0x0017880109058fc3"},"type":"device_joined"}'
Zigbee2MQTT:info  2020-11-23 21:16:18: MQTT publish: topic 'zigbee2mqtt-timdev2/bridge/event', payload '{"data":{"friendly_name":"0x0017880109058fc3","ieee_address":"0x0017880109058fc3","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2020-11-23 21:16:18: MQTT publish: topic 'zigbee2mqtt-timdev2/bridge/log', payload '{"message":{"friendly_name":"0x0017880109058fc3"},"type":"device_connected"}'
Zigbee2MQTT:info  2020-11-23 21:16:18: MQTT publish: topic 'zigbee2mqtt-timdev2/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x0017880109058fc3"},"type":"pairing"}'
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,13,69,193,194,49,194,49,195,143,5,9,1,136,23,0,142,217] +21ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,193,194,49,194,49,195,143,5,9,1,136,23,0,142,217] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 13 - 2 - 5 - 193 - [194,49,194,49,195,143,5,9,1,136,23,0,142] - 217 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - endDeviceAnnceInd - {"srcaddr":12738,"nwkaddr":12738,"ieeeaddr":"0x0017880109058fc3","capabilities":142} +23ms
  zigbee-herdsman:controller:log Device announce '0x0017880109058fc3' +9ms
Zigbee2MQTT:debug 2020-11-23 21:16:18: Device '0x0017880109058fc3' announced itself
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
Zigbee2MQTT:info  2020-11-23 21:16:18: MQTT publish: topic 'zigbee2mqtt-timdev2/bridge/event', payload '{"data":{"friendly_name":"0x0017880109058fc3","ieee_address":"0x0017880109058fc3"},"type":"device_announce"}'
Zigbee2MQTT:info  2020-11-23 21:16:18: MQTT publish: topic 'zigbee2mqtt-timdev2/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x0017880109058fc3"},"type":"device_announced"}'
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102] +3ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +2m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,194,49,0,113] +7ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,194,49,0,113] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [194,49,0] - 113 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":12738,"relaycount":0,"relaylist":[]} +11ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,194,49,0,113] +5ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,194,49,0,113] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [194,49,0] - 113 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":12738,"relaycount":0,"relaylist":[]} +6ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,18,69,130,194,49,0,194,49,1,64,142,11,16,82,128,0,0,44,128,0,0,127] +47ms
[... cut ...]
    at Generator.next (<anonymous>)
    at fulfilled (/Users/tim/projects/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:24:58)
Zigbee2MQTT:info  2020-11-23 21:17:04: MQTT publish: topic 'zigbee2mqtt-timdev2/bridge/log', payload '{"message":"Publish 'set' 'state' to 'eettafel' failed: 'Error: Command 1 genOnOff.toggle({}) failed (Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"eettafel"},"type":"zigbee_publish_error"}'
```

slaesh, 2020-11-23:
I can't imagine that, if it's working normally without the use of groups..? 🤔
The hardware doesn't care about groups or not.. It doesn't even know about that kinda info.. Just transmitting and receiving bytes..
Maybe koenkk can shed some light into this dark topic.. 😇😊

tim-brand, 2020-11-23:
Thanks for your reply! 
That's what I'm thinking too. I can manage the single bulp just fine.
However. I was was running your debug version of the firmware in my above test. I just updated it to the latest version of koenkk. Then I also got the same errors on controlling the single bulp. After some reboots of zigbee2mqtt, it was all working fine again. Except the groups.

tim-brand, 2020-11-25:
I've been playing with this issue for hours. Created a complete new environment with different pan-id, channel and network-key. I connected one bulp and added it to a new group. Still the same issue. Did many restarts and rejoins, but still the error remains. I had just one coordinator, one Hue bulp and one group, and still the error when controlling the group was the "no network route". I noticed the errorcode `205`, so I did some searching in the code and thought that this is a zigbee protocol code. I think I found the related code in [this document](https://www.nxp.com/docs/en/user-guide/JN-UG-3113.pdf). It's described as `ZPS_NWK_ENUM_NO_KEY` -> `0xCD` which is the 205 code -> `Security processing has failed on outgoing frame due to no key`. So it seems that it's missing a key there.

Finally I went back to my original setup and the network would let me boot. It kept saying that I had to choose a different pan_id. Then I did the trick with disconnecting the antenna and covering the usb-stick with a cup wrapped with tin foil, which helped starting it again. And surprisingly after that my groups in my original network were working again!

So, looking back on the issue. It started after I did some testing with a new network setup and went back to my original network, then the groups didn't work anymore. Some days later I started testing again this groups issue by creating a new network again. And after I moved back to my original network again, the groups are working again.

To me it seems like there's some kind of memory on the stick which is keeping some network (key) information. Also looking at the fact that at both times I had issues going back to my original network.

@Koenkk Is this some useful information? And can I do some more testing to get some information?
@slaesh I ordered a second stick as I don't want to abuse my live network everytime ;)

Koenkk, 2020-11-26:
I don't think this error code is related to the zigbee protocol but rather zstack itself, there error 205 defines no network route.

Since there are only reports of this error (in combination with groups) for the slaesh stick I expect it's something in the CC2652RB firmware (other sticks user the CC2652R firmware). If there are clear steps on how to reproduce this issue I can create a ticket on the Texas Instruments forum.

tim-brand, 2020-11-26:
Thanks @Koenk, I might know how to reproduce now, but will wait for my 2nd stick, to now abuse my home network

smathev, 2021-11-16:
@Koenkk 

I'm having this issue too, where zigbee2mqtt-defined groups are reporting "No network route" when I try to toggle them.

I can toggle all lightbulbs on/off in Home Assistant (HA), I can toggle the lightbulbs on/off in zigbee2mqtt (Z2M). I can create a group in HA and turn it on/off,  

If I create the same group in Z2M it reports "No route to network" when I interact with it in HA.  I'm still able to interact with the individual bulbs in HA and Z2M

Koenkk, 2021-11-16:
@smathev what adapter and firmware are you using?

smathev, 2021-11-16:
@Koenkk 
Thanks for getting back so fast! I'm loving your software, and using it for Home Assistant is a breeze! 
I'm using HA and Zigbee2mqtt as docker containers on my server. 

The adapter I'm using is from @slaesh - the cc2652-stik

I've recently run an update, and this is the information from the front-end of Zigbee2mqtt:
Zigbee2MQTT version : 1.22.0 commit: 0a1fccf
Coordinator type: zStack3x0
Coordinator revision: 20200715
Frontend version: 0.6.30

Koenkk, 2021-11-16:
@smathev you are using an old firmware, try updating to https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin

smathev, 2021-11-18:
@Koenkk 

It seems that this solved the issue. Sorry for not seeing the obvious issue at hand!

Tamadite, 2021-12-31:
I'm experiencing the same issue that is, not able to steer Zigbee devices when using CC2652p.

**The story**
1. 14 device Zigbee network.
2. CC2531 coordinator presenting sporadical unresponsive problems.
3. Migrating to CC2652P Ebyte E72-2G4M20S1E flashed with a mastered version of CC1352P2_CC2652P_other_coordinator_20210708. 
4. Changed port config only on Z2M. All worked fine and could steer Zigbee devices. Although it showed a 40% LQI improvement over the CC2531, one far from coordinator motion sensor could not engage even using same antenna as on CC2531. Moving sensor 3 meters closer helped but disengaged when moving it back to original placement.  
5. I flashed CC2652p with original CC1352P2_CC2652P_other_coordinator_20210708. No improvement.
6. I flashed CC2652p with development CC1352P2_CC2652P_other_coordinator_2021121. Better LQI values (235%). The far node could not engage however when moving it closer, engaged and kept engaged when moved back to original placement.
7. Suddenly, out of the blue, a couple days after I started to get Zigbee unresponsive problems as with old CC2531. Since then it does not matter the firmware I flash on CC2652p I experience the phenomenon exposed in this thread: Zigbee devices engage (except for the far one) but cannot be steered showing "no network route" error.

**Notes**
1) The flashing tool I use is [flasher_CC2538](https://smartlight.me/files/support/flashers/flasher_cc2538.zip).
2) I have tried egony's [firmware](https://github.com/egony/cc2652p_E72-2G4M20S1E/tree/master/firmware/coordinator) ver. 20210901 without success. As mentioned by Koenkk, egony warns on SDK bug.
3) Originally I had 1 group with 4 lamps and 1 remote bound to them. Not in use any longer as it got corrupted during testing & troubleshooting.
3) When I set the old CC2531 back and I change the port on Z2M config everything works as expected including the far node.
4) I am currently using my old CC2531.
5) Antenna characteristics are in specs according to antenna analyser.

Tamadite, 2022-01-07:
I gave up with the troubleshooting of the migration. Today I went for Koenkk [recommendation](https://github.com/Koenkk/zigbee2mqtt/issues/4135#issuecomment-682092658) given above. On the top of that one I also changed the panid in the configuration just to avoid the existing nodes showing up as error in the log because they were not paired yet. 
After a clean start and repairing everything is working fine (so far).

Littlesloths, 2022-07-04:
the "no network route" error is still active.... I have three Ikea tradfri remotes E1810. They worked before in that regard. Now I changed to ESP32 Zigbee Bridge Pro and 2/3 are not working. 
There are discussions on 
https://github.com/Koenkk/zigbee2mqtt/issues/7716
others think it is related to the IKEA firmware on the remote control. 
Any new ideas?