Koenkk/zigbee2mqtt #4139: New devices

zhapwey, 2020-08-20:
hello:

I would like to contribute some equipment that I have tried, but I do not know how to do this in the right way.

### **Devices that work properly**

**MOES ZigBee Switch Module**

information:
https://github.com/Koenkk/zigbee2mqtt/issues/3925

    {
        zigbeeModel: ['TS0121'],
        model: 'TS0121',
        description: 'switch Actuador 1 channel with input',
        supports: 'on/off',
        vendor: 'Moes',
        fromZigbee: [fz.on_off,fz.ignore_genOta],
        toZigbee: [tz.on_off],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genOnOff']);
            await configureReporting.onOff(endpoint);
        },
    },

**Gewis Zigbee swich Module 1 channel with input**
information:
https://www.gewiss.com/ww/en/products/experience-catalogue/catalogs/series/product/domotics/wireless-devices-zigbee/GWA1521

    {
        zigbeeModel: ['GWA1521_Actuator_1_CH_PF'],
        model: 'GWA1521_Actuator_1_CH_PF',
        description: 'switch Actuador 1 channel with input',
        supports: 'on/off',
        vendor: 'Gewiss',
        fromZigbee: [fz.on_off,fz.ignore_genOta],
        toZigbee: [tz.on_off],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genOnOff']);
            await configureReporting.onOff(endpoint);
        },
    },

**Gewis Zigbee swich Module 2 channel with input**
information:
https://www.gewiss.com/ww/en/products/experience-catalogue/catalogs/series/product/domotics/wireless-devices-zigbee/GWA1522
    {
        zigbeeModel: ['GWA1522_Actuator_2_CH'],
        model: 'GWA1522_Actuator_2_CH',
        description: 'switch Actuador 2 channel with input',
        supports: 'on/off',
        vendor: 'Gewiss',
        fromZigbee: [fz.generic_state_multi_ep],
        toZigbee: [tz.on_off],
        endpoint: (device) => {
            return {'l1': 1, 'l2': 2}
        },
    },

**Gewis Zigbee shutter actuator**
information:
https://www.gewiss.com/ww/en/products/experience-catalogue/catalogs/series/product/domotics/wireless-devices-zigbee/GWA1531
    {
        zigbeeModel: ['GWA1531_Shutter'],
        model: 'GWA1531_Shutter',
        description: 'shutter actuator',
        supports: 'on/off',
        vendor: 'Gewiss',
        fromZigbee: [fz.cover_position_tilt,fz.ignore_basic_report],
        toZigbee: [tz.cover_state, tz.cover_position_tilt],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genLevelCtrl']);
            await configureReporting.brightness(endpoint);
        },
    },

I have more zigbee devices that are not in the library but I can't get anything I'll put them in another post in case someone can help me with them

Best Regards,

Koenkk, 2020-08-20:
- MOES ZigBee Switch Module should already be supported (as zigbeeModel `TS0121` is already supported). Does it work in the latest zigbee2mqtt? 
- Gewis Zigbee swich Module 1 channel with input -> added
- Gewis Zigbee swich Module 2 channel with input -> added
- Gewis Zigbee shutter actuator -> added

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

zhapwey, 2020-08-25:
Yes,

{"id":23,"type":"Router","ieeeAddr":"0xec1bbdfffed3e162","nwkAddr":25411,"manufId":0,"manufName":"_TYZB01_iuepbmpv","powerSource":"Mains (single phase)","modelId":"TS0121","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,3,4,5,6,1794,2820],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"TS0121","manufacturerName":"_TYZB01_iuepbmpv","powerSource":1,"zclVersion":3,"appVersion":67}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca1564e","endpointID":1}]}},"appVersion":67,"zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1598341998301}

Koenkk, 2020-08-25:
Added now!

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-08-25 by Koenkk