Koenkk/zigbee2mqtt #4142: Device turns itself back on without MQTT or action in logs

LuukML, 2020-08-20:
Hi, before i start, I love your product! 

# Bug Report
## What happened
The bug that I have now encountered is that after switching off via hassIO and z2m the lamp switches itself on again after +/- 30 seconds at 100% values. I use mqtt explorer to monitor the MQTT traffic, there is no message and I see nothing else in my log.txt. 

I added this lamp myself because it was not yet included in the supported devices. The product I added is: Signify (philips) InteractPro 38519400

     {
         zigbeeModel: ['Xi InterAct'],
         model: '912401483130',
         vendor: 'Philips',
         description: 'Panel',
         supports: 'Panel form interact line',
         extend: generic.light_onoff_brightness, // (also tested with hue instead of generic, same behavior)
     },

Hass extension config line:
    '912401483130': [cfg.light_brightness],

these are the logs: https://hastebin.com/uliyixiliv.cs

## What did you expect to happen
that the lamp stays off when the lamp is turned off

## How to reproduce it (minimal and precise)
to be determined later, 

## Debug Info
Zigbee2MQTT version: 1.14.0
Adapter hardware: CC2531
Adapter firmware version: ?

Hope you can help!

LuukML, 2020-08-20:
Today spent some extra time testing and tried to adjust some settings via MQTT.

Test 1: change the Hue power on behavior with
MQTT 'zigbee2mqtt / Philips_panel1 / set' with data '{"hue_power_on_behavior": "off"}'.

The response to this is this log: https://hastebin.com/anisaculud.py

Test 2: turn off the lamp from hass and immediately after turning off the lamp turn off the gateway.

Outcome: lamp stays off and bug seems fixed. When turning on the gateway, the lamp shows the same behavior as the bug I reported. 

Because of this behavior I suspect that I am doing something wrong in the (translator's?)settings. And that z2m expects a response from the lamp but does not get it (or receives an error) and then falls back to a last known position

LuukML, 2020-08-20:
Again some tests and updates, I am now running your most  resente version of z2m.

here's the hersman log I turned off logging a few seconds after the lamp turned back on: https://hastebin.com/uxefuqakod.coffeescript

Koenkk, 2020-08-20:
Don't see anything suspicious in the log. Would need to sniff the traffic to dive further in what is happening. https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

LuukML, 2020-08-20:
Tnx for your quick response! I just looked, I turned off as many divices in the environment as possible to have minimal traffic. at the times of switching on or off I tried to mark the packages and provide information about what I did there. Is this the correct file type of the logs?

[logs pcapng.zip](https://github.com/Koenkk/zigbee2mqtt/files/5105289/logs.pcapng.zip)

Koenkk, 2020-08-23:
Really strange indeed as there are no commands send to the device after that. So the device itself does this. Are you able to check with the hue bridge if it shows the same behaviour there? I'm not really sure how to continue else tbh.

LuukML, 2020-08-24:
okay, I took some time today to test again. First I just sniffed traffic and watched when it turned on. it was difficult to see exactly where it jumps but that was with package mat as source: 0x0000 Destination: broadcast protocol: zigbee zdp length: 48 info: permit join request. this was the case a few times and also the first time.

I then tested with turning off the lamp and pulling the usb dongle out of the gateway to see if it turned on and that was not the case. I then no longer got the permit join request message.

I then turned on the config permit_join: false: the bulb has been off for quite some time now and seems to show normal behavior. Is there an option to send 1 packet in the zigbee network, just like with mqtt? I would like to see if that idd jumps through that message. have 'now tested it 3x with the same outcome (join on and off)

I see that when the gateway is rebooted the same message in the network traffic: '' Permit join request '' and that the lamp switches on again after switching on and off.

Unfortunately I can't connect the panel to the HUE brige because this is part of the interact PRO philips line and looks like they are blocking it on the HUE.

Is this behavior explainable to you?

Koenkk, 2020-08-24:
Good that you find the issue, but it's very strange behaviour for sure! You can enable permit join while running via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigpermit_join , if it's indeed the permit join packet, the bulb should immediately turn on after this.

LuukML, 2020-08-25:
hi, tested and indeed it was the permit_join package. is there a way to see how or why something reacts that way? I think you work around the corner from my work if you feel like it, I like to treat you to a beer as a thank you, and to show what I'm working on: D

Koenkk, 2020-08-25:
@LuukML it's very strange for this device to react this way to the command, however this is something in the firmware of the device so we cannot fix it from the zigbee2mqtt side. Due to Corona I haven't been in Eindhoven for a long time (working from home), will let you know when I'm there again!

LuukML, 2020-08-26:
cool thanks for the help and you're always welcome for a beer! for now i shale close as i dont think Philips will give us acces to the firmware

closed 2020-08-26 by LuukML