Koenkk/zigbee2mqtt #4143: Link quality of end devices not showing in case of router

xlammertink, 2020-08-20:
# Bug Report
## What happened
To be honest I'm not even sure if this is a Zigbee2MQTT issue or a router firmware issue, but guys can probably tell more about it. I'm using a CC2531 coordinator in combination with a CC2530+CC2590 router. It's all working well, except for reported link quality. Instead of showing the link of the end-device and the closest router/coordinator, the link between the router and the coordinator is how.

Example: A door sensor reports it's state via a router. The link between the door sensor and router has a quality of 48, the link quality between router and coordinator is 89. Now the link quality for the door sensor device shows up as 89 instead of 48.

## What did you expect to happen
I expected the link quality of the link between the end device and first hop is reported as link quality.

Example: A door sensor reports it's state via a router. The link between the door sensor and router has a quality of 48, the link quality between router and coordinator is 89. I expected the door sensor to show a link quality value of 48. The link quality of the router would show as 89 (as it does already).

## How to reproduce it (minimal and precise)
Make a setup with a coordinator and router. Connect end device with router.

## Debug Info
Zigbee2MQTT version: 1.14.3
Adapter hardware:
- Coordinator: CC2531
- Router: CC2530+CC2590 (WeBee)
Adapter firmware version:
- Coordinator: CC2531_DEFAULT_20190608
- Router: CC2530_CC2591_router_2019_02

zhapwey, 2020-08-20:
Sorry to get into your topic I see in the description of your rooter
Router: CC2530 + CC2591 (WeBee)
What equipment is that? I am looking for a cc2530 + cc2591 that is not on the plate but finished and if it could be seen better.

xlammertink, 2020-09-06:
Hi @zhapwey , I bought a random one at AliExpress. It’s just the PCB with antenna, wired with jumper wires to my RPi. It’s ugly, but it does the job. If you want a nice one, check out: https://www.zigbee2mqtt.io/how_tos/how_to_create_a_cc2530_router.html

xlammertink, 2020-09-06:
Btw, is there anyone that can say something related to the post? I still experience the issue that link quality of the router is shown in home assistant instead of the link to the device. Is this something related to Zigbee2mqtt, or related to something else?

htvekov, 2020-09-06:
I have the exact same experience and question as you @xlammertink 

I would also expect that link quality presented in HA for a zigbee end device would be link quality between device and parent (router).

Ciao !

zhapwey, 2020-09-07:
I see the same problem link coordinator-router 160 and between rooter-end device 45 and the device report is 160, all devices connected to this rooter publish the same link quality.

Best Regards,

xlammertink, 2020-09-28:
@Koenkk sorry for bugging you, but do you know if this is a Zigbee2MQTT bug or related to the firmware? And in case of the first, can you provide some pointers so I can see if I can fix this?

Koenkk, 2020-09-28:
The coordinator only receives the message from the router, this message does not include the linkquality of the previous message (from door sensor to router) and therefore the coordinator only knows the linkquality of the message from router -> coordinator.

closed 2020-11-06 by github-actions[bot]