Koenkk/zigbee2mqtt #4145: Need support adding Merten Wiser Roller Shutter Controll

7h0mas-R, 2020-08-20:
Hi all,

I use Zigbee2MQTT since November after an article in CT - nice work. Especially since I moved to HomeAssistant, I have a nice Zigbee network running.

I am now trying to replace old roller shutter controls with new Models from Merten. They are compatible with Wiser system and I succeeded in pairing them as described in the "How to support new devices" tutorial.
I made it up to step 2 in the tutorial and added the device information, however, in the logs I find no traces of any message coming from the shutter controller.
I suppose that I need a "configure:" section and tried to draw conclusions from looking at the configure section of the already Supported Schneider "Wiser radiator thermostat" WV704R0A0902.
However, since it is explicitly configuring a Thermostat, just copying it will not work, I suppose.
Can anybody give some guidance where to find the necessary info I need to put here or how to sniff that out somehow? 
How was the Configuration for the Wiser Thermostat researched? I also tried to get some ideas from a driver implementation on the Homey platform, that was made for a Dimmer from the same series, but that also did not provide the right clue yet.

Here is the response I get after pairing:
https://pastebin.com/raw/KePHrFHG

And here is my configuration code in devices.js so far:
https://pastebin.com/embed_js/4z7iFsv3

After restarting Zigbee2MQTT the device gets listed in the log:
https://pastebin.com/raw/KAD62YHG

From here I am lost, no messages are logged whatever I press on the controller.

Zigbee2MQTT version:
1.14.3 on Hass.Io
Adapter hardware: CC2531
Adapter firmware version: CC2531_DEFAULT_20190608

Koenkk, 2020-08-23:
> but I do not yet get any response (tried sending "position" as payload and also tried "open" and "close" with all sorts of quotes (and with and without escape characters).

What did you see in the zigbee2mqtt debug log?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

> Not sure what to make of the "Cannot read property 'bind' of undefined" message - could it still be a syntax error in my message (e.g. wrong quotes)? 

Can you share the `data/database.db` entry of this device?

7h0mas-R, 2020-08-24:
> What did you see in the zigbee2mqtt debug log?

```
debug 2020-08-22 21:38:51: Received MQTT message on 'zigbee2mqtt/0x588e81fffe1399d8/set' with data '"position": 40'
debug 2020-08-22 21:38:51: Publishing 'set' 'state' to '0x588e81fffe1399d8'
error 2020-08-22 21:38:51: Publish 'set' 'state' to '0x588e81fffe1399d8' failed: 'Error: Cluster 'closuresWindowCovering' has no command 'undefined''
debug 2020-08-22 21:38:51: Error: Cluster 'closuresWindowCovering' has no command 'undefined'
    at Object.getCommand (/zigbee2mqtt-1.14.3/node_modules/zigbee-herdsman/dist/zcl/utils.js:107:19)
    at Endpoint.<anonymous> (/zigbee2mqtt-1.14.3/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:362:37)
    at Generator.next (<anonymous>)
    at /zigbee2mqtt-1.14.3/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:27:71
    at new Promise (<anonymous>)
    at __awaiter (/zigbee2mqtt-1.14.3/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:23:12)
    at Endpoint.command (/zigbee2mqtt-1.14.3/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:360:16)
    at Object.convertSet (/zigbee2mqtt-1.14.3/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:258:26)
    at EntityPublish.onMQTTMessage (/zigbee2mqtt-1.14.3/lib/extension/publish.js:201:52)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.14.3/lib/controller.js:375:44)
```
 
> Can you share the `data/database.db` entry of this device?

```
{"id":26,"type":"Router","ieeeAddr":"0x588e81fffe1399d8","nwkAddr":36856,"manufId":4190,"manufName":"Schneider Electric","powerSource":"Mains (single phase)","modelId":"1GANG/SHUTTER/1","epList":[21,5],"endpoints":{"5":{"profId":260,"epId":5,"devId":514,"inClusterList":[0,3,4,5,258,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"dateCode":""}}},"binds":[]},"21":{"profId":260,"epId":21,"devId":260,"inClusterList":[0,3,2821,65303],"outClusterList":[3,5,6,8,25,258],"clusters":{"genBasic":{"attributes":{"modelId":"1GANG/SHUTTER/1","manufacturerName":"Schneider Electric","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":5,"hwVersion":1,"dateCode":"","swBuildId":"001.005.005 R"}}},"binds":[]}},"appVersion":1,"stackVersion":5,"hwVersion":1,"dateCode":"","swBuildId":"001.005.005 R","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1598124785637}
```

BTW: Is there a simple intro tutorial explaining the Zigbee Communication? Tried looking into the Spec and some stuff I found on Google, but I had difficulties getting an overview, since I have only limited time to research on this.

Koenkk, 2020-08-24:
The message should be `{"position": 40}` not `"position": 40`

7h0mas-R, 2020-08-24:
arrgggh... I have obviously been blindfolded... when messing around with the quotes and escape characters in the home assistant developer GUI I must have accidentaly deleted the parantheses at some point. Due to lack of experience with Java and YAML I did not even notice...
Works like a charm now from the developer page. Will try out everything and make a pull request when ready. 
And then I finally found the Zigbee Cluster Library documentation, which was an additional eyeopener to make more sense of the code...
In case others have similar problems, this is how you have to send the data from the home assistant built in "service" panel on the "developer" tab (with proper escaping and double quotes):

```
topic: zigbee2mqtt/0x588e81fffe1399d8/set
payload: "{\"position\": 40}"
```

Looking forward to trying to integrate some more of the Wiser components... but probably not before Christmas :)
Will update devices.js when I get there.
Thanks for the good support and sorry for the slightly sheepish startup problems!

closed 2020-08-24 by 7h0mas-R

Xaqumuti, 2023-03-24:
Hi,
Even if the topic is closed, in case it helps with other devices, you can download the Zigbee attributes specification for Schneider Electric / Merten devices from here:
https://shop.exchange.se.com/en-US/apps/103010/zigbee-interface-and-behaviour-documentation-for-schneider-electric-connected-devices
Only e-mail registration needed.
Cheers