Koenkk/zigbee2mqtt #4146: Request attribute update

hvanoch, 2020-08-20:
Is it possible to request a zigbee device to send an update?
I have a Xiaomi socket with power measurement. But it sometimes doesn't send any update for minutes. 

I searched through the docs but couldnt find any info on this topic.

Koenkk, 2020-08-23:
Try applying the following changes: https://github.com/Koenkk/zigbee-herdsman-converters/pull/1497/files and see if it works by sending to `zigbee2mqtt/FRIENDLY_NAME/get` payload `{"power":""}`

hvanoch, 2020-08-24:
I missed your comment. But I was working on my own PR. And I have a working solution. see https://github.com/Koenkk/zigbee-herdsman-converters/pull/1501

I tested your PR but it does not work. Logs or showing:
```
Zigbee2MQTT:error 2020-08-24 21:07:11: Publish 'get' 'power' to 'socket' failed: 'Error: Read 0x04cf8cdf3c7c3e92/1 genAnalogInput(["presentValue"], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 34146 - 1 - 3 - 12 - 1 after 10000ms)'
Zigbee2MQTT:debug 2020-08-24 21:07:11: Error: Read 0x04cf8cdf3c7c3e92/1 genAnalogInput(["presentValue"], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 34146 - 1 - 3 - 12 - 1 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
Zigbee2MQTT:info  2020-08-24 21:07:11: MQTT publish: topic 'home/zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'get' 'power' to 'socket' failed: 'Error: Read 0x04cf8cdf3c7c3e92/1 genAnalogInput([\"presentValue\"], {\"timeout\":10000,\"disableResponse\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Timeout - 34146 - 1 - 3 - 12 - 1 after 10000ms)'","meta":{"friendly_name":"socket"}}'
```

closed 2020-09-18 by Koenkk