Koenkk/zigbee2mqtt #4148: quick fix to support device with unsupported version to devices.js.

kerryandjane, 2020-08-21:
# Bug Report
## What happened
Supported device https://www.zigbee2mqtt.io/devices/HLC610-Z.html has many versions it seems. 
I see 'HOMA1002', 'HOMA0019' versions have support already. Out of the box my device wasn't. Quick fix needed.

I bought a new HLC610-Z and the version is HOMA0006. 

Adding 'HOMA0006' to the device in node_modules/zigbee-herdsman-converters/devices.js gets this device working as expected.
so it looks like this: 
    {                                         
        zigbeeModel: ['HOMA1002', 'HOMA0019', 'HOMA0006'],
        model: 'HLC610-Z',                  
        vendor: 'Shenzhen Homa',            
        description: 'Wireless dimmable controller',
        extend: generic.light_onoff_brightness,
    },  

## What did you expect to happen
I expected the device to be supported by default. Can you add this permanently add this update to device.js please to the next update?
## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2MQTT version: 1.14.3
Adapter hardware: CC2531
Adapter firmware version:latest

thankyou.

closed 2020-09-29 by github-actions[bot]

huub66, 2021-08-03:
@Koenkk can you also add model '053' to devices.js for HCL-610-Z, as this version has been shipped to me. Following above procedure made it working. Or do I need to create a new issue?

Koenkk, 2021-08-03:
@huub66 done.

huub66, 2021-08-04:
> @huub66 done.
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Thnx!