Koenkk/zigbee2mqtt #4150: Can not add Lonsonho (Lerlink) X702A 2 Gang Wall Switch with neutral line to my network

dpositive, 2020-08-21:
# Bug Report
## What happened
Device is not detected by Zigbee2mqtt 

## What did you expect to happen
Device is detected by Zigbee2mqtt, added to my network and ready to work (two switches and power monitor sensor are available in Home assistant integrations page)

## How to reproduce it (minimal and precise)
1. Connect device to AC power according to wiring diagram from the device instruction
2. Press and hold any button for at least 10 seconds till the green light starts blinking
3. Enable joining by setting `permit_join: true` in the configuration.

## Debug Info
Zigbee2MQTT version: 1.14.3 
Adapter hardware: CC2592
Adapter firmware version: zStack30x
[Debug log](https://hastebin.com/mobidosawe.apache)

Koenkk, 2020-08-23:
No, you need an additional stick (the CC2531 to be specific if you follow: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html).

One last thing I can imagine being the problem is that these devices only work on certain channels, e.g. channel 15, 20 and 25. Can you try that?

dpositive, 2020-08-23:
> Can you try that?

Could you please give me the link to manual how to see what channel I'm currently using and how to change it?

Koenkk, 2020-08-23:
@dpositive see `channel` of https://www.zigbee2mqtt.io/information/configuration.html#configuration (default is 11). But note that changing requires repairing of all devices! (and there is no guarantee that a different channel might fix issues with this device).

dpositive, 2020-08-24:
I offered sniffer stick, as soon as I receive it I will send a log output.

dpositive, 2020-08-27:
@Koenkk Sniffed the zigbee traffic when putting the device into pairing mode. Could you take a look?
[Lonsonho.pcapng.zip](https://github.com/Koenkk/zigbee2mqtt/files/5137774/Lonsonho.pcapng.zip)

Koenkk, 2020-08-27:
The device joins through a router: `0x00158d0004451699` but for some reason this device does not send the network key (you should see a `Transport key` message after the `Assocation response`). I would recommend turning this device off to make sure it doesn't join via it.

dpositive, 2020-08-27:
@Koenkk I've disconnected my Aqara switch, which is a router in my network, from AC power, reloaded zigbee2mqtt addon, applied permission for joining the network, connected Lonsohho switch to AC power and turned it on to the pairing mode. Nothing changed .
Here is new Wireshark log
[Lonsonho2.pcapng.zip](https://github.com/Koenkk/zigbee2mqtt/files/5138070/Lonsonho2.pcapng.zip)

Koenkk, 2020-08-28:
@dpositive now it seems that the lonsonho switch doesn't send the associate request. In case you own a CC2531, can you test with that and a zstack 12 firmware?

dpositive, 2020-08-28:
@Koenkk ok, I will try to do it this weekend. 
Where can I get zstack 12 firmware? 
I saw that zStack30x is available at 
https://www.zigbee2mqtt.io/getting_started/flashing_the_cc2531.html

Koenkk, 2020-08-28:
https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/default

dekamilv, 2020-09-01:
Hi. 
Have similar switch. Doesn't works neither with z-stack 3.0 nor with 1.2 :(

dpositive, 2020-09-01:
@dekamilv I had no time to flash my CC2531 with z-stack 1.2. And now I understand that there is no reason to try as it wouldn't work at all. Are you able to sniff traffic for z-stack 1.2 when Lonsonho is trying to connect?

dekamilv, 2020-09-01:
> @dekamilv I had no time to flash my CC2531 with z-stack 1.2. And now I understand that there is no reason to try as it wouldn't work at all. Are you able to sniff traffic for z-stack 1.2 when Lonsonho is trying to connect?

As per this manual? 
https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

dpositive, 2020-09-01:
@dekamilv Yes

dekamilv, 2020-09-01:
> @dekamilv Yes

Will try this evening.

dekamilv, 2020-09-01:
Not sure done everything right.

> Zigbee2MQTT:error 2020-09-01 21:00:08: Failed to interview '0xec1bbdfffe782433', device has not successfully been paired
Zigbee2MQTT:info  2020-09-01 21:00:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0xec1bbdfffe782433"},"type":"pairing"}'
Zigbee2MQTT:error 2020-09-01 21:00:18: Failed to execute LQI for '0xec1bbdfffe782433'
[01.09.2020.zip](https://github.com/Koenkk/zigbee2mqtt/files/5158031/01.09.2020.zip)

dpositive, 2020-09-01:
@Koenkk any ideas?

Koenkk, 2020-09-02:
@dekamilv the sniff is decrypted, I need your network key to decrypt it (you can send it to me on telegram: @koenkk). Please also provide the herdsman debug logging of the failed interview here.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

dekamilv, 2020-09-02:
> @dekamilv the sniff is decrypted, I need your network key to decrypt it (you can send it to me on telegram: @Koenkk). Please also provide the herdsman debug logging of the failed interview here.
>
> To [...]

Long log.... In addons' window catch this 

[02.09.2020.txt](https://github.com/Koenkk/zigbee2mqtt/files/5163157/02.09.2020.txt)

dekamilv, 2020-09-02:
> @dekamilv the sniff is decrypted, I need your network key to decrypt it (you can send it to me on telegram: @Koenkk). Please also provide the herdsman debug logging of the failed interview here.
>
> To [...]

Found full log....
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/5163174/log.txt)

Koenkk, 2020-09-02:
For some reason the device keep trying to rejoin the network even while the coordinator accepted it, I don't see a reason for the device to behave in that way. 

Further investigation would require a sniff when pairing it to the original gateway (in order to compare the behaviour). https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

dekamilv, 2020-09-03:
I don't have any "original" Tuyas' gateways. 
Sorry.

dekamilv, 2020-09-21:
@Koenkk  hi.
This weekend I am upgraded my hassio with [E18-TBH-27](https://www.aliexpress.com/item/4000833817933.html?spm=a2g0s.9042311.0.0.27424c4dSPhs13) flashed by Ver_3.0_CC2530_CC2592_20190523 and looks like it paired itself. At least switch stop blinking and now I see it in control panel:

[image]
[image]

And networks' map, for my mind, looks strange...
[image]
 

In case you have any interest I can make some logs as per your instructions.

Koenkk, 2020-09-21:
This should be supported in the latest dev branch already: https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

dpositive, 2020-09-21:
@Koenkk Do you mean support for this current switch?

Koenkk, 2020-09-21:
@dpositive yes, but in your case I don't know why it doesn't pair. Not sure how @dekamilv managed to do this.

dekamilv, 2020-09-21:
> This should be supported in the latest dev branch already: https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

No, it not. I am all the time using zigbee2mqtt-edge.

> @dpositive yes, but in your case I don't know why it doesn't pair. Not sure how @dekamilv managed to do this.
I am really did nothing. Only "move" to 3.0.

Koenkk, 2020-09-21:
@dekamilv can you provide the zigbee2mqtt startup log?

dekamilv, 2020-09-21:
> @dekamilv can you provide the zigbee2mqtt startup log?

`zigbee2mqtt-edge
  "ban": [],
  "whitelist": [],
  "queue": {},
  "experimental": {},
  "socat": {
    "enabled": false,
    "master": "pty,raw,echo=0,link=/dev/ttyZ2M,mode=777",
    "slave": "tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5",
    "restartdelay": 1.0,
    "initialdelay": 1.0,
    "options": "-d -d",
    "log": false
  },
  "zigbee_herdsman_debug": false
}Debug information end
./run.sh: line 24: [Info] Configuration backup found in /share/zigbee2mqtt/.configuration.yaml.bk. Skipping config backup.: No such file or directory
[Info] Socat is DISABLED and not started
2020-09-21T18:52:22: PM2 log: Launching in no daemon mode
2020-09-21T18:52:23: PM2 log: App [npm:0] starting in -fork mode-
2020-09-21T18:52:23: PM2 log: App [npm:0] online
> zigbee2mqtt@1.14.3-dev start /app
> node index.js
Zigbee2MQTT:info  2020-09-21 18:52:30: Logging to console and directory: '/share/zigbee2mqtt/log/2020-09-21.18-52-30' filename: log.txt
Zigbee2MQTT:info  2020-09-21 18:52:32: Starting Zigbee2MQTT version 1.14.3-dev (commit #)
Zigbee2MQTT:info  2020-09-21 18:52:32: Starting zigbee-herdsman...
Zigbee2MQTT:info  2020-09-21 18:52:35: zigbee-herdsman started
Zigbee2MQTT:info  2020-09-21 18:52:35: Coordinator firmware version: '{"meta":{"maintrel":2,"majorrel":2,"minorrel":7,"product":2,"revision":20190523,"transportrev":2},"type":"zStack30x"}'
Zigbee2MQTT:info  2020-09-21 18:52:35: Currently 11 devices are joined:
Zigbee2MQTT:info  2020-09-21 18:52:35: Bedroom_sensor (0x00158d00022738c4): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Zigbee2MQTT:info  2020-09-21 18:52:35: Bathroom_sensor (0x00158d000227ebb0): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor (EndDevice)
Zigbee2MQTT:info  2020-09-21 18:52:35: Bathroom_switch (0x00158d000238a76d): WXKG02LM - Xiaomi Aqara double key wireless wall switch (EndDevice)
Zigbee2MQTT:info  2020-09-21 18:52:35: Entrance_door (0x00158d0001c2de65): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
Zigbee2MQTT:info  2020-09-21 18:52:35: Corridor_PIR (0x00158d0001e562f9): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
Zigbee2MQTT:info  2020-09-21 18:52:35: Guestroom_button (0x00158d00023d2157): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
Zigbee2MQTT:info  2020-09-21 18:52:35: Bedroom_button (0x00158d0001f3de90): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
Zigbee2MQTT:info  2020-09-21 18:52:35: 0xec1bbdfffe782433 (0xec1bbdfffe782433): Not supported (Router)
Zigbee2MQTT:info  2020-09-21 18:52:35: Wall_relay (0x00158d00044c6b7d): LLKZMK11LM - Xiaomi Aqara wireless relay controller (Router)
Zigbee2MQTT:info  2020-09-21 18:52:35: Bedroom_radiator (0x842e14fffef154e3): TS0601_thermostat - TuYa Radiator valve with thermostat (EndDevice)
Zigbee2MQTT:info  2020-09-21 18:52:35: Guestroom_sensor (0x00158d0003fa737b): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor (EndDevice)
Zigbee2MQTT:info  2020-09-21 18:52:35: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2020-09-21 18:52:36: Connecting to MQTT server at mqtt://homeassistant
Zigbee2MQTT:info  2020-09-21 18:52:38: Connected to MQTT server
Zigbee2MQTT:info  2020-09-21 18:52:38: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:info  2020-09-21 18:52:38: MQTT publish: topic 'zigbee2mqtt/Bedroom_sensor', payload '{"battery":100,"humidity":53.73,"linkquality":62,"pressure":1012.5,"temperature":24.55,"voltage":3035}'
Zigbee2MQTT:info  2020-09-21 18:52:38: MQTT publish: topic 'zigbee2mqtt/Bathroom_sensor', payload '{"battery":100,"humidity":53.79,"linkquality":67,"temperature":24.74,"voltage":3005}'
Zigbee2MQTT:info  2020-09-21 18:52:38: MQTT publish: topic 'zigbee2mqtt/Bathroom_switch', payload '{"battery":100,"linkquality":23,"voltage":3015}'
Zigbee2MQTT:info  2020-09-21 18:52:38: MQTT publish: topic 'zigbee2mqtt/Entrance_door', payload '{"battery":100,"contact":true,"linkquality":54,"voltage":3045}'
Zigbee2MQTT:info  2020-09-21 18:52:38: MQTT publish: topic 'zigbee2mqtt/Corridor_PIR', payload '{"battery":100,"illuminance":0,"illuminance_lux":0,"linkquality":46,"occupancy":false,"voltage":3025}'
Zigbee2MQTT:info  2020-09-21 18:52:38: MQTT publish: topic 'zigbee2mqtt/Guestroom_button', payload '{"battery":100,"linkquality":75,"voltage":3042}'
Zigbee2MQTT:info  2020-09-21 18:52:38: MQTT publish: topic 'zigbee2mqtt/Bedroom_button', payload '{"battery":100,"linkquality":75,"voltage":3012}'
Zigbee2MQTT:info  2020-09-21 18:52:38: MQTT publish: topic 'zigbee2mqtt/Wall_relay', payload '{"consumption":0.78,"linkquality":80,"power":1.15,"state_l1":"ON","state_l2":"OFF","temperature":38}'
Zigbee2MQTT:info  2020-09-21 18:52:38: MQTT publish: topic 'zigbee2mqtt/Bedroom_radiator', payload '{"auto_lock":"MANUAL","away_preset_days":1,"away_preset_temperature":15,"boost_time":300,"child_lock":"UNLOCKED","comfort_temperature":20,"current_heating_setpoint":"15.0","eco_temperature":15,"force":"normal","holidays":[{"hour":6,"minute":0,"temperature":20},{"hour":8,"minute":0,"temperature":15},{"hour":11,"minute":30,"temperature":15},{"hour":12,"minute":30,"temperature":15},{"hour":17,"minute":30,"temperature":20},{"hour":22,"minute":0,"temperature":15}],"linkquality":46,"local_temperature":"21.5","local_temperature_calibration":"-4.0","max_temperature":30,"min_temperature":14,"position":0,"preset":"schedule","week":"7","w [... cut ...] etworkAddress":35888,"powerSource":"Battery","type":"EndDevice","vendor":"TuYa"},{"dateCode":"20160516","description":"MiJia temperature & humidity sensor","friendly_name":"Guestroom_sensor","hardwareVersion":30,"ieeeAddr":"0x00158d0003fa737b","lastSeen":1600688760699,"manufacturerID":4151,"manufacturerName":"LUMI","model":"WSDCGQ01LM","modelID":"lumi.sensor_ht","networkAddress":35926,"powerSource":"Battery","softwareBuildID":"3000-0001","type":"EndDevice","vendor":"Xiaomi"}],"type":"devices"}'`

Koenkk, 2020-09-21:
You are not on the latest edge (`Starting Zigbee2MQTT version 1.14.3-dev (commit #)`), to update you need to uinstall/install if I remember correctly.

dekamilv, 2020-09-21:
> You are not on the latest edge (`Starting Zigbee2MQTT version 1.14.3-dev (commit #)`), to update you need to uinstall/install if I remember correctly.

Sorry. I didn't know that.

dekamilv, 2020-09-21:
> You are not on the latest edge (`Starting Zigbee2MQTT version 1.14.3-dev (commit #)`), to update you need to uinstall/install if I remember correctly.

Thank you very much. After manual update switch recognized (as 4 gang instead of only 2 i have, but it is much better than nothing).

[image]

Koenkk, 2020-09-21:
@dekamilv fixed, 

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

dpositive, 2020-09-21:
For me changing my z2m to "version":"1.14.4-dev" doesn't help. Waiting for Tuya original gateway be shipped to sniff traffic while pairing my switch with it.

barelby, 2020-09-24:
Hello. I have the same problem. Switch blinking with red/green lights and no any reaction in assistant page

dekamilv, 2020-09-24:
> Hello. I have the same problem. Switch blinking with red/green lights and no any reaction in assistant page

Z-Stack_Home_1.2 ?

barelby, 2020-09-24:
I don't know firmware version of stick (cc2538) - I bought ready for use device. But after  migrating to zigbee2mqttassistant-dev and Zigbee2mqtt 1.14.4.2 everything works fine.

NorDroN, 2020-10-14:
Hi!
I have the same problem.
cc2531 (zStack12, 20190608)
z2m (1.15.0)

Shaquu, 2020-10-14:
I have similar switch (1 gang) and it is working for me on the latest release version.
If you have a problem while pairing then try restarting Pi and getting it closer to the switch.

NorDroN, 2020-10-15:
I've put it closer as possible (≈10cm) and I've restarted my Pi (with stick) a lots. But it didn't help me.
I thought that my stick didn't work, but I could add some Xiaomi devices and see rows in logs z2m about this.
When I try to pair Lonsonho I can't see something new in logs .

dpositive, 2020-10-20:
@Koenkk finally today I received my Moes HY368 Radiator valve with original gateway. I sniffed the zigbee traffic when putting the Lonsonho switch into pairing mode with original Tuya gateway. Could you take a look?
* Lonsonho switch trying to connect to original Tuya gateway
[LonsonhoAddingToOriginalHub.pcapng.zip](https://github.com/Koenkk/zigbee2mqtt/files/5410706/LonsonhoAddingToOriginalHub.pcapng.zip)

P.S.:
I also have the same problem with adding my Moes HY368 Radiator valve to z2m. Could you also look at the sniffed the zigbee traffic for:
* Moes HY368 Radiator valve trying to connect to z2m
[TRVaddingToZ2M.pcapng.zip](https://github.com/Koenkk/zigbee2mqtt/files/5410688/TRVaddingToZ2M.pcapng.zip)
* Moes HY368 Radiator valve trying to connect to original Tuya gateway
[TRVaddingToOriginalHub.pcapng.zip](https://github.com/Koenkk/zigbee2mqtt/files/5410694/TRVaddingToOriginalHub.pcapng.zip)

Koenkk, 2020-10-22:
@dpositive I've checked your Lonsonho sniff, but it does not contain a pairing sequence.

You mention: 

> Lonsonho switch trying to connect to original Tuya gateway

But did it actually connect to it? 

I see the same in your TRV to TuYa gateway sniff.

dpositive, 2020-10-22:
Yes. They are actually connected and added to my Tuya Smart app. Should I try to sniff some other way? I can make any check you'll need. Also I can give you temwiever access if it is more convenient for you.

dpositive, 2020-10-22:
@Koenkk this is crazy. I've turned off all zigbee devices in my home, started zboss sniffer, turned on Tuya gateway on to pairing mode and nothing. Looks like sniffer doesn't see Tuya gateways messages. Than after I turned my TRV or Switch on to pairing mode I saw a few beacon requests and nothing else. But devices connected to Tuya gateway successfully. Also I cannot seу the traffic while sending commands to the devices thugh the Tuya app. But devices respond for the commands correctly.
It is weird but when I connect Xiaomi devices to my z2m I can see all the traffic

dpositive, 2020-10-22:
@Koenkk I've found the reason why there ware no pairing sequence.  My Tuya gateway works on 14th channel.
Sending you  sniffed the zigbee traffic for both switch and TRV devices. Hope it will help. I hate Tuya UI and could use its devices only with z2m!

[channel14.zip](https://github.com/Koenkk/zigbee2mqtt/files/5425978/channel14.zip)

Koenkk, 2020-10-23:
@Hacker-CB @kirovilya do you know if the CC2538 firmware contains the following change: https://github.com/Koenkk/Z-Stack-firmware/blob/f2a7aa17fba716c72dd46d4821328f7daeb1e238/coordinator/Z-Stack_3.x.0/firmware.patch#L260 ?

kirovilya, 2020-10-24:
@Koenkk no zgAllowRejoinsWithWellKnownKey variable in z-stack 3.0.2 and BDB_DEFAULT_TC_REQUIRE_KEY_EXCHANGE is False

Koenkk, 2020-10-24:
@dpositive I want to rule out any firmware issue, can you try if it work with a zigbee 1.2 cc2531 stick?

dpositive, 2020-10-24:
@Koenkk, do you need to have sniffed traffic with cc2531? Or I just can flash my sniffer stick coordinator firmware?

Koenkk, 2020-10-24:
@dpositive just flash the cc2531 with yoru coordinator firmware (don't need sniff). Use https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/default

dpositive, 2020-10-24:
@Koenkk I've flashed my cc2531 with CC2531ZNP-Prod.hex coordinator firmware.
Here is the flashing log:
pi@raspberrypi:~/flash_cc2531 $ ./cc_write CC2531ZNP-Prod.hex
  ID = b524.
  reading line 15490.
  file loaded (15497 lines read).
writing page 128/128.
verifying page 128/128.
 flash OK.

Than I disconnected cc2531 wires from my raspberry and  rebooted it. After Home Assistant was started Zigbee2MQTT says:
[z2mLog.html.zip](https://github.com/Koenkk/zigbee2mqtt/files/5433655/z2mLog.html.zip)

dpositive, 2020-10-24:
@Koenkk I flashed my cc2531 with coordinator firmware from the link. 
Had some problem with z2m starting but after I deleted coordinator_backup.json and database.db files from zigbee2mqtt folder everything worked correctly. 
All my previously added devices started to work with new coordinator. 
But unfortunately Tuya's TRV and switch still can not be added.

Koenkk, 2020-10-26:
- I see the device tries to join through `0x00158d0004451699`, what device is this?
- Can you try to pair the device very close to the coordinator (so it joins via that).

dpositive, 2020-10-28:
@Koenkk 0x00158d0004451699 - is my Aqara double key wired wall switch.
I turned it off and tried to add my switch and TRVs to my network through CC2538 stick - nothing changed
Than I tried to add them through CC2530 and all my Tuya devices were added!
TRVs were added correctly
Switch was added as X702 but actually it is X702A and should have exactly the same icon as X712A (has mechanic buttons instead of sensor buttons). Nevertheless it works correctly.
So CC2538 firmware should be fixed and updated.

dpositive, 2020-11-28:
Is there any chance to flash my cc2538 with new fixed firmware?

Koenkk, 2020-11-28:
@dpositive I don't maintain the CC2538 myself, I think @kirovilya can do this.

kirovilya, 2020-11-30:
@dpositive We communicated via telegram - you need to update the firmware version to 20201010 https://github.com/jethome-ru/zigbee-firmware/tree/master/ti/coordinator/cc2538_cc2592

dpositive, 2020-11-30:
@kirovilya, thank you so much! Closing the issue

closed 2020-11-30 by dpositive