Koenkk/zigbee2mqtt #4168: Xiaomi WXKG03LM devices stop executing automations

quequi, 2020-08-23:
Hi

Suddenly all my WXKG03LM devices not activate the automations. In zigbee2mqtt log files appears "single" and "double" actions but the automations aren't launched. If I execute the automation manually works perfectly.

Following I share log events and an automation.

zigbee2mqtt log
https://pastebin.com/YS9kUtSW

WXKG03LM device automation
https://pastebin.com/hC7ZzyWh

These are the versions that I have in my environment:

HassOS 4.12
Home Assistant 0.114.3
zigbee2mqtt 1.14.3

Any help it's appreciated.

Thanks in advance.

Regards

Koenkk, 2020-08-23:
Fixed!

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

quequi, 2020-08-30:
Hi Koenkk

I confirm that the problem has been solved in Dev Branch. 

I have switched to edge version. Thank you for fixing the problem.

Regards

closed 2020-08-30 by quequi

SVhuMMer, 2020-10-09:
I hate to say it, but it happened again: select Aqara switches stopped triggering automations under 1.15.0 or latest edge. All other zigbee devices seem to work properly :(