Koenkk/zigbee2mqtt #4171: Unable to unbind Philips Hue dimmer from deCONZ coordinator

vdkeybus, 2020-08-23:
# Bug Report
## What happened
Paired Hue remote to network with deCONZ coordinator. Pairing is successful, configuration is successful. Pressing a button produces Zigbee messages. However, trying to unbind the remote from the coordinator always fails:

Aug 23 22:45:36 zacate npm[1323974]: Zigbee2MQTT:debug 2020-08-23 22:45:36: Received MQTT message on 'zigbee2mqtt/bridge/unbind/0x0017880104ad2466' with data 'coordinator'
Aug 23 22:45:36 zacate npm[1323974]: Zigbee2MQTT:debug 2020-08-23 22:45:36: unbinding cluster 'genScenes' from '0x0017880104ad2466' to 'Coordinator'
Aug 23 22:45:36 zacate npm[1323974]: Zigbee2MQTT:error 2020-08-23 22:45:36: Failed to unbind cluster 'genScenes' from '0x0017880104ad2466' to 'Coordinator' (TypeError: Cannot read property 'message' of undefined)
Aug 23 22:45:36 zacate npm[1323974]: Zigbee2MQTT:info  2020-08-23 22:45:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"cluster":"genScenes","from":"0x0017880104ad2466","to":"Coordinator"},"type":"device_unbind_failed"}'
Aug 23 22:45:36 zacate npm[1323974]: Zigbee2MQTT:debug 2020-08-23 22:45:36: unbinding cluster 'genOnOff' from '0x0017880104ad2466' to 'Coordinator'
Aug 23 22:45:36 zacate npm[1323974]: Zigbee2MQTT:error 2020-08-23 22:45:36: Failed to unbind cluster 'genOnOff' from '0x0017880104ad2466' to 'Coordinator' (TypeError: Cannot read property 'message' of undefined)
Aug 23 22:45:36 zacate npm[1323974]: Zigbee2MQTT:info  2020-08-23 22:45:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"cluster":"genOnOff","from":"0x0017880104ad2466","to":"Coordinator"},"type":"device_unbind_failed"}'
Aug 23 22:45:36 zacate npm[1323974]: Zigbee2MQTT:debug 2020-08-23 22:45:36: unbinding cluster 'genLevelCtrl' from '0x0017880104ad2466' to 'Coordinator'
Aug 23 22:45:37 zacate npm[1323974]: Zigbee2MQTT:error 2020-08-23 22:45:37: Failed to unbind cluster 'genLevelCtrl' from '0x0017880104ad2466' to 'Coordinator' (TypeError: Cannot read property 'message' of undefined)
Aug 23 22:45:37 zacate npm[1323974]: Zigbee2MQTT:info  2020-08-23 22:45:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"cluster":"genLevelCtrl","from":"0x0017880104ad2466","to":"Coordinator"},"type":"device_unbind_failed"}'
Aug 23 22:45:37 zacate npm[1323974]: Zigbee2MQTT:error 2020-08-23 22:45:37: Failed to unbind

Binding the remote to the lamp is successful, though:

Aug 23 22:46:19 zacate npm[1323974]: Zigbee2MQTT:debug 2020-08-23 22:46:19: Received MQTT message on 'zigbee2mqtt/bridge/bind/0x0017880104ad2466' with data '0x00178801083b7407'
Aug 23 22:46:19 zacate npm[1323974]: Zigbee2MQTT:debug 2020-08-23 22:46:19: binding cluster 'genScenes' from '0x0017880104ad2466' to '0x00178801083b7407'
Aug 23 22:46:21 zacate npm[1323974]: Zigbee2MQTT:debug 2020-08-23 22:46:21: Received Zigbee message from '0x0017880104ad2466', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":4,"time":0,"type":0,"unknown1":3145728,"unknown2":0}' from endpoint 2 with groupID null
Aug 23 22:46:21 zacate npm[1323974]: Zigbee2MQTT:debug 2020-08-23 22:46:21: Received Zigbee message from '0x0017880104ad2466', type 'commandOffWithEffect', cluster 'genOnOff', data '{"effectid":0,"effectvariant":0}' from endpoint 1 with groupID null
Aug 23 22:46:22 zacate npm[1323974]: Zigbee2MQTT:debug 2020-08-23 22:46:22: Received Zigbee message from '0x0017880104ad2466', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":4,"time":1,"type":2,"unknown1":3145728,"unknown2":0}' from endpoint 2 with groupID null
Aug 23 22:46:22 zacate npm[1323974]: Zigbee2MQTT:info  2020-08-23 22:46:22: MQTT publish: topic 'zigbee2mqtt/0x0017880104ad2466', payload '{"action":"off-press","battery":87.5,"counter":1,"duration":0,"linkquality":130,"update_available":false}'
Aug 23 22:46:22 zacate npm[1323974]: Zigbee2MQTT:info  2020-08-23 22:46:22: MQTT publish: topic 'zigbee2mqtt/0x0017880104ad2466', payload '{"action":"","battery":87.5,"counter":1,"linkquality":130,"update_available":false}'
Aug 23 22:46:22 zacate npm[1323974]: Zigbee2MQTT:info  2020-08-23 22:46:22: MQTT publish: topic 'zigbee2mqtt/0x0017880104ad2466/action', payload 'off-press'
Aug 23 22:46:24 zacate npm[1323974]: Zigbee2MQTT:info  2020-08-23 22:46:24: Successfully bound cluster 'genScenes' from '0x0017880104ad2466' to '0x00178801083b7407'
Aug 23 22:46:24 zacate npm[1323974]: Zigbee2MQTT:info  2020-08-23 22:46:24: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"cluster":"genScenes","from":"0x0017880104ad2466","to":"0x00178801083b7407"},"type":"device_bind"}'
Aug 23 22:46:24 zacate npm[1323974]: Zigbee2MQTT:debug 2020-08-23 22:46:24: binding cluster 'genOnOff' from '0x0017880104ad2466' to '0x00178801083b7407'
Aug 23 22:46:25 zacate npm[1323974]: Zigbee2MQTT:info  2020-08-23 22:46:25: Successfully bound cluster 'genOnOff' from '0x0017880104ad2466' to '0x00178801083b7407'
Aug 23 22:46:25 zacate npm[1323974]: Zigbee2MQTT:info  2020-08-23 22:46:25: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"cluster":"genOnOff","from":"0x0017880104ad2466","to":"0x00178801083b7407"},"type":"device_bind"}'
Aug 23 22:46:25 zacate npm[1323974]: Zigbee2MQTT:debug 2020-08-23 22:46:25: binding cluster 'genLevelCtrl' from '0x0017880104ad2466' to '0x00178801083b7407'
Aug 23 22:46:25 zacate npm[1323974]: Zigbee2MQTT:info  2020-08-23 22:46:25: Successfully bound cluster 'genLevelCtrl' from '0x0017880104ad2466' to '0x00178801083b7407'
Aug 23 22:46:25 zacate npm[1323974]: Zigbee2MQTT:info  2020-08-23 22:46:25: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"cluster":"genLevelCtrl","from":"0x0017880104ad2466","to":"0x00178801083b7407"},"type":"device_bind"}'

The lamp cannot be controlled directly from the remote, presumably because it was not previously unbound from the coordinator.

## What did you expect to happen
Unbind successful. Subsequent bind leading to lamp controllable by remote. Using the CC2531, this procedure worked.

## Debug Info
MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"410f6a0","coordinator":{"meta":{"maintrel":0,"majorrel":38,"minorrel":88,"product":0,"revision":"0x26580700","transportrev":0},"type":"ConBee2"},"log_level":"debug","network":{"channel":15,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.14.3-dev"}'
Zigbee2MQTT version: 
Adapter hardware:
Adapter firmware version:

Koenkk, 2020-08-24:
Can you provide the herdsman debug log of the coordinator unbind attempt?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

cvhoang, 2020-08-25:
I'm having the same issue as well. [Here](https://gist.github.com/cvhoang/5f81597d55e4386810dca6a40ccbbd93) is the relevant log.

vdkeybus, 2020-08-25:
Attached it. Thanks for looking into it.

(BTW how can you direct the npm output to a file; neither > nor 2> logged
anything. Had to use cut-and-paste.)

  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010 +5s
  zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 137 +5s
  zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 137 srcAddr: 0x4c0d destAddr: 0x0 profile id: 0x104 cluster id: 0xfc00 +34ms
  zigbee-herdsman:deconz:frameParser response payload: 29,11,16,15,0,4,0,0,48,0,33,0,0 +1ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":15,"manufacturerCode":4107,"commandIdentifier":0},"Payload":{"button":4,"unknown1":3145728,"type":0,"unknown2":0,"time":0}},"address":19469,"endpoint":2,"linkquality":154,"groupID":null}' +5s
Zigbee2MQTT:debug 2020-08-25 15:28:16: Received Zigbee message from '0x0017880104ad2466', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":4,"time":0,"type":0,"unknown1":3145728,"unknown2":0}' from endpoint 2 with groupID null
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010 +27ms
  zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 138 +52ms
  zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 138 srcAddr: 0x4c0d destAddr: 0x0 profile id: 0x104 cluster id: 0xfc00 +25ms
  zigbee-herdsman:deconz:frameParser response payload: 29,11,16,17,0,4,0,0,48,2,33,0,0 +1ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":17,"manufacturerCode":4107,"commandIdentifier":0},"Payload":{"button":4,"unknown1":3145728,"type":2,"unknown2":0,"time":0}},"address":19469,"endpoint":2,"linkquality":154,"groupID":null}' +52ms
Zigbee2MQTT:debug 2020-08-25 15:28:16: Received Zigbee message from '0x0017880104ad2466', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":4,"time":0,"type":2,"unknown1":3145728,"unknown2":0}' from endpoint 2 with groupID null
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010 +64ms
  zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 139 +100ms
  zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 139 srcAddr: 0x6da7 destAddr: 0x0 profile id: 0x104 cluster id: 0x300 +38ms
  zigbee-herdsman:deconz:frameParser response payload: 8,109,10,3,0,33,107,97 +0ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":109,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":3,"dataType":33,"attrData":24939}]},"address":28071,"endpoint":1,"linkquality":154,"groupID":0}' +102ms
Zigbee2MQTT:debug 2020-08-25 15:28:16: Received Zigbee message from '0x00158d00013d3202', type 'attributeReport', cluster 'lightingColorCtrl', data '{"currentX":24939}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-08-25 15:28:16: MQTT publish: topic 'zigbee2mqtt/licht_polyvalente_ruimte_2', payload '{"brightness":0,"color":{"x":0.3805,"y":0.3769},"color_temp":453,"state":"OFF"}'
Zigbee2MQTT:info  2020-08-25 15:28:16: MQTT publish: topic 'zigbee2mqtt/0x00158d00013d3202', payload '{"brightness":0,"color":{"x":0.3805,"y":0.3769},"color_temp":453,"linkquality":154,"state":"OFF"}'
  zigbee-herdsman:controller:endpoint DefaultResponse 0x00158d00013d3202/1 768(10, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) +5s                                                                                                                                                                                                                                                                        
  zigbee-herdsman:deconz:adapter zclFrame.payload: +5s
  zigbee-herdsman:deconz:adapter { cmdId: 10, statusCode: 0 } +1ms
  zigbee-herdsman:deconz:adapter no response expected +0ms
  zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0x6da7 EP:1 SeqNr. 140 request id: 32 +51ms
  zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 32 status: 0 +47ms
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110 +8ms
  zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 141 +51ms
  zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0x6da7 request id: 32 confirm status: 0 +42ms
  zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send +85ms
Zigbee2MQTT:info  2020-08-25 15:28:16: MQTT publish: topic 'zigbee2mqtt/0x0017880104ad2466', payload '{"action":"off-press","battery":100,"counter":1,"duration":0,"linkquality":154,"update_available":false}'
Zigbee2MQTT:info  2020-08-25 15:28:16: MQTT publish: topic 'zigbee2mqtt/0x0017880104ad2466', payload '{"action":"","battery":100,"counter":1,"linkquality":154,"update_available":false}'
[... cut ...]
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110 +8ms
  zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 160 +50ms
  zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0x39e7 request id: 37 confirm status: 0 +42ms
  zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send +71ms
  zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 161 +9s
  zigbee-herdsman:deconz:frameParser device state: 10100010 +9s

Koenkk, 2020-08-25:
@ChrisHae could you look at this?

ChrisHae, 2020-08-26:
Yes, I will have a look

ChrisHae, 2020-09-04:
I found an error in the unbind function. I have testet it and for me unbinding Philips Dimmer switch and binding to a light worked. I could dim and put light on/off.

I have uploaded the fix to the zigbee-herdsman repository.

vdkeybus, 2020-09-04:
Thanks for the fix. Is this in a PR or already in the repo I can download ?

Koenkk, 2020-09-04:
@vdkeybus will be in the dev branch in 2 hours from now.

vdkeybus, 2020-09-05:
I'm not familiar with how herdsman is installed in the zigbee2mqtt.

I'm on the zigbee2mqtt dev branch, I updated (ran update.sh) but I still
cannot unbind. I'm on commit 410f6a0f95fcd837d2f04b90b6708b1d1c4a1a8c from
Aug 14, though.

I did lose my modifications from devices.js in zigbee-herdsman-converters
in this process, so update must have done something.

How can I check for the herdsman fix ?

vdkeybus, 2020-09-05:
I think I see, the zigbee-herdsman has its own repo, and the one that is
automatically created lags behind. I have a version 117 and I see the
zigbee-herdsman git repo's master is at 130.

I tried creating the zigbee-herdsman repo in the zigbee2mqtt one and now I
get

Sep 05 14:15:01 zacate npm[3143743]: internal/modules/cjs/loader.js:638
Sep 05 14:15:01 zacate npm[3143743]:     throw err;
Sep 05 14:15:01 zacate npm[3143743]:     ^
Sep 05 14:15:01 zacate npm[3143743]: Error: Cannot find module
'zigbee-herdsman'

I never understood very well why we need to register the presence of a file
when it already is (by its very existence in the filesystem)... Totally
obvious to the rest of the world, I should have ran a magic command to
register. What is this command ?

Thanks.

jeroen.vandenkeybus@gmail.com>:

vdkeybus, 2020-09-05:
How can I run zigbee2mqtt with the GitHub zigbee-herdsman and
zigbee-herdsman-converters ?

I want to commit my own changes so they aren't undone with an update of
zigbee2mqtt. I just destroyed my zigbee-herdsman-converters again.

Thanks

Koenkk, 2020-09-05:
Once you are on the dev branch, you can do `git pull` to get all latest changes, afterwards execute `npm ci`, this will install the zigbee-herdsman/zigbee-herdsman-converters version defined in the `package.json` file (which is almost always the latest). Note that `npm ci` will remove any manual changes to zigbee-herdsman-converters.

vdkeybus, 2020-09-05:
npm ci gave me version 117 ?

vdkeybus, 2020-09-05:
I found the trouble: I am on a forked repo to commit the changes I made to
support new devices. I discovered that these changes need to be made in
zigbee-herdsman-converters which isn't under that git repo and I plainly
forgot about it. Merged Koenkk's repo back in and now it works as it should.

I'd still like to know what the best way is to use a fork of
zigbee-herdsman-converters to submit a PR for a device that is added ?

I successfully tested the unbind from coordinator:

Sep 05 17:19:59 zacate npm[3163541]: Zigbee2MQTT:debug 2020-09-05 17:19:59:
unbinding cluster 'genOnOff' from '0x0017880104ad2466' to 'Coordinator'
Sep 05 17:20:00 zacate npm[3163541]: Zigbee2MQTT:info  2020-09-05 17:20:00:
Successfully unbound cluster 'genOnOff' from '0x0017880104ad2466' to
'Coordinator'
Sep 05 17:20:00 zacate npm[3163541]: Zigbee2MQTT:info  2020-09-05 17:20:00:
MQTT publish: topic 'zigbee2mqtt/bridge/log', payload
'{"message":{"cluster":"genOnOff","from":"0x0017880104ad2466","to":"Coordinator"},"type":"device_unbind"}'
Sep 05 17:20:00 zacate npm[3163541]: Zigbee2MQTT:debug 2020-09-05 17:20:00:
unbinding cluster 'genLevelCtrl' from '0x0017880104ad2466' to 'Coordinator'
Sep 05 17:20:00 zacate npm[3163541]: Zigbee2MQTT:info  2020-09-05 17:20:00:
Successfully unbound cluster 'genLevelCtrl' from '0x0017880104ad2466' to
'Coordinator'
Sep 05 17:20:00 zacate npm[3163541]: Zigbee2MQTT:info  2020-09-05 17:20:00:
MQTT publish: topic 'zigbee2mqtt/bridge/log', payload
'{"message":{"cluster":"genLevelCtrl","from":"0x0017880104ad2466","to":"Coordinator"},"type":"device_unbind"}'

and successfully did a bind to lamp:

Sep 05 17:21:39 zacate npm[3163541]: Zigbee2MQTT:debug 2020-09-05 17:21:39:
Received MQTT message on 'zigbee2mqtt/bridge/bind/0x0017880104ad2466' with
data '0x00178801083b7407'
Sep 05 17:21:39 zacate npm[3163541]: Zigbee2MQTT:debug 2020-09-05 17:21:39:
binding cluster 'genScenes' from '0x0017880104ad2466' to
'0x00178801083b7407'
Sep 05 17:21:40 zacate npm[3163541]: Zigbee2MQTT:info  2020-09-05 17:21:40:
Successfully bound cluster 'genScenes' from '0x0017880104ad2466' to
'0x00178801083b7407'
Sep 05 17:21:40 zacate npm[3163541]: Zigbee2MQTT:info  2020-09-05 17:21:40:
MQTT publish: topic 'zigbee2mqtt/bridge/log', payload
'{"message":{"cluster":"genScenes","from":"0x0017880104ad2466","to":"0x00178801083b7407"},"type":"device_bind"}'
Sep 05 17:21:40 zacate npm[3163541]: Zigbee2MQTT:debug 2020-09-05 17:21:40:
binding cluster 'genOnOff' from '0x0017880104ad2466' to '0x00178801083b7407'
Sep 05 17:21:40 zacate npm[3163541]: Zigbee2MQTT:info  2020-09-05 17:21:40:
Successfully bound cluster 'genOnOff' from '0x0017880104ad2466' to
'0x00178801083b7407'
Sep 05 17:21:40 zacate npm[3163541]: Zigbee2MQTT:info  2020-09-05 17:21:40:
MQTT publish: topic 'zigbee2mqtt/bridge/log', payload
'{"message":{"cluster":"genOnOff","from":"0x0017880104ad2466","to":"0x00178801083b7407"},"type":"device_bind"}'
Sep 05 17:21:40 zacate npm[3163541]: Zigbee2MQTT:debug 2020-09-05 17:21:40:
binding cluster 'genLevelCtrl' from '0x0017880104ad2466' to
'0x00178801083b7407'
Sep 05 17:21:41 zacate npm[3163541]: Zigbee2MQTT:info  2020-09-05 17:21:41:
Successfully bound cluster 'genLevelCtrl' from '0x0017880104ad2466' to
'0x00178801083b7407'
Sep 05 17:21:41 zacate npm[3163541]: Zigbee2MQTT:info  2020-09-05 17:21:41:
MQTT publish: topic 'zigbee2mqtt/bridge/log', payload
'{"message":{"cluster":"genLevelCtrl","from":"0x0017880104ad2466","to":"0x00178801083b7407"},"type":"device_bind"}'

Koenkk, 2020-09-05:
> I'd still like to know what the best way is to use a fork of
zigbee-herdsman-converters to submit a PR for a device that is added ?

There are some good guides available, e.g. https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request . An alternative would be to post your changes here so I can add them.

> and successfully did a bind to lamp:

Good that your issue is fixed now, assuming this can be closed since OP issue has been solved.

closed 2020-09-05 by Koenkk

cvhoang, 2020-09-06:
Is the fix in `latest-dev` Docker tag?

Koenkk, 2020-09-06:
yes

cvhoang, 2020-09-06:
It's not working for me. I assume it's because I have forked `zigbee-herdsman-converts` and added new device? Anyway, I've created a pull request.

cvhoang, 2020-09-08:
Still doesn't work for me with the latest version of dev Docker image. Any idea @Koenkk? Thanks.

elliot-wood, 2020-10-20:
Is this the same bug? Using koenkk/zigbee2mqtt:latest-dev as of 20 Oct 2020.

Here I'm trying to unpair a Hue Dimmer Switch from the coordinator, using a deconz adapter:

```Zigbee2MQTT:debug 2020-09-03 10:40:28: Loaded state from file /app/data/state.json
Zigbee2MQTT:info  2020-09-03 10:40:28: Logging to console and directory: '/app/data/log/2020-09-03.10-40-27' filename: log.txt
Zigbee2MQTT:debug 2020-09-03 10:40:28: Removing old log directory '/app/data/log/2020-09-03.09-51-53'
Zigbee2MQTT:info  2020-09-03 10:40:28: Starting Zigbee2MQTT version 1.15.0-dev (commit #466e963)
Zigbee2MQTT:info  2020-09-03 10:40:28: Starting zigbee-herdsman...
Zigbee2MQTT:debug 2020-09-03 10:40:28: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[20],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"deconz","path":"/dev/ttyAMA0"}}'
CREATED DECONZ ADAPTER
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[48,167,189,25,44,25,199,44,134,59,21,84,70,229,31,58],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[20]},"serialPort":{"path":"/dev/ttyAMA0","adapter":"deconz"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","adapter":{"concurrent":null}}' +0ms
  zigbee-herdsman:deconz:driver Opening with /dev/ttyAMA0 +0ms
  zigbee-herdsman:deconz:driver Serialport opened +13ms
  zigbee-herdsman:controller:log Started with result 'resumed' +22ms
  zigbee-herdsman:controller:log Injected database: true, adapter: true +0ms
  zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 1 paramId: 1 +190ms
  zigbee-herdsman:deconz:frameParser MAC: 0x00212effff05bb16 +0ms
  zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 2 paramId: 7 +207ms
  zigbee-herdsman:deconz:frameParser NWKADDR: 0 +199ms
Zigbee2MQTT:info  2020-09-03 10:40:29: zigbee-herdsman started
  zigbee-herdsman:deconz:driver send read firmware version request from queue. seqNr: 3 +215ms
  zigbee-herdsman:deconz:frameParser read firmware version response - version: 0,7,82,38 +216ms
Zigbee2MQTT:info  2020-09-03 10:40:29: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":82,"product":0,"revision":"0x26520700","transportrev":0},"type":"ConBee2/RaspBee2"}'
  zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 4 paramId: 5 +201ms
  zigbee-herdsman:deconz:frameParser PANID: 1a62 +200ms
  zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 5 paramId: 11 +202ms
  zigbee-herdsman:deconz:frameParser APS_EXT_PANID: 0xdddddddddddddddd +203ms
  zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 6 paramId: 28 +204ms
  zigbee-herdsman:deconz:frameParser CHANNEL: 20 +203ms
  zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 7 paramId: 24 +209ms
  zigbee-herdsman:deconz:frameParser NETWORK_KEY: hidden +213ms
Zigbee2MQTT:debug 2020-09-03 10:40:30: Zigbee network parameters: {"channel":20,"extendedPanID":"0xdddddddddddddddd","panID":6754}
Zigbee2MQTT:info  2020-09-03 10:40:30: Currently 3 devices are joined:
Zigbee2MQTT:info  2020-09-03 10:40:30: Bedside Left (0x00178801102897b5): 9290012573A - Philips Hue white and color ambiance E26/E27/E14 (Router)
Zigbee2MQTT:info  2020-09-03 10:40:30: Bedroom Ceiling (0x00178801102a41b7): 9290012573A - Philips Hue white and color ambiance E26/E27/E14 (Router)
Zigbee2MQTT:info  2020-09-03 10:40:30: Bedroom Dimmer Switch (0x0017880102c3e5a1): 324131092621 - Philips Hue dimmer switch (EndDevice)
Zigbee2MQTT:warn  2020-09-03 10:40:30: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2020-09-03 10:40:30: Allowing new devices to join.
Zigbee2MQTT:warn  2020-09-03 10:40:30: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2020-09-03 10:40:30: Zigbee: allowing new devices to join.
  zigbee-herdsman:controller:log Permit joining +1s
  zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xfffc EP:0 SeqNr. 8 request id: 1 +213ms
  zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 1 status: 0 +218ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1 +21ms
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101110 +148ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1 +147ms
query aps data confirm
  zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 9 +238ms
query aps data indication
  zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 10 +206ms
  zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 10 srcAddr: 0x9e10 destAddr: 0xfffc profile id: 0x0 cluster id: 0x0 +465ms
  zigbee-herdsman:deconz:frameParser response payload: 2,22,187,5,255,255,46,33,0,0,0 +1ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1 +22ms
query aps data confirm
[... cut ...]
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +100ms
query aps data confirm
  zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 60 +714ms
  zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xfffc request id: 13 confirm status: 0 +727ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +13ms
  zigbee-herdsman:deconz:adapter PERMIT_JOIN - 0 seconds +2s
Zigbee2MQTT:info  2020-09-03 10:42:31: zigbee-herdsman stopped
```

ChrisHae, 2020-10-20:
Just tested it with the current version. Binding and Undbing of Hue Dimmer switch works as intended. 
`zigbee-herdsman:deconz:adapter UNBIND RESPONSE - addr: 0x57a1 status: 0 +0ms
Zigbee2MQTT:info  2020-10-20 15:01:36: Successfully unbound cluster 'genLevelCtrl' from '0x001788011038d303' to 'Coordinator'`

Your Log shows that the gateway did not receive a response from the dimmer switch. 
So did you press a button on the switch before you send the request? That should keep the switch awake for a few seconds. Or perhaps bring it closer to the gateway?

elliot-wood, 2020-10-21:
@ChrisHae I've tried holding down one of the face buttons while sending off the unpair command, and it still does the same thing. I'm holding the remote next to the RaspBee module.