Koenkk/zigbee2mqtt #4173: Fyrtur blinds status inverted

rosenstand, 2020-08-24:
# Bug Report
## What happened

So, I've been using the IKEA Fyrtur blinds with zigbee2mqtt for a couple of months now. They work great and we (usually) control them with our Hue dimmer switch (through HA) by using the on-hold-release and off-hold-release actions to trigger an automation that calls cover.set_cover_position with either position: 0 (down) or position: 100 (up). Even binding the included remote to the blinds (through a group) works great.

I've been noticing that in Lovelace, the controls didn't seem to be correct, i.e. I could press the "down" button, but not the "up" button, when the blinds were down. I thought maybe the blinds didn't report the position correctly and went on with life. This evening I noticed in the history, that the state actually seem to be inverted. That is, when the blinds are down they are "Open" and when they are up they are "Closed".

[image]

Are anybody else seeing this, or am I doing something wrong?

## What did you expect to happen

The blinds to be "Open" when they are up and "Closed" when down.

## How to reproduce it (minimal and precise)

Pair Fyrtur blinds, add to Lovelace.

## Debug Info
Zigbee2MQTT version: 1.14.3
Adapter hardware: CC1352P-2
Adapter firmware version: 20200417 (just noticed new firmware is out, will update right away!)

Koenkk, 2020-08-24:
When setting position 100, is the cover open or closed?

rosenstand, 2020-08-25:
The cover is physically rolled up when at 100.
But the state shows as closed.
I would expect it to be open.

rosenstand, 2020-08-25:
Now they appear to report right. Only change since opening this is that I updated firmware on the cc1352-p2. I’ll close this and investigate further If it happens again. I suspect it is periodical :( as my early observations suggested.

closed 2020-08-25 by rosenstand