Koenkk/zigbee2mqtt #4175: Kwikset Smart Lock 914: state still working, but actions stop working

laurentbeaumer, 2020-08-25:
Hi, 

I have:

- Ubuntu 18.04 running on armhf (odroid xu4)
- ConBee 2
- Zigbee2MQTT
- Mosquitto
- Home Assistant
- Kwikset Smart ZigBee Lock 914

The installation is only week old, I believe I am running the latest both software and firmware.

The state is still updated in home assistant, however the actions (i.e. lock/unlock) suddenly stop working after few days (I was showing off to my father-in-law, one minute it was working, the next one it wasn't)

It worked once yesterday (i.e. days later), but stop working on the 2nd attempt and after.

I inspected the log but didn't see anything relevant.

I restarted one by one all three services.

There is a similar thread saying, it could be an issue with the authorization token not being refreshed at expiry.

I am happy to dig in the logs if it helps, but I am assume it is not an isolated issue. 

I am looking for a past similar issue which has been resolves, and then I'll reproduce the same steps.

Rebooting the device or repairing is not a long term viable solution (I haven't tried).

Thanks

laurentbeaumer, 2020-08-25:
[image]

Koenkk, 2020-08-25:
- Did you ever use and disabled the report feature? https://www.zigbee2mqtt.io/information/report.html
- Can you try to configure the device via: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure ? Does it work again after this? (make sure to wakeup the device before doing otherwise configure might fail).

laurentbeaumer, 2020-08-26:
Thanks, I'll check

laurentbeaumer, 2020-08-26:
Not sure if it helps, I have in the log: 
"Zigbee2MQTT:error 2020-08-25 07:42:37: Failed to execute routing table for 'Kwikset Smart Lock 914'"

laurentbeaumer, 2020-08-27:
Worth noting that I add the lock manually in the configuration.yaml, and I couldn't lock/unlock.

laurentbeaumer, 2020-08-27:
I remove the battery from the lock, and put them back. The lock is normally responding (i.e. locking and unlocking). I'll wait 1 or 2 days to see it the issue reappears, if not, I'll close the issue.

laurentbeaumer, 2020-09-07:
The issue happened again after restarting zigbee2mqtt.io, the above solution fixed (again) the issue.

As long as the service is not restarted, the lock works normally.

closed 2020-09-07 by laurentbeaumer