Koenkk/zigbee2mqtt #4176: New device ?

frlp542, 2020-08-25:
hello guys

I have some new devices I would like to add.
How do I get them in the library ?
do I need to create a pull request ?

Thanks for the feedback

Filip

https://www.ecobright.be/zigbee-wandschakelaar/zigbee-wandschakelaar-draadloos-zwart-2-zones/
    {
        zigbeeModel: ['ED-10011'],
        model: 'ED-10011',
        vendor: 'EcoBright',
        description: 'Zigbee 4 button wall switch Black',
        supports: 'action',
        fromZigbee: [
            fz.command_on, fz.command_off, fz.command_move, fz.command_stop,
            fz.battery,
        ],
        toZigbee: [],
        meta: {multiEndpoint: true},
    },

McGiverGim, 2020-08-25:
I suppose that if you are able to create the PR, yes, is the fastest way. I think they must be added into the converters repo: https://github.com/Koenkk/zigbee-herdsman-converters/pulls you can see some PR with devices added there.

Seeing your text, I suppose you have read it, but you have instructions https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

frlp542, 2020-08-25:
> I suppose that if you are able to create the PR, yes, is the fastest way. I think they must be added into the converters repo: https://github.com/Koenkk/zigbee-herdsman-converters/pulls you can see [...]

Didn't see it at first as i only found how to add to the documentation. and converters. My device doesn't require a new convertor that's where i was lost.
I will try the pullrequest.

Thanks for the feedback

closed 2020-10-03 by github-actions[bot]