Koenkk/zigbee2mqtt #4177: Pairing Hue Tap with Conbee II fails

wzbfyb, 2020-08-25:
When I try to pair the Hue Tap switch, I get 

`2020-08-25T09:04:51.786Z zigbee-herdsman:deconz:frameParser received unknown command - id 25`

in the log
My Network is on channel 15, thus I pressed the two-dot button, for 10 seconds.
Immediately after I press the button I get the unknown command message in the log. The message is the same for any button from the device. I try to pair it within reach of two Philips Hue Bulbs. the rest of my bulbs are Innr.

## Debug Info
Zigbee2MQTT version: 1.14.3
Adapter hardware: Conbee II
Adapter firmware version: 26580700

Koenkk, 2020-08-25:
@ChrisHae could you look at this?

ChrisHae, 2020-08-26:
I will have a look, I think the green power command isn't implemented yet.

ChrisHae, 2020-08-28:
I think green power implementation will be done next week.

wzbfyb, 2020-08-28:
That’s absolutely awesome, thanks!
Time to get rid of my enocean light switches.

ChrisHae, 2020-09-04:
I have uploaded Green Power implementation to zigbee-herdsman master branch. Perhaps you have to wait until it is integrated in zigbee2mqtt or ha-plugin.

I couldn't test it with HA because I have no installation here. but I could pair the Hue tap with zigbee2mqtt and receive messages from it. Would be nice if you could test it in a real life case with Home Assistant.

wzbfyb, 2020-09-04:
hi!

actually I use it with node red, but I can try with HA as well, thanks a lot!

Koenkk, 2020-09-04:
Changes will be in hassio edge in 2 hours from now.

wzbfyb, 2020-09-05:
Awesome. When will this be in the docker images?

Koenkk, 2020-09-05:
it's already (latest-dev)

wzbfyb, 2020-09-18:
i can confirm it works with Conbee II on 1.14.4-dev latest

Koenkk, 2020-09-18:
Thanks for confirming, I assume this can be closed now.

closed 2020-09-18 by Koenkk