Koenkk/zigbee2mqtt #4178: Salus SP600 no longer reporting status

derekoharrow, 2020-08-25:
# Bug Report

I have two Salus SP600 power monitoring plug sockets. Neither of them is reporting status anymore - neither the switch on/off status, nor the power consumption status.

I have re-paired them, and they re-joined the network just fine. I can also control them no problem, but they are not routinely updating their status. They both used to, but both have stopped over the last week or two.

The only time I seem to receive state information is when changing the on/off switch status - log below.

```
debug 2020-08-25 10:55:08: Received MQTT message on 'zigbee2mqtt/Tumble_Dryer/set' with data '{"state": "OFF"}'
debug 2020-08-25 10:55:08: Publishing 'set' 'state' to 'Tumble_Dryer'
info  2020-08-25 10:55:08: MQTT publish: topic 'zigbee2mqtt/Tumble_Dryer', payload '{"energy":86.64,"linkquality":15,"state":"OFF","last_seen":"2020-08-25T10:55:08+01:00"}'
debug 2020-08-25 10:55:13: Received MQTT message on 'zigbee2mqtt/Tumble_Dryer/set' with data '{"state": "ON"}'
debug 2020-08-25 10:55:13: Publishing 'set' 'state' to 'Tumble_Dryer'
info  2020-08-25 10:55:13: MQTT publish: topic 'zigbee2mqtt/Tumble_Dryer', payload '{"energy":86.64,"linkquality":15,"state":"ON","last_seen":"2020-08-25T10:55:13+01:00"}'
debug 2020-08-25 10:55:31: Received Zigbee message from 'Washing_Machine', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"manufacturerCode":4216,"imageType":38,"fileVersion":33640736}' from endpoint 9 with groupID 0
debug 2020-08-25 10:55:34: Received Zigbee message from 'Washing_Machine', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"manufacturerCode":4216,"imageType":38,"fileVersion":33640736}' from endpoint 9 with groupID 0
debug 2020-08-25 10:55:37: Received Zigbee message from 'Washing_Machine', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"manufacturerCode":4216,"imageType":38,"fileVersion":33640736}' from endpoint 9 with groupID 0
debug 2020-08-25 10:55:40: Received Zigbee message from 'Washing_Machine', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"manufacturerCode":4216,"imageType":38,"fileVersion":33640736}' from endpoint 9 with groupID 0
debug 2020-08-25 10:55:43: Received Zigbee message from 'Washing_Machine', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"manufacturerCode":4216,"imageType":38,"fileVersion":33640736}' from endpoint 9 with groupID 0
```

## Debug Info
Version of Zigbee2MqttAssistant (this software): 0.3.157 (0.3.157+Branch.master.Sha.ea388c3bbc2d43c4c9c62f27bb6764911ba4acc9) Release
Version of Zigbee2Mqtt: 1.14.3
Coordinator version: 20190619
Coordinator Zigbee Id: 0x00124b000e0eaa6f

Koenkk, 2020-08-25:
- Did you ever use and disabled the report feature? https://www.zigbee2mqtt.io/information/report.html
- Can you try to configure the device via: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure ? Does it work again after this?

PedroLamas, 2020-08-25:
I just ran a quick test with my SP600 and a 50W lamp, it seems to be working fine and reporting the values correctly!

PedroLamas, 2020-08-25:
@derekoharrow have you fully rebooted the SP600 (by fully turning the power off and on to the plug)?

derekoharrow, 2020-08-25:
> * Did you ever use and disabled the report feature? https://www.zigbee2mqtt.io/information/report.html
> * Can you try to configure the device via: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure ? Does it work again after this?

Adding report:true into the config yaml file seems to have fixed it - thank you!

closed 2020-08-25 by derekoharrow