Koenkk/zigbee2mqtt #4179: Samotech SM311 stopped reporting values

PedroLamas, 2020-08-25:
# Bug Report

I have 2 Samotech SM311 (generic of Sunricher ZG2835RAC) that stopped reporting values since I updated to the latest release.

These were working fine before but now something must have changed to cause this problem!

Looking at the description, this might be related to https://github.com/Koenkk/zigbee2mqtt/issues/4178

## Debug Info
Zigbee2MQTT version: 1.14.3
Adapter hardware: CC26X2R1
Adapter firmware version: 20200417

PedroLamas, 2020-08-25:
@Koenkk could this issue somehow be related with this change? https://github.com/Koenkk/zigbee-herdsman-converters/commit/7bd1eb8882803a03ad227723a4c249e56908c751

PedroLamas, 2020-08-25:
Here are the full logs: https://pastebin.com/7EZTkqcy

Koenkk, 2020-08-25:
- Did you ever use and disabled the report feature? https://www.zigbee2mqtt.io/information/report.html
- Can you try to configure the device via: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure ? Does it work again after this?

PedroLamas, 2020-08-25:
I've never knowingly enabled the reporting feature (never needed it before).

Already use the forced device config message, got a success message on the log but unfortunately that didn't fix the problem.

PedroLamas, 2020-08-25:
I just restarted the 2 Samotech SM311 dimmers I have (by turning the Lights mains power off and on) and now they are back working fine, so I guess this is a problem with the device... closing this issue for now, I'll keep an eye if this happens again!

closed 2020-08-25 by PedroLamas