Koenkk/zigbee2mqtt #4185: New device: Moes (Tuya?) ZigBee Thermostat Programmable Temp Controller BHT-002-GCLZBW

insipiens, 2020-08-26:
## What happened: Device is not supported
## What did you expect to happen: Expected to follow the procedure to add the device and review debug level logs - logs are empty. wireshark output included but outside my experience level to resolve
## How to reproduce it (minimal and precise): pair this new device?

## Debug Info
Zigbee2MQTT version: 1.14.3
Adapter hardware: CC2531
Adapter firmware version: current (last month)

Moes ZigBee Smart Thermostat Programmable Temperature Controller for Water/Gas Boiler BHT-002-GCLZBW 

Device website: https://www.moeshouse.com/collections/smart-thermostat/products/white-zigbee-smart-thermostat-programmable-temperature-controller-zigbee-hub-required-for-water-gas-boiler

User manual: https://support.gw-ec.com/upload/qa/2018/11/19/3019d10504894c7327d18fc07389bbe6.pdf

I've attached the work[?] I did in the attached text file: proforma device.js and resulting outputs etc.

[Moes Zigbee Thermostat.txt](https://github.com/Koenkk/zigbee2mqtt/files/5126862/Moes.Zigbee.Thermostat.txt)

Any advice on progressing from here appreciated

ginkage, 2020-08-26:
Try editing your devices.js like so:
```
    {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_aoclfnxz'}],
        model: 'BHT-002-GCLZB',
        vendor: 'Moes',
        description: 'Room thermostat Water/Gas Boiler',
        supports: 'thermostat, temperature',
        meta: {tuyaThermostatSystemMode: common.TuyaThermostatSystemModes},
        fromZigbee: [fz.tuya_thermostat, fz.tuya_thermostat_on_set_data, fz.ignore_basic_report],
        toZigbee: [
            tz.tuya_thermostat_child_lock, tz.tuya_thermostat_window_detection, tz.tuya_thermostat_valve_detection,
            tz.tuya_thermostat_current_heating_setpoint, tz.tuya_thermostat_system_mode, tz.tuya_thermostat_auto_lock,
            tz.tuya_thermostat_calibration, tz.tuya_thermostat_min_temp, tz.tuya_thermostat_max_temp,
            tz.tuya_thermostat_boost_time, tz.tuya_thermostat_comfort_temp, tz.tuya_thermostat_eco_temp,
            tz.tuya_thermostat_force,
        ],
    },
```

NE8XT, 2020-08-26:
> Try editing your devices.js like so:
> 
> ```
>     {
>         fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_aoclfnxz'}],
>         model: 'BHT-002-GCLZB',
>         vendor: 'Moes',
>         description: 'Room thermostat Water/Gas Boiler',
>         supports: 'thermostat, temperature',
>         meta: {tuyaThermostatSystemMode: common.TuyaThermostatSystemModes},
>         fromZigbee: [fz.tuya_thermostat, fz.tuya_thermostat_on_set_data, fz.ignore_basic_report],
>         toZigbee: [
>             tz.tuya_thermostat_child_lock, tz.tuya_thermostat_window_detection, tz.tuya_thermostat_valve_detection,
>             tz.tuya_thermostat_current_heating_setpoint, tz.tuya_thermostat_system_mode, tz.tuya_thermostat_auto_lock,
>             tz.tuya_thermostat_calibration, tz.tuya_thermostat_min_temp, tz.tuya_thermostat_max_temp,
>             tz.tuya_thermostat_boost_time, tz.tuya_thermostat_comfort_temp, tz.tuya_thermostat_eco_temp,
>             tz.tuya_thermostat_force,
>         ],
>     },
> ```
> 
> then re-pair your device.
=============================================
Zigbee2MQTT:info  2020-08-26 21:35:58: Successfully interviewed '0xbc33acfffe62f313', device has successfully been paired
Zigbee2MQTT:info  2020-08-26 21:35:58: Device '0xbc33acfffe62f313' is supported, identified as: Moes Room thermostat Water/Gas Boiler (BHT-002-GCLZB_thermostat)
Zigbee2MQTT:info  2020-08-26 21:35:58: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0xbc33acfffe62f313","model":"BHT-002-GCLZB_thermostat","vendor":"Moes","description":"Room thermostat Water/Gas Boiler","supported":true}}'

**Results in Z2M, how to test?**

insipiens, 2020-08-26:
Thanks @ginkage That certainly produced some results in the log 

`Aug 26 21:14:26 raspberrypi bash[17181]: Zigbee2MQTT:info  2020-08-26 21:14:26: Device '0xbc33acfffe62f32a' joined
Aug 26 21:14:26 raspberrypi bash[17181]: Zigbee2MQTT:info  2020-08-26 21:14:26: Starting interview of '0xbc33acfffe62f32a'
Aug 26 21:14:26 raspberrypi bash[17181]: Zigbee2MQTT:info  2020-08-26 21:14:26: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0xbc33acfffe62f32a"}}'
Aug 26 21:14:26 raspberrypi bash[17181]: Zigbee2MQTT:info  2020-08-26 21:14:26: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0xbc33acfffe62f32a"}}'
Aug 26 21:14:26 raspberrypi bash[17181]: Zigbee2MQTT:info  2020-08-26 21:14:26: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0xbc33acfffe62f32a"}}'
Aug 26 21:14:41 raspberrypi bash[17181]: Zigbee2MQTT:info  2020-08-26 21:14:41: Successfully interviewed '0xbc33acfffe62f32a', device has successfully been paired
Aug 26 21:14:41 raspberrypi bash[17181]: Zigbee2MQTT:info  2020-08-26 21:14:41: Device '0xbc33acfffe62f32a' is supported, identified as: Moes Room thermostat Water/Gas Boiler (BHT-002-GCLZB)
Aug 26 21:14:41 raspberrypi bash[17181]: Zigbee2MQTT:info  2020-08-26 21:14:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0xbc33acfffe62f32a","model":"BHT-002-GCLZB","vendor":"Moes","description":"Room thermostat Water/Gas Boiler","supported":true}}'
Aug 26 21:15:15 raspberrypi bash[17181]: zigbee-herdsman-converters:siterwell_gs361: NOT RECOGNIZED DP #536 with data {"type":"Buffer","data":[0,0,0,225]}
Aug 26 21:16:15 raspberrypi bash[17181]: zigbee-herdsman-converters:siterwell_gs361: NOT RECOGNIZED DP #536 with data {"type":"Buffer","data":[0,0,0,225]}
Aug 26 21:16:20 raspberrypi bash[17181]: zigbee-herdsman-converters:siterwell_gs361: NOT RECOGNIZED DP #528 with data {"type":"Buffer","data":[0,0,0,30]}
Aug 26 21:16:20 raspberrypi bash[17181]: zigbee-herdsman-converters:siterwell_gs361: NOT RECOGNIZED DP #528 with d..............................................
Aug 26 21:16:26 raspberrypi bash[17181]: zigbee-herdsman-converters:siterwell_gs361: NOT RECOGNIZED DP #1060 with data {"type":"Buffer","data":[1]}
*edited for brevity*`

So that is the default (unrecognised) for TuYa convertors, so maybe not Tuya at all. 

I got my hands on a Moes zigbee gateway this morning, so also linked it there. Despite getting the transport (network) key It's _very,very chatty_ and  its difficult to gather the codes [without experience?]

Appreciate the help

ginkage, 2020-08-26:
Hm... Judging by the modelID, it should've been TuYa, but then again, maybe it does need a special converter to implement. Oh well, at least we tried.

Utyff, 2020-08-27:
Hi @ginkage, I tried your devices.js update.
Device successfully joined and interviewed but converters are not working and no one parameters added to device.
Could you help with converters?

Here short log after join device
_debug 2020-08-27 03:19:13: Received Zigbee message from '0xbc33acfffe5ec839', type 'commandGetData', cluster 'manuSpecificTuyaDimmer', data '{"status":0,"transid":22,"dp":101,"fn":0,"data":{"type":"Buffer","data":[6,0,40,11,30,42,13,30,44,17,30,46,6,0,48,12,0,46,14,30,44,17,30,42,6,0,38,12,30,40,14,30,42,18,30,40]}}' from endpoint 1 with groupID 0
debug 2020-08-27 03:19:19: Received Zigbee message from '0xbc33acfffe5ec839', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 2020-08-27 03:19:22: Received Zigbee message from '0xbc33acfffe5ec839', type 'raw', cluster 'manuSpecificTuyaDimmer', data '{"type":"Buffer","data":[25,14,36,0,10]}' from endpoint 1 with groupID 0
debug 2020-08-27 03:19:22: No converter available for 'TS0601_thermostat' with cluster 'manuSpecificTuyaDimmer' and type 'raw' and data '{"type":"Buffer","data":[25,14,36,0,10]}'_

Full log here:
https://cloud.mail.ru/public/Ajwb/4SvBqiXFK

insipiens, 2020-08-27:
@Utyff I've done some further testing, the default log (on failed conversion has been really useful) and have found the following message (DP/data). These are all messages from the Thermostat in response to pressing buttons.

Green shaded entries I am certain of, the others I need to confirm further. 

[image]

insipiens, 2020-08-29:
` Zigbee2MQTT:info  2020-08-30 00:03:19: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe62f32a', payload '{"standby":"YES","linkquality":31,"local_temperature":"18.5","activated":"ON","set_temperature":"17.0","target_temperature":"15.0","Manual_set_temperature":"10.0","child_lock":"UNLOCKED"}'
`
Progress, of sorts - just amending my local installation. The codes are all different to Tuya codes currently provided. I've added them to the Tuya converter as extra codes for now.

Will look at the toZigbee.js converter next. 

attached the analysis of tozigbee message content
[image]

Illivion, 2020-09-04:
Hi, @insipiens. I am trying to make some protocol analysis too, but it's my first experience and i have some troubles. I can't catch transport key for decrypting payloads. I've tried many times to pair device with hub, but 'transport key' frame doesn't occur. I ask for advice

insipiens, 2020-09-04:
@Illivion 

You want to search for the "Transport key" identifier in the Info column in wireshark, you can reorder it by clicking the info column header.
 
Attached a screenshot of the process , locating the Transport key in the Info column and then locate the data for the key in the Zigbee security header. 

You'll need to enter this in the preferences for the zigbee protocol - AES-128 encryption.

I would recommend putting "zbee_aps" in the display filter once set up so you see relevant messages for Zigbee Home automation.

[image]

Koenkk, 2020-09-07:
Assuming this can be closed now because changes have been added.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-09-07 by Koenkk

way2many, 2020-09-08:
Is this Opentherm? I couldn't find the information on the website.

insipiens, 2020-09-08:
Not opentherm, uses just Tuya or Smart life app.

Zigbee2mqtt support has been added to the Dev branch and should be in the master at next update

poisondima, 2020-09-23:
@Koenkk, I have a thermostat for a warm floor BHT-002-**GB**LZB, in the dev version it was defined as "Moes Room thermostat water / gas boiler (BHT-002-**GC**LZB)". The difference is that it has 2 temperature sensors: internal and external.
```
info  2020-09-23 15:37:05: Successfully interviewed '0xbc33acfffe62f395', device has successfully been paired
info  2020-09-23 15:37:05: Device '0xbc33acfffe62f395' is supported, identified as: Moes Room thermostat water/gas boiler (BHT-002-GCLZB)
info  2020-09-23 15:37:05: MQTT publish: topic 'homeassistant/lock/0xbc33acfffe62f395/child_lock/config', payload '{"availability_topic":"zigbee2mqtt/bridge/state","command_topic":"zigbee2mqtt/0xbc33acfffe62f395/set/child_lock","device":{"identifiers":["zigbee2mqtt_0xbc33acfffe62f395"],"manufacturer":"Moes","model":"Room thermostat water/gas boiler (BHT-002-GCLZB)","name":"0xbc33acfffe62f395","sw_version":"Zigbee2MQTT 1.14.4-dev"},"json_attributes_topic":"zigbee2mqtt/0xbc33acfffe62f395","name":"0xbc33acfffe62f395_child_lock","payload_lock":"LOCK","payload_unlock":"UNLOCK","state_locked":"LOCKED","state_topic":"zigbee2mqtt/0xbc33acfffe62f395","state_unlocked":"UNLOCKED","unique_id":"0xbc33acfffe62f395_child_lock_zigbee2mqtt","value_template":"{{ value_json.child_lock }}"}'
info  2020-09-23 15:37:05: MQTT publish: topic 'homeassistant/climate/0xbc33acfffe62f395/climate/config', payload '{"action_template":"{% set values = {'idle':'off','heat':'heating','cool':'cooling','fan only':'fan'} %}{{ values[value_json.running_state] }}","action_topic":"zigbee2mqtt/0xbc33acfffe62f395","availability_topic":"zigbee2mqtt/bridge/state","current_temperature_template":"{{ value_json.local_temperature }}","current_temperature_topic":"zigbee2mqtt/0xbc33acfffe62f395","device":{"identifiers":["zigbee2mqtt_0xbc33acfffe62f395"],"manufacturer":"Moes","model":"Room thermostat water/gas boiler (BHT-002-GCLZB)","name":"0xbc33acfffe62f395","sw_version":"Zigbee2MQTT 1.14.4-dev"},"json_attributes_topic":"zigbee2mqtt/0xbc33acfffe62f395","max_temp":"30","min_temp":"5","mode_command_topic":"zigbee2mqtt/0xbc33acfffe62f395/set/system_mode","mode_state_template":"{{ value_json.system_mode }}","mode_state_topic":"zigbee2mqtt/0xbc33acfffe62f395","modes":["off","auto","heat"],"name":"0xbc33acfffe62f395_climate","temp_step":1,"temperature_command_topic":"zigbee2mqtt/0xbc33acfffe62f395/set/current_heating_setpoint","temperature_state_template":"{{ value_json.current_heating_setpoint }}","temperature_state_topic":"zigbee2mqtt/0xbc33acfffe62f395","temperature_unit":"C","unique_id":"0xbc33acfffe62f395_climate_zigbee2mqtt"}'
info  2020-09-23 15:37:05: MQTT publish: topic 'homeassistant/sensor/0xbc33acfffe62f395/linkquality/config', payload '{"availability_topic":"zigbee2mqtt/bridge/state","device":{"identifiers":["zigbee2mqtt_0xbc33acfffe62f395"],"manufacturer":"Moes","model":"Room thermostat water/gas boiler (BHT-002-GCLZB)","name":"0xbc33acfffe62f395","sw_version":"Zigbee2MQTT 1.14.4-dev"},"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0xbc33acfffe62f395","name":"0xbc33acfffe62f395_linkquality","state_topic":"zigbee2mqtt/0xbc33acfffe62f395","unique_id":"0xbc33acfffe62f395_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
info  2020-09-23 15:37:05: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Room thermostat water/gas boiler","friendly_name":"0xbc33acfffe62f395","model":"BHT-002-GCLZB","supported":true,"vendor":"Moes"},"type":"pairing"}'
```

insipiens, 2020-09-23:
> @Koenkk, I have a thermostat for a warm floor BHT-002-**GB**LZB, in the dev version it was defined as "Moes Room thermostat water / gas boiler (BHT-002-**GC**LZB)". The difference is that it has 2 [...]

@poisondima Do you know the DP of the second temperature sensor? I'm sure we can add it and amend the naming to indicate the family instead of a specific model.

Not sure how the HA configuration should be amended for a second temperature, if that is a concern...

poisondima, 2020-09-23:
@insipiens I have only CC2531, I don't have "Moes zigbee gateway"

insipiens, 2020-09-23:
There’s an error clause which should show any unknown dp codes, such as a second temperature when the log is set to debug....that should give the code.

poisondima, 2020-09-24:
@insipiens I got such a log, but there are no errors.
Tell me how to turn the thermostat on and off through HA? When I press a physical button on the device, the "running" ON / OFF attribute changes. Also, when the heating relay is turned on, the "heating" ON / OFF attribute changes.
[log.zip](https://github.com/Koenkk/zigbee2mqtt/files/5274239/log.zip)

insipiens, 2020-09-24:
@poisondima , I can make some changes which you can copy. Just a little concerned if your thermostat has different behaviour than mine - without the hub how do you know what functions it has that can be controlled through Zigbee?

On mine it seems to have two ontrollable behaviours:

- turn (RUNNING = OFF/ON)
- Turn Program: ON/OFF

Which of those do you want as the buttons? and which button does what out of the permitted system modes in HS [“auto”, “off”, “cool”, “heat”, “dry”, “fan_only”]

The only other function mine has is programming the thermostat but I don't know HA has an interface to do that plus I am sure Z2M can handle lengths of hex required.

If you're a github guru you could then handle the PR (I'm new to Github)

Let me know.....

I have changed mine this afternoon so the buttons turn on/off RUNNING, and a preset for PROGRAM Off/ON

poisondima, 2020-09-24:
Need a switch that will turn the device on and off.
If "RUNNING = OFF" then "state" and hvac_modes = off
If "RUNNING = ON" then "state" and hvac_modes = heat, and the attribute "heating: ON / OFF" changes itself depending on the set temperature.
"State" and hvac_modes = auto, should include the schedule set on the device itself, but I see no reason to programming the thermostat there, it is much more convenient to use automations in the Home Assistant.
[image]
hvac_modes: auto, heat, off
I am as new to github as you are. I don't know what "PR" is. I thought maybe you posted it to your repository https://github.com/insipiens/zigbee-herdsman-converters, but the last change there was 22 days ago.

insipiens, 2020-09-24:
So here is how mine now looks, It's the thermostat when you click the elipse in the top right corner.

- Operation: turn Off/On  [off, heat]  (on the normal thermostat view - these are the visible mode buttons) however they seem to just turn the thermostat itself off/on

- Preset: Hold/Program  (equivalent to use manual/use program )

I'll do a little more analysis tonight, if you think it looks acceptable I'll share the code changes with you so you can test it.

on your second temperature issue I don't know how to proceed without access to your model of thermostat.
[image]
[image]

poisondima, 2020-09-24:
@insipiens Where do I get this fix and try it? :)

insipiens, 2020-09-24:
I'll post it here for you to try later

insipiens, 2020-09-24:
I have included the surrounding coding for the device as well as the change made, make backups of each file before changing...

3 files need changing, I've arranged the coding sections below as follows.
**description of change
Format....filelocation/filename
code snippet**

Hope its clear........YAML is very particular about the indentation used. Best of luck :)

_changed the parameters of the climate configuration section for device BHT-002-GCLZB that generates the HA thermostat_

file location: opt/zigbee2mqtt/lib/extension/homeassisant.js

```
    'BHT-002-GCLZB': [cfg.lock_child_lock, climate(5, 30, 'current_heating_setpoint', 1, ['off', 'heat'], [], ['hold', 'program']),
    ],
```
**Added 2 converters to the device.js file - tz.moes_thermostat_mode, tz.moes_thermostat_standby**
/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices.js
```
    {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_aoclfnxz'}],
        model: 'BHT-002-GCLZB',
        vendor: 'Moes',
        description: 'Moes BHT series Thermostat',
        supports: 'thermostat, temperature',
        fromZigbee: [fz.moes_thermostat],
        toZigbee: [tz.moes_thermostat_child_lock, tz.moes_thermostat_current_heating_setpoint, tz.moes_thermostat_mode, tz.moes_thermostat_standby],
    },
```
**Added the two converters >>  moes_thermostat_mode & moes_thermostat_standby**
/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/toZigbee.js
```
    moes_thermostat_child_lock: {
        key: ['child_lock'],
        convertSet: async (entity, key, value, meta) => {
            sendTuyaCommand(entity, 296, 0, [1, value==='LOCK' ? 1 : 0]);
        },
    },
    moes_thermostat_current_heating_setpoint: {
        key: ['current_heating_setpoint'],
        convertSet: async (entity, key, value, meta) => {
            const temp = value;
            const payloadValue = utils.convertDecimalValueTo2ByteHexArray(temp);
            sendTuyaCommand(entity, 528, 0, [4, 0, 0, ...payloadValue]);
        },
    },
    moes_thermostat_mode: {
        key: ['preset'],
        convertSet: async (entity, key, value, meta) => {
            sendTuyaCommand(entity, 1026, 0, [1, value === 'hold' ? 0 : 1]);
            sendTuyaCommand(entity, 1027, 0, [1, value === 'program' ? 0 : 1]);
        },
    },
    moes_thermostat_standby: {
        key: ['system_mode'],
        convertSet: async (entity, key, value, meta) => {
            sendTuyaCommand(entity, 257, 0, [1, value === 'off' ? 0 : 1]);
        },
    },
```

insipiens, 2020-09-24:
Ah, also....on reading the instructions for mine, but they cover all models, it mentions that the floor temperature can be displayed by holding the temperature down button for 5 seconds. I wonder if that would create the missing DP code in the log?

poisondima, 2020-09-25:
Thank! Now at least I can turn the thermostat on and off!
The thermostat state still does not change to "off / heat" shows "unknown", so it does not display the operating mode.
Changing presets works: if I choose hold, the manual mode is turned on, if I choose program, the schedule mode is turned on. But, like the state, it does not remember and when opening these values are always empty.
[image]
Probably does not save values due to the same attributes: "preset_mode" and "preset_modes".
[image]
I know and tried to hold the button for 5 seconds to display the floor temperature, it shows the temperature on the screen, but does not send new DP codes.
In any case, thanks again for the help and the work done!

insipiens, 2020-09-25:
I’ll check the preset memory issue, would be a shame for leaving a little left not done.

Cheers

insipiens, 2020-09-25:
@poisondima Try this change:

I've marked the modifications with inline comment //mod23092020

File: /opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/fromZigbee.js
```
const moesThermostat = (model, msg, publish, options, meta) => {
    const dp = msg.data.dp;
    const data = msg.data.data;
    const dataAsDecNumber = utils.convertMultiByteNumberPayloadToSingleDecimalNumber(data);
    let temperature;
    /*See tuyaThermostat above for message structure comment */
    switch (dp) {
    case 257: // 0x0101 Thermostat on standby = OFF, running = ON
        return {system_mode: dataAsDecNumber ? 'heat' : 'off'};   //mod23092020
    case 296: // 0x2801 Changed child lock status for moes thermostat
        return {child_lock: dataAsDecNumber ? 'LOCKED' : 'UNLOCKED'};
    case 263: // 0x0701 Changed child lock status
        return {child_lock: dataAsDecNumber ? 'LOCKED' : 'UNLOCKED'};
    case 528: // 0x1002 set temperature
        temperature = dataAsDecNumber;
        return {current_heating_setpoint: temperature};
    case 536: // 0x1802 moes room temperature
        temperature = (dataAsDecNumber / 10).toFixed(1);
        return {local_temperature: temperature};
    case 556: // 0x2c02 Temperature calibration
        temperature = (dataAsDecNumber / 10).toFixed(1);
    case 1026: // 0x0204 Changed program mode for moes thermostat
        return {preset_mode: dataAsDecNumber ? 'program' : 'hold'};          //mod23092020
    case 1027: // 0x0304 Changed manual mode status for moes thermostat
        return {manual_mode: dataAsDecNumber ? 'ON' : 'OFF'};
    case 1060: // 0x2404 Moes Thermostat is Open or Closed
        return {Heat: dataAsDecNumber ? 'OFF' : 'ON'};
    default: // The purpose of the codes 1041 & 1043 are still unknown
        console.log(`zigbee-herdsman-converters:Moes BHT-002-GCLZB: NOT RECOGNIZED DP #${
            dp} with data ${JSON.stringify(data)}`);
    }
};
```

insipiens, 2020-09-25:
Also, I did notice in the smart life app that you can choose which temperature sensor reports, 3 choices..so probably that would be how to get floor sensor temperature reported instead of the internal sensor

poisondima, 2020-09-25:
[image]
Now turning on and off works and the state is displayed in HA correctly.
I don't have a tuya hub for zigbee and can't use the app.
I have two BHT-002-GBLW (Wi-Fi) and there you can get into additional settings if you hold down the 2 left buttons, in which you can select the temperature of the floor sensor, air sensor, or both at the same time. Unfortunately, in the version with Zigbee, it is impossible to enter this mode, the instructions also do not describe this, perhaps all these settings were transferred to the smart life application.

insipiens, 2020-09-25:
@poisondima 

it's your lucky day...I found the temperature choice setting

Here's a quick and dirty solution, make the following changes:

_Add a new converter to the devices file, tz.moes_thermostat_sensor_
File: /opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices.js
 ```
   {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_aoclfnxz'}],
        model: 'BHT-002-GCLZB',
        vendor: 'Moes',
        description: 'Moes BHT series Thermostat',
        supports: 'thermostat, temperature',
        fromZigbee: [fz.moes_thermostat],
        toZigbee: [tz.moes_thermostat_child_lock, tz.moes_thermostat_current_heating_setpoint, tz.moes_thermostat_mode, tz.moes_thermostat_standby, tz.moes_thermostat_sensor],
    },

```
Add the following new converter clause to the toZigbee.js file, to the end of the Moes thermostat entries

file: /opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/toZigbee.js

```
    moes_thermostat_sensor: {
        key: ['sensor'],
        convertSet: async (entity, key, value, meta) => {
            sendTuyaCommand(entity, 1067, 0, [1, value]);
        },
    },
```

Now you need to transmit an mqtt message from HA's mqtt integration. 

Topic will be like: _zigbee2mqtt/Moes_Room_Thermostat/set/sensor_   yours will have a different name.

value to send is one of [0|1|2] where 0 is the room temperature sensor, 1 is AL, 2 is OU 
no idea which one you need, you might want to experiment but it should show a symbol indicating floor heating on the device.

If the message command works you should get an error message in the log like: 
Moes BHT-002-GCLZB: NOT RECOGNIZED DP #1067 with data {"type":"Buffer","data":[1]}

This is the device confirming the change by responding with the new setting but we'd have to set up another rule in fromZigbee.js to convert that to something useful.  

Bon chance!

poisondima, 2020-09-26:
It really is a lucky day!
If I use 0 or 1 it looks like this:
[image]
If I use 2:
[image]
I need the setting exactly set number 2 sensor, which is built into the floor. Can this attribute be done like a switch in HA?

insipiens, 2020-09-26:
I’m not sure, but you could try making one in HA...if yiu needed actions you can automate the mqtt message publishing.

Are you satisfied with this now, so I can asj for it to be added to the dev branch?

poisondima, 2020-09-26:
Well, of course! this is just great, now I will buy another zigbee and sell the wifi version))

Falconpage, 2020-09-26:
Thanks for the great job done for this unit.
Have you found a way to set the temperature compensation ?

insipiens, 2020-09-26:
@Falconpage Funny you mention that, just finished doing that one.

Here's what I've done, be happy for feedback:

_/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/toZigbee.js_ 
Added a new converter:

```
    moes_thermostat_calibration: { 
        key: ['local_temperature_calibration'],
        convertSet: async (entity, key, value, meta) => {
            const temp = value;
            if (value < 0){ value = 4096+value}
            const payloadValue = utils.convertDecimalValueTo2ByteHexArray(value);
            sendTuyaCommand(entity, 539, 0, [4, 0, 0, ...payloadValue]); 
        },
    },
```
Add the converter to the moes entry in ../devices.js 
```

{
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_aoclfnxz'}],
        model: 'BHT-002-GCLZB',
        vendor: 'Moes',
        description: 'Moes BHT series Thermostat',
        supports: 'thermostat, temperature',
        fromZigbee: [fz.moes_thermostat],
        toZigbee: [tz.moes_thermostat_child_lock, tz.moes_thermostat_current_heating_setpoint, tz.moes_thermostat_mode, tz.moes_thermostat_standby, tz.moes_thermostat_sensor, tz.moes_thermostat_calibration],
    },
```

Add new converrter to _/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters```
/fromZigbee.js_

I also tidied up the response for changing the thermostat sensor choice so it's a little clearer.

```
const moesThermostat = (model, msg, publish, options, meta) => {
    const dp = msg.data.dp;
    const data = msg.data.data;
    const dataAsDecNumber = utils.convertMultiByteNumberPayloadToSingleDecimalNumber(data);
    let temperature;
    /*See tuyaThermostat above for message structure comment */
    switch (dp) {
    case 257: // 0x0101 Thermostat on standby = OFF, running = ON
        return {system_mode: dataAsDecNumber ? 'heat' : 'off'};
    case 296: // 0x2801 Changed child lock status for moes thermostat
        return {child_lock: dataAsDecNumber ? 'LOCKED' : 'UNLOCKED'};
    case 263: // 0x0701 Changed child lock status
        return {child_lock: dataAsDecNumber ? 'LOCKED' : 'UNLOCKED'};
    case 528: // 0x1002 set temperature
        temperature = dataAsDecNumber;
        return {current_heating_setpoint: temperature};
    case 536: // 0x1802 moes room temperature
        temperature = (dataAsDecNumber / 10).toFixed(1);
        return {local_temperature: temperature};
    case 539: // Calibration
        temperature = dataAsDecNumber;
        //for negative values produce complimentary hex (equivalent to negative values)
        if (temperature > 4000){ temperature = temperature-4096} 
        return {local_temperature_calibration: temperature};
    case 1026: // 0x0204 Changed program mode for moes thermostat
        return {preset_mode: dataAsDecNumber ? 'program' : 'hold'};          
    case 1027: // 0x0304 Changed manual mode status for moes thermostat
        return {manual_mode: dataAsDecNumber ? 'ON' : 'OFF'};
    case 1060: // 0x2404 Moes Thermostat is Open or Closed
        return {Heat: dataAsDecNumber ? 'OFF' : 'ON'};
    case 1067: // 0x2b04 Temperature sensor selected
    switch (dataAsDecNumber) {
        case 0:
            return {sensor: 'IN'};
            break;
        case 1:
            return {sensor: 'AL'};
            break;
        case 2:
            return {sensor: 'OU'};
            break;
        default:
            return {sensor: 'Not supported'};
        }
    default: // The purpose of the codes 1041 & 1043 are still unknown
        console.log(`zigbee-herdsman-converters:Moes BHT-002-GCLZB: NOT RECOGNIZED DP #${
            dp} with data ${JSON.stringify(data)}`);
    }
};
```

Odd thing is I got some error responses for the following codes, hadn't got those before...

```
Sep 26   15:56:48 raspberrypi bash[25927]: zigbee-herdsman-converters:Moes   BHT-002-GCLZB: NOT RECOGNIZED DP #530 with data   {"type":"Buffer","data":[0,0,0,70]}
--
Sep 26 15:56:48 raspberrypi   bash[25927]: zigbee-herdsman-converters:Moes BHT-002-GCLZB: NOT RECOGNIZED DP   #531 with data   {"type":"Buffer","data":[0,0,0,35]}
Sep 26 15:56:48 raspberrypi   bash[25927]: zigbee-herdsman-converters:Moes BHT-002-GCLZB: NOT RECOGNIZED DP   #532 with data   {"type":"Buffer","data":[0,0,0,5]}
```

I know that 531 and 532 are maximum and minimum allowed range of target temperature, but what is 530 with value 70?

Cheers

poisondima, 2020-09-26:
Calibration and Temperature sensor selected works!
[image]

Falconpage, 2020-09-26:
Thanks @insipiens !
It works, but I have to commit -4 for local_temperature_calibration topic to set -1°c below the current value.
I think there is a -3 default calibration on my unit.

insipiens, 2020-09-26:
@Falconpage Yes, it's an absolute change from the factory level measurement. put in 0 as your number, check the reading then put in the required calibration change. Should make sense then.

insipiens, 2020-09-26:
I seem to have found another problem - might just be my thermostat but sometimes it doesn't actually switch on the heating when in manual mode after setting the set target above read temperature. 

Also this happens if I physically change the settingon the device (not just through zigbee/HA) so not sure its a zigbee related issue.

Anyone else found this?

On a positive note: working on having the set program reported from the device - could I get it to send it from HA?  That'd be nice!

Falconpage, 2020-09-27:
I found the heating switch on only when a full digit temperature is below the target .

Room : 19.5°c and target 20°c -> no heat
Room : 19°c and target 20°c -> heat

Calibration works well, I think this setting is lost after main lost. We will have to do some automation for that.
I found an issue on this MOES Thermostat, horrible time accuracy. About 10 to 15 minutes a day.
Have you see this on yours ?

insipiens, 2020-09-27:
> I found the heating switch on only when a full digit temperature is below the target .
>
> Room : 19.5°c and target 20°c -> no heat
> Room : 19°c and target 20°c -> heat
>
> Calibration works well, I think [...]

I hope mine is just faulty then as it doesn't switch always despite temp difference.
 
I found the calibration remained intact all the time. Even after setting it with the smart life app and then pairing it to Z2M

Yes, the timekeeping is terrible. It does get local time from the Tuya/Moes hub but not via a Zigbee HA message - not sure that's possible with Z2M. Might have to be a case of manaaging the program from HA instead.

I've added this so you get the program in the Z2M logs - insert it after the switch(dp) line in file _fromZigbee.js_

```
    case 101: // set schedule  
    return {Program: [
        {p1: data[0]+'h:'+data[1]+'m '+data[2]+'°C'},
        {p2: data[3]+'h:'+data[4]+'m '+data[5]+'°C'},
        {p3: data[6]+'h:'+data[7]+'m '+data[8]+'°C'},
        {p4: data[9]+'h:'+data[10]+'m '+data[11]+'°C'},
        {sa1: data[12]+'h:'+data[13]+'m '+data[14]+'°C'},
        {sa2: data[15]+'h:'+data[16]+'m '+data[17]+'°C'},
        {sa3: data[18]+'h:'+data[19]+'m '+data[20]+'°C'},
        {sa4: data[21]+'h:'+data[22]+'m '+data[23]+'°C'},
        {su1: data[24]+'h:'+data[25]+'m '+data[26]+'°C'},
        {su2: data[27]+'h:'+data[28]+'m '+data[29]+'°C'},
        {su3: data[30]+'h:'+data[31]+'m '+data[32]+'°C'},
        {su4: data[33]+'h:'+data[34]+'m '+data[35]+'°C'},
        ]
    };
```

Then you can see what the program will be doing at the "wrong time" ;)

Falconpage, 2020-09-27:
@insipiens , can you post your toZigbee.js / fromZigbee.js ans devices.js ?
Thanks :)

insipiens, 2020-09-27:
@Falconpage Time issue seems resolved, certainly mine is keeping accurate time now - not sure why it wasn't before as the fix has been in place for > 1 year. Maybe a pairing issue. Can you repair and check?

Attached .txt files of the entries for toZigbee.js, fromZigbee.js, device.js, and /lib/extension/homeassistant.js just for this device.

[Moes_Thermo_lib_extension_homeassistantjs.txt](https://github.com/Koenkk/zigbee2mqtt/files/5288950/Moes_Thermo_lib_extension_homeassistantjs.txt)
[Moes_thermostat_entry_in_toZigbeejs.txt](https://github.com/Koenkk/zigbee2mqtt/files/5288951/Moes_thermostat_entry_in_toZigbeejs.txt)
[Moes_thermostat_entry_in_fromZigbeejs.txt](https://github.com/Koenkk/zigbee2mqtt/files/5288952/Moes_thermostat_entry_in_fromZigbeejs.txt)
[Moes_thermostat_entry_in_devicejs.txt](https://github.com/Koenkk/zigbee2mqtt/files/5288953/Moes_thermostat_entry_in_devicejs.txt)

 I think it is pretty well complete now. Will look to create a PR to Dev in the coming week or so.

Falconpage, 2020-09-27:
Great job, I will try those settings with my setup.
One more thing, I'm new to home assistant, where do you find thermostat card needed values ?  I can't find it and I love understand what I'm doing.

Thanks again

insipiens, 2020-09-27:
to create this specific card go to the HA MQTT integration, locate the room thermostat and you should see an "add to lovelace" choice....

Falconpage, 2020-09-27:
I know that, I just want to understand where you find that "system_mode" is needed with values 'off' and 'heat' to be able to display it on the thermostat card.

insipiens, 2020-09-27:
Just one final comment, concerning the problem I had changing temperature and the device not providing heat. On my device there's a setting called minimum temperature, which one would assume was the minimum allowed to be chosen - it was factory set to 5 degrees.

When I set a target temperature 5 degrees above the room temperature it would switch on heating - so decided to create a converter to change the minimum temp setting. Set that from 5 to 0 and now the temperature set works correctly.

the converter is included in the file attached previously.

insipiens, 2020-09-27:
Here and here as per the MQTT addition.

[image]

[image]

Falconpage, 2020-09-28:
I see that I've got a different thermostat card, I will check if my homeassistant.js is up to date with this value :     'BHT-002-GCLZB': [cfg.lock_child_lock, climate(5, 30, 'current_heating_setpoint', 1, ['off', 'heat'], [], ['hold', 'program']),
and remove, pair and restart HASS to see if the climate entity is good.

Have you find a documentation about the syntax in use here to declare a climate entity into the homeassistant.js ?

insipiens, 2020-09-28:
The details are created based on the settings in honeassistant.js

Sounds like you need to update yours like Below and then restart the z2m service

file location: opt/zigbee2mqtt/lib/extension/homeassisant.js

```
    'BHT-002-GCLZB': [cfg.lock_child_lock, climate(5, 30, 'current_heating_setpoint', 1, ['off', 'heat'], [], ['hold', 'program']),
    ],
```

Falconpage, 2020-09-28:
Clearly I need to update :)

Sorry for my poor english, as I said I like understand what I do.
If I want help the community in the futur I need to build my own documentation.
Can you tell me where you find documentation about the climate entity build syntax that you puted in the homeassistant.js ?
If you haven't time to help me I will undestand.

insipiens, 2020-09-28:
I just figured it out by trial and error

iquix, 2020-09-28:
I was also playing around with this device for a while.
Actually I'm a Smartthings user, and I have made my own DTH for Smartthings for this thermostat.
(for someone who is interested, my DTH is at my github repository)

Whenever I change the target temperature, both of 0x1002(target temperature) status packet and 0x2404(open/close) status packet floods like 30~40 times with same contents of data. (with only difference in tuya packet id header)

I want to know that whether it happens to your device with zigbee2mqtt.

insipiens, 2020-09-28:
@iquix yes, exactly the same. My guess is it might be because the actual switch on/off heat takes a moment and they wanted to keep the message flow going long enough to capture that information....

iquix, 2020-09-28:
> @iquix yes, exactly the same. My guess is it might be because the actual switch on/off heat takes a moment and they wanted to keep the message flow going long enough to capture that information....

Thank you for your reply, @insipiens.

I was wondering if it requires some sort of ACK packet to be sent to the thermostat, to stop that flooding.

I don't have tuya zigbee hub nor CC2531 with packet capturing firmware, so I could not test it out how it supposed to work on original tuya zigbee hub. But this flooding is so annoying, especially in cloud based zigbee system such as Smartthings.

Hacker-CB, 2020-09-28:
I'm implemeting similar thermostat _TYST11_wv90ladg (issue https://github.com/Koenkk/zigbee2mqtt/issues/1856 )
ETOP HT-08: http://www.etopcontrols.com/thermostat/detail/HT-08.html
Most of DPs are different.
Now I  finished all modes, presets and schedule.
One thing which I don't know how to implement - set time on thermostat. Standard cluster `genTime` does not work. Seems that I it use some custom DP. 
Is anybody knows, how to set time on Tuya device?

insipiens, 2020-09-28:
@iquix  in Z2M we can put a parameter in the config which limits repeat messages, so not something we’ve needed to look at. Sorry

@Hacker-CB that is already implemented in Z2M . There’s an attribute from a cluster library function that the device uses to read time. It doesn’t/shouldn’t need configuring for a particular device. I found it worked after re-pairing the device again.

Here’s where I asked that question and found out that. [https://github.com/Koenkk/zigbee-herdsman-converters/issues/1597](url)

Hacker-CB, 2020-09-28:
@insipiens I know about `genTime` cluster. Unfortunatelly, it does now work with ETOP Tuya thermostat.

Falconpage, 2020-09-28:
@insipiens , homeassisant.js fixed. Everythinh working great !  Good job.

I will work on home assistant side to light up flame only when "Heat":"ON".
Time not syncing with my thermostat, I will work on that.

Hacker-CB, 2020-09-28:
@Falconpage we found how to set time on thermostats.
I will submit PR tomorrow.

[100 more replies]