Koenkk/zigbee2mqtt #4186: Nothing is happening

giuliandenicola1, 2020-08-26:
#What happened
I paired IKEA remote (IKEA E1524/E1810) successfully to Z2M, it even sends battery, link quality and software update information via MQTT. Because of MQTT it is also addes to home assistant with no problems, so far so good.

The problem now is if I press a button it does absolutely nothing, no logs, no MQTT, just nothing.

## What did you expect to happen
That it sent out a state or MQTT topic with `toggle` , `toggle_hold`, etc.

## How to reproduce it (minimal and precise)
Buy adapter from link below -> run zigbee2mqtt docker container -> pair ikea remote -> press buttons

## Debug Info
Zigbee2MQTT version: 1.14.3
Adapter hardware: CC2531 ( [AliExpress ](https://nl.aliexpress.com/item/4000816404730.html?spm=a2g0s.9042311.0.0.74cc4c4dGZvwXr))

MrHarok, 2020-08-26:
I've got the same problem. I've done two things: updating HA from 12.3 to 14.3 and Zigbee2mqtt to most recent version (past version wasn't older than 3 months but I don't know the specific version I was running.

When I check the logbook it says:
0x680ae2fffe3df10a_action changed to
Afstandbediening open/dicht werkkamer_click changed to

So it looks like that 'something' is changing but it doesn't record the change. To be honest I dont know how it behaved before. Something else I noticed is that the remote has now two options: Click and Action.

Hope this helps figuring out the problem. If I have some spare them I'm happy to help

Koenkk, 2020-08-26:
@MrHarok you are experiencing a different issue (#4168), already fixed in the dev branch

@giuliandenicola1 what firmware are you running? https://www.zigbee2mqtt.io/getting_started/flashing_the_cc2531.html#how-to-check-the-installed-firmware-version

giuliandenicola1, 2020-08-26:
> @giuliandenicola1 what firmware are you running? https://www.zigbee2mqtt.io/getting_started/flashing_the_cc2531.html#how-to-check-the-installed-firmware-version

It's 20180507 but I don't have a CC debugger..

Koenkk, 2020-08-26:
This device is indeed not working with that firmware, you need to upgrade to a recent one. (https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/default)

closed 2020-08-26 by Koenkk