Koenkk/zigbee2mqtt #4187: Reporting of Innr RB 285 C broken

theFork, 2020-08-26:
Apparently, the reporting of the RB 285 C bulb by Innr is broken. When I turn on/off the bulb via a bound Ubisys C4 controller, the corresponding switch in home assistant does not flip. There is also no reaction in the Z2M log. This worked quiet well for a while, but a few days ago it stopped.

I assume, that the update to 1.14.3 brought this issue since I updated to it a few days ago. This release includes support for the "FL 120 C Innr Color Flex LED strip", maybe this is related?

Version of Zigbee2Mqtt: 1.14.3
Coordinator version: 20200417 (ZZH!)

This is my current `database.db`, the device not reporting anymore is the `0x00158d00038f2a98`:

```
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b001e17f0bd","nwkAddr":0,"manufId":0,"epList":[242,47,13,12,110,11,8,6,5,4,3,2,1],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outClusterList":[1280,1282],"clusters":{},"binds":[]},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[]},"47":{"profId":260,"epId":47,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"242":{"profId":41440,"epId":242,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]}},"interviewCompleted":true,"meta":{},"lastSeen":null}
{"id":2,"type":"Router","ieeeAddr":"0x001fee00000030b7","nwkAddr":58457,"manufId":4338,"manufName":"ubisys","powerSource":"Mains (single phase)","modelId":"S2 (5502)","epList":[1,2,3,4,5,200,232,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"S2 (5502)","manufacturerName":"ubisys","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":1,"hwVersion":5,"dateCode":"20180517-DE-FB0","swBuildId":""}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001e17f0bd","endpointID":1}]},"2":{"profId":260,"epId":2,"devId":266,"inClusterList":[0,3,4,5,6],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"swBuildId":""}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001e17f0bd","endpointID":1}]},"3":{"profId":260,"epId":3,"devId":1,"inClusterList":[0,3],"outClusterList":[5,6,8],"clusters":{"genBasic":{"attributes":{}}},"binds":[]},"4":{"profId":260,"epId":4,"devId":1,"inClusterList":[0,3],"outClusterList":[5,6,8],"clusters":{"genBasic":{"attributes":{}}},"binds":[]},"5":{"profId":260,"epId":5,"devId":1281,"inClusterList":[0,1794,2820],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}},"seMetering":{"attributes":{"multiplier":1,"divisor":1000,"instantaneousDemand":0}}},"binds":[{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b001e17f0bd","endpointID":1}]},"200":{"profId":49155,"epId":200,"devId":0,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"232":{"profId":260,"epId":232,"devId":1287,"inClusterList":[0,21,64512],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":1,"stackVersion":1,"hwVersion":5,"dateCode":"20180517-DE-FB0","zclVersion":2,"interviewCompleted":true,"meta":{"reporting":1,"configured":3},"lastSeen":1593354831079}
{"id":3,"type":"Router","ieeeAddr":"0x001fee0000001815","nwkAddr":58695,"manufId":4338,"manufName":"ubisys","powerSource":"Mains (single phase)","modelId":"S2 (5502)","epList":[1,2,200,232,242,3,4,5],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"S2 (5502)","manufacturerName":"ubisys","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":1,"hwVersion":5,"dateCode":"20170718-DE-FB0","swBuildId":""}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001e17f0bd","endpointID":1}]},"2":{"profId":260,"epId":2,"devId":266,"inClusterList":[0,3,4,5,6],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"swBuildId":""}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001e17f0bd","endpointID":1}]},"3":{"profId":260,"epId":3,"devId":1,"inClusterList":[0,3],"outClusterList":[5,6,8],"clusters":{"genBasic":{"attributes":{}}},"binds":[]},"4":{"profId":260,"epId":4,"devId":1,"inClusterList":[0,3],"outClusterList":[5,6,8],"clusters":{"genBasic":{"attributes":{}}},"binds":[]},"5":{"profId":260,"epId":5,"devId":1281,"inClusterList":[0,1794,2820],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}},"seMetering":{"attributes":{"multiplier":1,"divisor":1000,"instantaneousDemand":0}}},"binds":[{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b001e17f0bd","endpointID":1}]},"200":{"profId":49155,"epId":200,"devId":0,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"232":{"profId":260,"epId":232,"devId":1287,"inClusterList":[0,21,64512],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":1,"stackVersion":1,"hwVersion":5,"dateCode":"20170718-DE-FB0","zclVersion":2,"interviewCompleted":true,"meta":{"configured":3,"reporting":1},"lastSeen":1597824524493}
{"id":4,"type":"Router","ieeeAddr":"0x001fee0000003e72","nwkAddr":56825,"manufId":4338,"manufName":"ubisys","powerSource":"Mains (single phase)","modelId":"J1 (5502)","epList":[1,2,200,232,242,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":514,"inClusterList":[0,3,4,5,258],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"J1 (5502)","manufacturerName":"ubisys","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":1,"hwVersion":6,"dateCode":"20200609-DE-FB0"}},"closuresWindowCovering":{"attributes":{"currentPositionLiftPercentage":255,"currentPositionTiltPercentage":255}}},"binds":[{"cluster":258,"type":"endpoint","deviceIeeeAddress":"0x00124b001e17f0bd","endpointID":1}]},"2":{"profId":260,"epId":2,"devId":515,"inClusterList":[0,3],"outClusterList":[5,258],"clusters":{"genBasic":{"attributes":{}}},"binds":[]},"3":{"profId":260,"epId":3,"devId":1281,"inClusterList":[0,1794,2820],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}}},"binds":[]},"200":{"profId":49155,"epId":200,"devId":0,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"232":{"profId":260,"epId":232,"devId":1287,"inClusterList":[0,21,64512],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":1,"stackVersion":1,"hwVersion":6,"dateCode":"20200609-DE-FB0","zclVersion":2,"interviewCompleted":true,"meta":{"reporting":1},"lastSeen":1593023462816}
{"id":5,"type":"Router","ieeeAddr":"0x001fee0000002cad","nwkAddr":26830,"manufId":4338,"manufName":"ubisys","powerSource":"Mains (single phase)","modelId":"J1 (5502)","epList":[1,2,200,232,242,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":514,"inClusterList":[0,3,4,5,258],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"J1 (5502)","manufacturerName":"ubisys","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":1,"hwVersion":6,"dateCode":"20180920-DE-FB0"}},"closuresWindowCovering":{"attributes":{"currentPositionTiltPercentage":255,"currentPositionLiftPercent [... cut ...] of Sweden","powerSource":3,"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20170302","swBuildId":"1.2.214"}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":87}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b001e17f0bd","endpointID":1}]}},"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20170302","swBuildId":"1.2.214","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1,"reporting":1},"lastSeen":1598415998005}
```

sjorge, 2020-08-26:
If you physically cut the power to the bulb and restore it, do you see the setting up reporting for [genOnOff, genLevelCtrl, ..] messages?

Occasionally my coordinator goes out of memory and it fails to setup reporting.

theFork, 2020-08-27:
When doing a power cycle, and then switch the bulb off (via bound C4), the log looks as follows. I've filtered the log for the controlling C4 and the Innr bulb:

```
Zigbee2MQTT:info  2020-08-27 11:53:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"regal_innr_eg_ez"}}'
Zigbee2MQTT:info  2020-08-27 11:53:17: MQTT publish: topic 'zigbee2mqtt/regal_innr_eg_ez', payload '{"brightness":254,"linkquality":66,"state":"ON","color_temp":295,"color":{"x":0.4136,"y":0.3941}}'
Zigbee2MQTT:info  2020-08-27 11:53:18: MQTT publish: topic 'zigbee2mqtt/regal_innr_eg_ez', payload '{"brightness":254,"linkquality":72,"state":"ON","color_temp":295,"color":{"x":0.4136,"y":0.3941}}'
Zigbee2MQTT:info  2020-08-27 11:53:18: MQTT publish: topic 'zigbee2mqtt/regal_innr_eg_ez', payload '{"brightness":254,"linkquality":69,"state":"ON","color_temp":295,"color":{"x":0.4136,"y":0.3941}}'
Zigbee2MQTT:info  2020-08-27 11:53:28: MQTT publish: topic 'zigbee2mqtt/c4_eg_wz', payload '{"update_available":true,"linkquality":60,"action":"2_toggle"}'
Zigbee2MQTT:info  2020-08-27 11:53:29: MQTT publish: topic 'zigbee2mqtt/regal_innr_eg_ez', payload '{"brightness":254,"linkquality":21,"state":"OFF","color_temp":295,"color":{"x":0.4136,"y":0.3941}}'
Zigbee2MQTT:info  2020-08-27 11:53:30: MQTT publish: topic 'zigbee2mqtt/regal_innr_eg_ez', payload '{"brightness":254,"linkquality":24,"state":"OFF","color_temp":295,"color":{"x":0.4136,"y":0.3941}}'
Zigbee2MQTT:info  2020-08-27 11:53:31: MQTT publish: topic 'zigbee2mqtt/regal_innr_eg_ez', payload '{"brightness":254,"linkquality":21,"state":"OFF","color_temp":295,"color":{"x":0.4136,"y":0.3941}}'
Zigbee2MQTT:info  2020-08-27 11:53:39: MQTT publish: topic 'zigbee2mqtt/regal_innr_eg_ez', payload '{"brightness":254,"linkquality":18,"state":"OFF","color_temp":295,"color":{"x":0.4136,"y":0.3941}}'
Zigbee2MQTT:info  2020-08-27 11:53:39: MQTT publish: topic 'zigbee2mqtt/regal_innr_eg_ez', payload '{"brightness":254,"linkquality":18,"state":"ON","color_temp":295,"color":{"x":0.4136,"y":0.3941}}'
Zigbee2MQTT:info  2020-08-27 11:53:39: MQTT publish: topic 'zigbee2mqtt/regal_innr_eg_ez', payload '{"brightness":254,"linkquality":18,"state":"ON","color_temp":295,"color":{"x":0.4136,"y":0.3941}}'
Zigbee2MQTT:info  2020-08-27 11:53:39: MQTT publish: topic 'zigbee2mqtt/regal_innr_eg_ez', payload '{"brightness":254,"linkquality":18,"state":"ON","color_temp":295,"color":{"x":0.4136,"y":0.3941}}'
Zigbee2MQTT:info  2020-08-27 11:53:39: MQTT publish: topic 'zigbee2mqtt/regal_innr_eg_ez', payload '{"brightness":254,"linkquality":18,"state":"ON","color_temp":295,"color":{"x":0.4136,"y":0.3941}}'
Zigbee2MQTT:info  2020-08-27 11:53:39: MQTT publish: topic 'zigbee2mqtt/regal_innr_eg_ez', payload '{"brightness":254,"linkquality":18,"state":"ON","color_temp":302,"color":{"x":0.416432237808611,"y":0.395982547448827}}'
Zigbee2MQTT:info  2020-08-27 11:53:40: MQTT publish: topic 'zigbee2mqtt/regal_innr_eg_ez', payload '{"brightness":254,"linkquality":21,"state":"ON","color_temp":295,"color":{"x":0.4136,"y":0.3941}}'
Zigbee2MQTT:info  2020-08-27 11:53:40: MQTT publish: topic 'zigbee2mqtt/regal_innr_eg_ez', payload '{"brightness":254,"linkquality":21,"state":"ON","color_temp":302,"color":{"x":0.416432237808611,"y":0.395982547448827}}'
Zigbee2MQTT:info  2020-08-27 11:53:40: MQTT publish: topic 'zigbee2mqtt/regal_innr_eg_ez', payload '{"brightness":254,"linkquality":21,"state":"ON","color_temp":296,"color":{"x":0.4142,"y":0.3944}}'
Zigbee2MQTT:info  2020-08-27 11:53:43: MQTT publish: topic 'zigbee2mqtt/regal_innr_eg_ez', payload '{"brightness":254,"linkquality":18,"state":"ON","color_temp":302,"color":{"x":0.4177,"y":0.396}}'
```

So there are no messages containing information about setting up reporting. The funny thing is, that reporting now works again.

One thing I should mention: In the last few days we had a power outage of the entire street we live in. Could it be possible that this is related? Am I assuming correctly, that all devices would somehow rejoin when the power is restored? Could this be more related to the bulb FW than the coordinator and Z2M?

What would be a recommendable procedure after a power outage when you have a bunch of devices? Turn on the fuses one at a time after restoring Z2M/coordinator?

sjorge, 2020-08-27:
I think the messages might be of log level debug,.... but a power outage could explain it.

If I cut power to all my lights at once and restore it, they don't all come back with working reporting as the coordinator will be too busy and fail, there was a recent change in dev to retry on those failures, which makes things better but if 20+ bulbs rejoin, some are missing missing reporting.

Rejoining them in small batches usually works fine for me though, I'm using S1's for some (need to get more for the other lights) so I can just physically turn them on in sets of 4-5 bulbs and spread then out over a few minutes.

theFork, 2020-08-27:
OK, thanks for the explanations. Since this retry mechanism is under development I think we can close this issue. Is there an ongoing discussion/issue regarding the retry mechanism? I would volunteer to apply a full-house outage if testing is required ;)

sjorge, 2020-08-27:
#4109 is the issue

theFork, 2020-08-27:
Tanks!

closed 2020-08-27 by theFork