Koenkk/zigbee2mqtt #4188: Xiaomi Door Sensor Contact Payload Missing

jani339, 2020-08-26:
# xiaomi  Door Sensor does not send contact state
## I installed Zigbee2MQTT on my Ubuntu Virtual machine. Im able to pair devices. I paired a xiaomi window/door sensor and a xiaomi motion detector.

I checked afterwards the Logs of Zigbee2MQTT with a successful pairing (already supported by Zigbee2MQTT). The motion detector is sending properly all  values:
info  2020-08-26 14:54:49: MQTT publish: topic 'zigbee2mqtt/Bewegungssensor', payload '{"battery":100,"voltage":3035,"linkquality":70,"**occupancy":false**,"device":{"friendlyName":"Bewegungssensor","model":"RTCGQ01LM","ieeeAddr":"0x00158d000303bd7e","networkAddress":53973,"type":"EndDevice","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","applicationVersion":11}}'

The Windows Sensor is missing the contact values:
info  2020-08-26 14:54:42: MQTT publish: topic 'zigbee2mqtt/0x00158d0003746561', payload '{"battery":100,"voltage":3015,"linkquality":26,"device":{"friendlyName":"0x00158d0003746561","model":"MCCGQ01LM","ieeeAddr":"0x00158d0003746561","networkAddress":27318,"type":"EndDevice","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","applicationVersion":10}}'

## What did you expect to happen
Im expecting in the Window Sensor Payload a string with "Contact":On or "Contact":Off

## How to reproduce it (minimal and precise)
It´s simply by pair the device. There is never a message with contact information.

## Debug Info
Zigbee2MQTT version: zigbee2mqtt@1.14.3
Adapter hardware: CC2531
Adapter firmware version: Newest

Koenkk, 2020-08-26:
Can you provide the herdsman debug logging when toggling contact of the sensor?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

jani339, 2020-08-27:
> Can you provide the herdsman debug logging when toggling contact of the sensor?
> 
> To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

I paired it again to receive a log, its the device --> 0x00158d00036ca00f:
[image]

By open/closing actions is no message send/logged :(, so i paired it again to give you some more information.

Koenkk, 2020-08-27:
- You don't see anything at all in the log? 
- Can you try replacing the battery and make keep the device very close to the coordinator? 
- Are you able to verify if the device works as expected with a Xiaomi gateway?

jani339, 2020-09-01:
> * You don't see anything at all in the log?

No there is nothing in the Logs. :(
And the Sensors working as expected with the APP (Sorry its based on German, but it shows the OPEN/CLOSING times of the sensors)
[image]

Koenkk, 2020-09-02:
> Could there be an Issue with the door sensor Configs that the Payload is not parsed correctly?

I don't think so, we should atleast see something in the herdsman debug logging.

I'm afraid the only way to continue here is to sniff the traffic when triggering the contact when its paired to Zigbee2MQTT (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

closed 2020-10-11 by github-actions[bot]