Koenkk/zigbee2mqtt #4189: all devices connect to coordinator, not to routers

sander816, 2020-08-26:
# Bug Report
## What happened
All devices connect to coordinator, none to my 3 other routers 

## What did you expect to happen
That devices connect to the nearest router, because they lose connection aftyer a while

## How to reproduce it (minimal and precise)
?

## Debug Info
Zigbee2MQTT version:
Adapter hardware: CC2531
Adapter firmware version: revision":20190608
Zigbee2MQTT version 1.14.3-dev (commit #5f8b9be)

[image]

Koenkk, 2020-08-26:
It's hard for me to read the linkquality values, but they look good (100+). So probably those devices have a stronger connection via the coordinator then the routers. Note that in Zigbee devices itself are responsible for choosing their parent, this is not something that can be "steered" via Zigbee2MQTT.

insipiens, 2020-08-26:
I had a similar issue, when pairing all devices it helps to pair the routers first to the coordinator (or other routers), then pair the end devices...even then some found a router or coordinator that seemed further away than seemed logical.

Does this happen even when re-pairing a device close to a router (centimeters?)

sander816, 2020-08-27:
Ok, I agree
But my cc2531 is hitting memory limits with errors because I am getting to the maximum devices. That's why I have a router, but no device used it

I wi

sander816, 2020-08-27:
Zigbee: allowing new devices to join.
Aug 27 20:35:02 rpi3.local npm[9386]: Zigbee2MQTT:error 2020-08-27 20:35:02: Failed to call 'BridgeLegacy' 'onMQTTMessage' (Error: SREQ '--> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":63,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,150,2,11,254,0]}}' failed with status '(0x10: MEM_ERROR)' (expected '(0x00: SUCCESS)')
Aug 27 20:35:02 rpi3.local npm[9386]:     at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:291:27)
Aug 27 20:35:02 rpi3.local npm[9386]:     at Generator.next (<anonymous>)
Aug 27 20:35:02 rpi3.local npm[9386]:     at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:24:58))

Koenkk, 2020-08-28:
@sander816 in case you are hitting memory issues, I would recommend switching to the source routing firmware since it has some more memory available. https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

closed 2020-10-05 by github-actions[bot]