Koenkk/zigbee2mqtt #4192: Need to power-cycle CC26X2R1 after reboot to get latest koenkk/zigbee2mqtt container to connect to it

maichai, 2020-08-27:
## What happened
After reboot the herdsman will not start due to getting no connection to the coordinator device. zigbee2mqtt will not work.

## What did you expect to happen
After reboot it should be possible to connect to the coordinator device.

To fix the new problem situation I need to power-cycle the coordinator hardware  after reboot.

Actually it worked like expected since over a year. After initial flashing of the device I did not change the firmware, so the new behavior is not driven by a new firmware version.

To me it seems more pausible that a docker container upgrade led introduced a buggy software on the host side.

## How to reproduce it (minimal and precise)
reboot

## Debug Info
Zigbee2MQTT version: zigbee2mqtt 1.14.3 running in Docker container koenkk/zigbee2mqtt
with --device=/dev/serial/by-id/usb-Texas_Instruments_XDS110__03.00.00.07__Embed_with_CMSIS-DAP_L43002EF-if00:/dev/coordinator
using /devcoordinator in the configuration.yaml

Adapter hardware: CC26X2R1
Adapter firmware version: '{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20191106}}'

Koenkk, 2020-08-27:
I think this has been discussed in #1429, seems its caused by a certain combination of setup/drivers/os. Not sure if there is a fix available.

wixoff, 2020-08-28:
Yes, this has been a problem for me ever since I started using the CC26X2R1 board.  I have tried a few different ways of performing USB resets before starting up the container (I'm using podman rather than docker), but none of them help -- unplug-replug seems to be the only solution.

It's not ideal, but I bought a USB cable with an inline power switch to avoid having to pull the cord so often.

In #1429 some people do report that a Raspberry Pi doesn't have this problem.

maichai, 2020-08-28:
Listen guys, I cannot quite follow you on this theory, because:
I wrote that I was using the **same board** with the **same firmware** and the **same OS** for many months **without any problems**. 
I cannot see how this combination would now suddenly create a problem, since **none** of these things changed when the problem started.
The only thing that **may** have changed is the version of the koenkk/zigbee2mqtt container.

How do you think you current theory fits into this picture? I m honestly curious.

Koenkk, 2020-08-29:
could you downgrade Zigbee2mqtt to figure out in what version it was still working?

maichai, 2020-08-30:
Not sure how to do this right now, this docker UI does not let me select an old version. What is the command line option for requesting a particular version?

Koenkk, 2020-08-30:
Look at https://www.zigbee2mqtt.io/information/docker.html#running but replace `koenkk/zigbee2mqtt` with e.g. `koenkk/zigbee2mqtt:1.14.2`. You can find all version numbers here: https://hub.docker.com/repository/docker/koenkk/zigbee2mqtt/tags?page=1

maichai, 2020-09-02:
OK I am using 1.14.2 now. I may be able to reboot that server later on and will let you know.

maichai, 2020-09-13:
OK I tried all versions released in 2020 and it did not fix the issue. I must have been hallucinating??

closed 2020-10-21 by github-actions[bot]