Koenkk/zigbee2mqtt #4193: Livolo switch as light

MiguelAngelLV, 2020-08-27:
## The bug
The livolo switchs shows a switch, but, in the most of cases, it are used to turn on / off lights, so, we need create a light using switch plataform

## What did you expect to happen

The livolo switch could be a parameter to shows a light o switch, with switch as default for compatibility.

xppx99, 2020-09-11:
Hello, this is quite similar with the Aqara D1 smart switch. I totally agree with the sugestion MiguelAlgelLV does.

closed 2020-10-19 by github-actions[bot]

MiguelAngelLV, 2021-04-20:
Will this feature be implemented? Should I create new issue or this will be reponed?

Koenkk, 2021-04-20:
You can try the opposite of https://www.zigbee2mqtt.io/integration/home_assistant.html#exposing-switch-as-a-light