Koenkk/zigbee2mqtt #4196: Ikea Tradfri LED1738G7 Inerview Failed

RestOp, 2020-08-27:
Hello,

i'm on Hassio Docker.
Using CC2531 with 3.0 Firmware - https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.0.x/bin/CC2531_20190425.zip
CC2531 Connected Directly in Raspberry. Keeping Lamp directly close to antenna of CC2531

Have a lot of other devices that pairs and work perfect.
But these two lamps i can not get paired.
I reseter them many times.
But always get the same:
Zigbee2MQTT:info  2020-08-27 23:24:46: Device '0x680ae2fffe924be9' joined
Zigbee2MQTT:info  2020-08-27 23:24:46: Starting interview of '0x680ae2fffe924be9'
Zigbee2MQTT:info  2020-08-27 23:24:46: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x680ae2fffe924be9"}}'
Zigbee2MQTT:info  2020-08-27 23:24:46: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x680ae2fffe924be9"}}'
....
Zigbee2MQTT:error 2020-08-27 23:27:05: Failed to interview '0x680ae2fffe924be9', device has not successfully been paired
Zigbee2MQTT:info  2020-08-27 23:27:05: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x680ae2fffe924be9"}}'

By the way - lapms blinks right after reset and then after 1-2 minutes.

Koenkk, 2020-08-28:
Did you check: https://www.zigbee2mqtt.io/information/FAQ.html#interview-fails ?

RestOp, 2020-08-28:
> Did you check: https://www.zigbee2mqtt.io/information/FAQ.html#interview-fails ?

Of course!
Do i need to post zigbee-herdsman debug log?

Koenkk, 2020-08-28:
Did you also try with touchlink? https://www.zigbee2mqtt.io/information/touchlink.html#factory-reset-device

RestOp, 2020-08-28:
> Did you also try with touchlink? https://www.zigbee2mqtt.io/information/touchlink.html#factory-reset-device

No, will try. Thanks!
Tell you result.

RestOp, 2020-08-28:
> Did you also try with touchlink? https://www.zigbee2mqtt.io/information/touchlink.html#factory-reset-device

No result.

`Zigbee2MQTT:info  2020-08-28 23:47:56: Starting touchlink factory reset...
Zigbee2MQTT:info  2020-08-28 23:47:56: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"touchlink","message":"reset_started","meta":{"status":"started"}}'
Zigbee2MQTT:warn  2020-08-28 23:48:21: Failed to factory reset device through Touchlink
Zigbee2MQTT:info  2020-08-28 23:48:21: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"touchlink","message":"reset_failed","meta":{"status":"failed"}}'`

Koenkk, 2020-08-29:
@RestOp look that the device does not stay in the network, do you have any other zigbee networks running? (tradfri/hue/xiaomi bridge?)

RestOp, 2020-08-29:
> @RestOp look that the device does not stay in the network, do you have any other zigbee networks running? (tradfri/hue/xiaomi bridge?)

No, nothing at all.
I have only cc2531 stick and no other gateways. Maybe any neighbour has one, but i do not think it can cause problems or?

Koenkk, 2020-08-29:
Could be that this is a bug in the zigbee 3.0 firmware of the CC2531 (it's not a recommended firmware for this device). Could you try switching to 1.2 firmware? (https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing). Unfortunately this will require repairing of your devices, you also have to delete `data/coordinator_backup.json` after flashing.

RestOp, 2020-08-29:
> Could be that this is a bug in the zigbee 3.0 firmware of the CC2531 (it's not a recommended firmware for this device). Could you try switching to 1.2 firmware? [...]

I have one more stick with 1.2 firmware, but i do not use it because i have some 3.0 devices.
I can try with it.

But the question - which firmware do i have to use, to make it work 3.0 on cc2531?

Koenkk, 2020-08-29:
I don't recommend 3.0 on cc2531, the cc2531 is not powerful enough for it.

Note that 3.0 devices will work perfectly fine with a 1.2 coordinator.

RestOp, 2020-08-29:
> Note that 3.0 devices will work perfectly fine with a 1.2 coordinator.

Really? I did not knew it!

If i get 3.0 stick out and put on its place 1.2 - will i need to repair all the devices and delete data/coordinator_backup.json ?
Or can i use 2 sticks togeather? I do not want to repair all the devices, because some of them are hard reachable ;-(((

Koenkk, 2020-08-29:
Ah, I forgot that 3.0 -> 1.2 is a special case and if I remember correctly does not require repairing. Just flash and start :)

RestOp, 2020-08-29:
> Ah, I forgot that 3.0 -> 1.2 is a special case and if I remember correctly does not require repairing. Just flash and start :)

I have no cable more to flash (((
But as i mentioned i have one more stick with 1.2. The only question how to use it for not to repair device, if there a way of course...

Koenkk, 2020-08-29:
Can you try to just plug and start with that stick?

RestOp, 2020-08-29:
Made it. Paired in few seconds without any problems!  Thanks a lot!
3.0 device seems aso to work.
But Buld was recongnized as LED1903C5/LED1835C6, instead of LED1836C6 (E14).
And here is not contorls for dimming and color 
[image]
I hope it is not problem!

Thanks one more time!!!

closed 2020-08-29 by RestOp