Koenkk/zigbee2mqtt #4197: Fix SNZB-02

akonstster, 2020-08-28:
# Bug Report
SONOFF Temperature and humidity sensor (SNZB-02) doesnt display any data  after pairing.
Im use mqtt discovery home assistant create entitys, but states **Unknown**. 

logs:
https://hastebin.com/cuwihuzake.py

## Debug Info
Zigbee2MQTT version:  1.14.3
Adapter hardware: CC2531
Adapter firmware version: 20190608

nanobra1n, 2020-08-28:
Did you wait long enough? Mine reports only every hour especially if the ambient temperature is pretty stable.

tomdudu38, 2020-08-28:
hi, ùy sensor snzb-04 do not work anymore seems to be since 1.14.2
my logs
https://github.com/danielwelch/hassio-zigbee2mqtt/issues/372

tomdudu38, 2020-08-29:
hello, mine report nothings.

akonstster, 2020-08-29:
Wait 1 week, and nothing happens.

nanobra1n, 2020-08-30:
How far away from the coordinator is your sensor? The CC2531's range isn't great especially through walls. If it isn't already, try placing it close to the coordinator. 

For what it's worth, I removed and repaired mine today and it reports at least every hour (1.14.3 with a CC1352P-2).

tomdudu38, 2020-08-31:
hi,i did same,removed and repaired and it seems to work .
it saw an hour, or less.,( few minutes)ex today:
Zigbee2MQTT:info  2020-08-31 17:31:55: MQTT publish: topic 'zigbee2mqtt/0x00124b001b783bed', payload '{"temperature":23.07,"linkquality":92,"humidity":62,"battery":80,"voltage":2900}'
Zigbee2MQTT:info  2020-08-31 17:32:35: MQTT publish: topic 'zigbee2mqtt/0x00124b001b783bed', payload '{"temperature":23.07,"linkquality":92,"humidity":60.5,"battery":80,"voltage":2900}'
Zigbee2MQTT:info  2020-08-31 17:33:15: MQTT publish: topic 'zigbee2mqtt/0x00124b001b783bed', payload '{"temperature":23.07,"linkquality":94,"humidity":61.8,"battery":80,"voltage":2900}'
Zigbee2MQTT:info  2020-08-31 17:33:32: MQTT publish: topic 'zigbee2mqtt/0x00124b001b781d08', payload '{"temperature":23.91,"linkquality":52,"humidity":52.61,"battery":80,"voltage":2900}'
Zigbee2MQTT:info  2020-08-31 17:34:11: MQTT publish: topic 'zigbee2mqtt/0x00124b001b795657', payload '{"state":"OFF","linkquality":52}'
Zigbee2MQTT:info  2020-08-31 17:38:05: MQTT publish: topic 'zigbee2mqtt/0x00124b001b783bed', payload '{"temperature":23.07,"linkquality":94,"humidity":60.6,"battery":80,"voltage":2900}'
Zigbee2MQTT:info  2020-08-31 17:38:39: MQTT publish: topic 'zigbee2mqtt/0x00124b001b781d08', payload '{"temperature":23.91,"linkquality":52,"humidity":51.59,"battery":80,"voltage":2900}'
Zigbee2MQTT:info  2020-08-31 17:39:11: MQTT publish: topic 'zigbee2mqtt/0x00124b001b795657', payload '{"state":"OFF","linkquality":52}'
Zigbee2MQTT:info  2020-08-31 17:40:48: MQTT publish: topic 'zigbee2mqtt/0x00124b001b783bed', payload '{"temperature":23,"linkquality":92,"humidity":60.6,"battery":80,"voltage":2900}'
Zigbee2MQTT:info  2020-08-31 17:40:59: MQTT publish: topic 'zigbee2mqtt/0x00124b001b781d08', payload '{"temperature":23.91,"linkquality":52,"humidity":50.52,"battery":80,"voltage":2900}'
Zigbee2MQTT:info  2020-08-31 17:42:21: MQTT publish: topic 'zigbee2mqtt/0x00124b001b783bed', payload '{"temperature":23,"linkquality":94,"humidity":60.6,"battery":80,"voltage":2900}'

closed 2020-10-09 by github-actions[bot]