Koenkk/zigbee2mqtt #4198: IKEA Tradfri Dimmer (E1743) not sending "key presses"

Havoc85, 2020-08-28:
# Bug Report
## What happened
I successfully paired a IKEA Tradfri Dimmer, but I don't see any events of the MQTT Device in Homeassistant.

Also unbinding of the device works:
https://hastebin.com/ramedesilo.coffeescript

As you can see here: Homeassistant added the device correctly and even get the battery and linkquality information:
[image]

## What did you expect to happen
See an event (action or click event) when I press the dimmer button.

## How to reproduce it (minimal and precise)
- pair a dimmer
- unbind the dimmer
- press a button

## Debug Info
Zigbee2MQTT version: 1.14.3
I don't know which exact adapter I'm using and on which firmware it's running - is there any advice how I get that on hass?

Koenkk, 2020-08-28:
It could be that your firmware is too old (these devices require a recent firmware), you can check the version via: https://www.zigbee2mqtt.io/getting_started/flashing_the_cc2531.html#how-to-check-the-installed-firmware-version

Havoc85, 2020-08-29:
It looks, that I have Version 20181024 (Coordinator firmware version: '{"type":"zStack12","meta":"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20181024}}'). Is there any way to check, which firmware I need exactly?

Koenkk, 2020-08-30:
At least 20190217 is needed (https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/CHANGELOG.md#20190217). But I would recommend updating to the latest, this will fix your issue.

closed 2020-08-30 by Koenkk

Havoc85, 2020-09-01:
Yesterday I updated to the latest Firmware. It fixed my issue. Thanks you.

geidorei, 2022-01-16:
Fantastic post! After much time trying to find the issue. Off to AliExpress to order the bits needed. Thanks folks!