Koenkk/zigbee2mqtt #4200: Gledopto GL-FL-006TZS is not in the support list.

audiobah, 2020-08-28:
Hi !

I created a pull request( #1502 at zigbee-herdsman-converters [Link](https://github.com/Koenkk/zigbee-herdsman-converters/pull/1502) ) for support of the GL-FL-006TZS to supported devices.
It was committed by Koenkk, but the device is not in the supported list on the zigbee2mqtt.io and in hassio as well (Maybe this device is not (yet) supported by Zigbee2Mqtt?).
What I am doing wrong?

nanobra1n, 2020-08-28:
Your PR has been committed to the dev branch 3 days ago. It won't be available until the next stable release (probably in a couple of weeks, just judging by the recent release cycles). Documentation gets updated then as well. 

If you don't want to wait that long you need to install the zigbee2mqtt-edge add-on and use that one instead. Note that the -edge add on doesn't get automatically updated, you need to uninstall and reinstall to get the latest version.   https://github.com/danielwelch/hassio-zigbee2mqtt#updating-the-add-on-and-zigbee2mqtt-library

audiobah, 2020-08-28:
Thank you! now it's clear.

Koenkk, 2020-08-28:
@nanobra1n thanks for helping.

The next release will be on next wednesday.

closed 2020-08-28 by Koenkk