Koenkk/zigbee2mqtt #4202: TS0201 does not update frequently enough.

SargonofAssyria, 2020-08-28:
The TS0201 does only update a few times per day while temperature and humidity change a lot more.
The device presents it self as: Model is TS0201 by _TZ2000_a476raq2.
The steps of temp changes are high. Sometimes more then 2 degrees, but mostly 0.7 - 0.9 degrees between updates.
The setting I use in the configuration.yaml are:
    temperature_precision: 2
    temperature_calibration: -2
I have used other 'precisions' but the result is the same.
Is there a 'hidden' setting about the update frequency?
I have a Cresta, TFA TS34C nearby and that device change dozens of times per day in increments of 0.1 degree.

## What did you expect to happen
The TS0201 should have (almost) same kind if updates.
## Debug Info
Zigbee2MQTT version: 1.14.3-dev (commit #a78d0f6)
Adapter hardware: CC1352P-2
Adapter firmware version: 20200805

Koenkk, 2020-08-28:
You can check if it can be configured more often by changing the entry in `devices.js` to:

```js
{
    zigbeeModel: ['TS0201'],
    model: 'TS0201',
    vendor: 'TuYa',
    description: 'Temperature & humidity sensor with display',
    supports: 'temperature and humidity',
    fromZigbee: [fz.battery, fz.temperature, fz.humidity],
    toZigbee: [],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        const binds = ['msTemperatureMeasurement'];
        await bind(endpoint, coordinatorEndpoint, binds);
        await configureReporting.temperature(endpoint, {min: 1, max: 5, change 1});
    },
},
```

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html how to edit `devices.js`.

sjorge, 2020-08-28:
I too have a bunch of them, they do indeed go hours without updates at times. A heiman next to it to compared logged 2-3 deg change over the time.

Not sure they support reporting, I got back an error. I will also try again next time I am at my computer if @SargonofAssyria doesn’t beet me too it

sjorge, 2020-08-28:
Just a heads up, there is an error in your new device entry, there needs to be a : behind change.

SargonofAssyria, 2020-08-29:
I got messages that the device was configured and succesfully interviewed, but some what later I got.
`error 2020-08-29 09:46:30: Failed to configure 'Woonkamer', attempt 3 (Error: ConfigureReporting 0xec1bbdfffea1474e/1 msTemperatureMeasurement([{"attribute":"measuredValue","minimumReportInterval":1,"maximumReportInterval":5,"reportableChange":1}], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 29275 - 1 - 13 - 1026 - 7 after 10000ms)
    at Timeout._onTimeout (/home/pi/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7))
`

SargonofAssyria, 2020-08-29:
Half a day later. The Cresta had several increments of 0.1 degrees. The TS0201 still is at the same temperature.
'Last seen' date and time still from the start of zigbee2mqtt.
So the solution did not work, also seen by the error in the logfile.

Koenkk, 2020-08-29:
@SargonofAssyria so it seems that we cannot configure the device to report more often unfortunately

SargonofAssyria, 2020-08-29:
To bad. Tnx for the effort.

closed 2020-08-29 by SargonofAssyria

Bartekn86, 2020-08-29:
ive asked tuya to update firmware to fix problem, i have same issue

sjorge, 2020-08-30:
Yep, same issue with mine I just retried, same error I as when i tried it a few weeks ago too. Device was awake because I kept it online by pushing the little button, to rule out it taking a nap.

SargonofAssyria, 2020-08-30:
After I left the device at rest, it seems to have a default step size of 0.7 degrees.
Is that something that could be investigated further, changing the step size?

jandy123, 2020-10-16:
In a desperate attempt to improve the reporting frequency (or alternatively decrease the reporting threshold), I've noticed that the device reports data every 1 minute if the button is continuously kept pressed. I, however, have no clue what the implication on battery life are. If anyone could do some measurements, I'd be very curious...

SargonofAssyria, 2020-10-16:
I tried that too, and it works some times. But I also get messages about leaving the network, so I have to include it again.
I looks like that every time you push the button Temp and hum is sent.
I tried to dissemble it without breaking it, but did not succeed yet.

jandy123, 2020-10-16:
@SargonofAssyria : I've managed to dissemble it without breaking and block the button in the pushed position. So far so good, for me it reports data every minute and the signal symbol on the LCD is blinking like every second. While this works here, I doubt battery life wilkl be great, but, as I said, I cannot measure such small current with the tools I have....

To disassemble, just start in the battery compartment,  inserting some screwdriver forcing a bit, and at the same time popping the plastic clips all around. There are like 5 clips, I'd recommend starting below where there are a few holes (it's only one clip there).

SargonofAssyria, 2020-10-16:
Tnx, for the instructions how to open the TS0201. I block the switch and see the updates coming every minute.
Now wait how the battery holds.

cpuks, 2020-10-21:
Any reports on battery life after modification as otherwise it's useless due to no updates.

jandy123, 2020-10-21:
As I said, that's no way for me to measure actual current, so just looking at the battery voltage it's still at the starting value.

So, looks fine here !

SargonofAssyria, 2020-10-22:
After the modification ( 1 week ago ) battery is still at 100% and voltage dropped from 2.9 Volt to 2.8 Volt.
They where not fresh batteries.
So there is some impact, but have to wait until the devices dies on battery power.

jandy123, 2020-11-07:
Well, battery life is just about two weeks with good batteries. In my opinion this is simply not acceptable...

sjorge, 2020-11-07:
I switched all of mine to the new sonoff ones, they u take annoying cell batteries but atleast update often enough for me to get nice data. And they were super cheap in bulk.

SargonofAssyria, 2020-11-07:
Mine are after more then 3 weeks still at 100% and voltage 2.700
Which "new sonoff ones"?

sjorge, 2020-11-07:
https://www.itead.cc/sonoff-snzb-02-zigbee-temperature-and-humidity-sensor.html

Data I am getting from them is pretty consistent, even have one outside and it’s holding up fine!

~ sjorge

SargonofAssyria, 2020-11-08:
Tnx, I have a Xiaomi look a like. But this one does not have a display.

Bartekn86, 2020-11-08:
is there any firmware update ?

SargonofAssyria, 2020-11-08:
Device does not support OTA.

Bartekn86, 2020-11-08:
how they flashed firmware first time ? :-)

sjorge, 2020-11-08:
> how they flashed firmware first time ? :-)

Factory bakes them in, sometimes making fuses burn in the result to make it truly read only... 
Theoretically you could probbaly use a romflasher to dump and then flash the chip again. But you need:
- figure out the exact chip and pin layout of the rom
- a compatible rom flasher
- a device with a new firmware to dump
- dump the firmare
- flash the firmware on the rom chip of a device with the old firmware
- oops nope, you need to back that one up first in case the new one works
- hope it still works after.

Realistically speaking if the device does not support Zigbee OTA you're not going to get fimrware updates for it, and if it does... it's still a difficult to get the firmware files.

SargonofAssyria, 2020-11-14:
Mine are after more then 4 weeks still at 100% and voltage 2.600
I think the 100% is wrong, but device still working, although voltage is going down rapidly.

jandy123, 2020-11-15:
Well, that's good to know. Mine stopped working when voltage dropped to
around 2.0V. This does happen fast, though.

Anyways, 4 weeks is quite decent. Which batteries, btw. ?

falmanna, 2020-11-22:
I have 20 of these sensors, the battery lasts for 2 months at max. This is definitely not acceptable since it is being advertised as a one-year battery lifetime, and in reality, it is not even 1/4 of that.

sjorge, 2020-11-22:
Yeah the super sensitive settings dropped mine by a whole lot in a week, so think they will last about a month to two like this.

The followup of .5 change and 15 minutes max seems to do better...

But maybe we need to play with minimum too.

change: .5
max: 1h
min: 5 minutes

For example, although giving the user control to set this to their preference is ultimately better I guess, as that might not be frequent enough for some... as that is a maximum of 12 updates per hour... which to me personally is plenty.

~ sjorge

jandy123, 2020-11-22:
@sjorge: Did you manage to change the settings ? I'm unable to change those on mine. 

Which z2m version you use ? Also, could you please test for instance if you can set a change of 0.1 ? In my case I'm stuck with 0.7 degree for the reportable step size, see post above by  @SargonofAssyria.

sjorge, 2020-11-22:
Doh :( I was confusing this thread with the one for the SONOFF one :( where I also posted in recently,... no I was unable to change the default bindings on these :( I also tried a full remove/join cycle but they seem to stubornly ignore the requests.

Even without then on the network, they seem to drain battery rather fast (so just display) I was using ikea AAA in mine. So not sure if it is due to poor battery quality or not.

jandy123, 2020-11-22:
@sjorge Thought so ;). No problem. 

===

Well, I guess I'll wire mine to some 3.3V power supply and keep the button pressed ;).

SargonofAssyria, 2020-12-31:
Today battery level in Domoticz came to 1%
In Z2M battery is still at 100, Voltage is lowered to 2.100 Volt.
I will change batteries with 2 lithium Energizer batteries and will see how long they hold.

Newan, 2020-12-31:
I have too devices that hold the battery up to 6 month. The another two devices hold only for weeks. I see no different between this devices, but only 2 devices works

srtxema, 2021-01-13:
Same problem. Only update 3 times/day with variations until 5 degrees.

srtxema, 2021-01-14:
[image]
show temperature of 2 devices
pink line is tuya ts0201. 6 times / 24 hours
red line is sonoff snzb-02. 3 times /hour

jandy123, 2021-01-14:
Yep, so it's complete trash. Only decent way is to hook it up to some power adapter and keep the button pressed...

srtxema, 2021-01-14:
the problem with sonoff is that it has no display. any alternative similar to TS0201 but that works well?

bprus, 2021-01-19:
I have the same problem with TS0201, using deconz. Sometimes it doesn't update for as long as 18 hours. Has anyone found any solution with display? Previously I used Xiaomi BLE sensor with display, but it often wasn't updated due to connectivity issues (however, it updated every few minutes, so still better than TS0201).

cpuks, 2021-01-19:
As mentioned above couple times - hardware hack is only solution + I suggest using lithium battery.

cdemi, 2021-07-24:
Just to let everyone know that I recently added Tuya SM0201 which is a very similar device and it suffers the same problem.

https://github.com/Koenkk/zigbee-herdsman-converters/pull/2833

greko, 2021-08-03:
I tried to set the precision values for the TS0201, but I don't see the difference in reporting between different settings. I do it in configuration.yaml in that way:
devices:
  '0x5c0272fffec3da7d':
    friendly_name: '0x5c0272fffec3da7d'
    debounce: 1
    humidity_precision: 1
    temperature_precision: 1

Did anyone succeed in this kind of configuration? 
My issue is that the env conditions are changing quite fast which causes the sensor reporting every 1 or 2 minute, which drains battery fast. I'd like to change it to report less frequently.

Another question: what is the unit for humidity_precision and temperature_precision? Are these predefined constants: 0,1,2 like described in documentation, or they are the % or degrees of celsious respectively?

falmanna, 2021-08-03:
@greko  I am really familiar with zigbee2mqtt as I used this sensor in openahb zigbee before, but in general, there should be a config in Zigbee to configure min and max reporting interval time regardless of value. If it is possible here then that might help you by setting it to e.g. 10 mins or more.

greko, 2021-08-03:
@falmanna Thanks for response. I tried the way as @Koenkk suggested in this thread, but I am receving similar output as @SargonofAssyria ;-/ (timeout)

falmanna, 2021-08-04:
@greko yeah, I am not sure if it is working for me or the sensor is just reporting slowly. Overall it is really a bad sensor, but it looks good!

jandy123, 2021-10-17:
Since I like the looks of these devices, I decided to add a low-power Attyni13A to "push" the button every 10 minutes and force a new update and reading.

Waiting for the Attynis to arrive. Will let know how it goes, in case people still interested.

eloo, 2021-11-02:
@jandy123 
thanks for this workaround. 

i've applied it to all 4 of my ts0201 and so far its working really good. 
i hope the battery is not going to be drained too fast but we will see. 

i have also slightly improved your setup. so i can get the VCC from a pad like the others
[image]

my code 
```
#include <avr/interrupt.h>
#include <avr/sleep.h>

// Pin 4 for LED
volatile int led = 1; //pin 4 (ATtiny leg 3)
volatile int count = 0;

ISR(WDT_vect) {
  count = count +1;
  //93 with 1.2 internal oscillator, about 15 min
  if(count > 90)
  {
    digitalWrite(led, LOW);   // push button; when pushed, button is pulled down by 10k resistor in the device, so this should be safe 
    delay(150);               // wait for 100ms
    digitalWrite(led, HIGH);    // release button
    count = 0;
  }
  //sleep_mode();u
}

void setup() {
  // initialize the digital pin as an output.
  pinMode(led, OUTPUT);
  digitalWrite(led, HIGH); //when released/default, button is pulled up by 100k resistor in the device
  
  pinMode(2, INPUT_PULLUP);
  pinMode(3, INPUT_PULLUP);
  pinMode(0, INPUT_PULLUP);
  
  // prescale timer to 8s so we can measure current
  WDTCR |= (1<<WDP3 )|(0<<WDP2 )|(0<<WDP1)|(1<<WDP0); // 8s
  // Enable watchdog timer interrupts
  WDTCR |= (1<<WDTIE);
  // Disable ADC
  ADCSRA &= ~(1<<ADEN);

  ACSR = (1<<ACD); //Disable the analog comparator
  DIDR0 = 0x3F; //Disable digital input buffers on all ADC0-ADC5 pins.  
  
  sei(); // Enable global interrupts
  
  // Use the Power Down sleep mode
  set_sleep_mode(SLEEP_MODE_PWR_DOWN);
  
  for (;;) {
    sleep_mode();   // go to sleep and wait for interrupt...
  }
  
}
// the loop routine runs over and over again forever:
void loop() {
}
```
and i've programmed the ATTiny13 with USBasp with the following settings in Arduino IDE

BOD: disabled
Clock: 4,8 Mhz

and here is a comparison:

a fixed one
[image]
and a not fixed one
[image]

I hope this useful for others as well to make their TS0201 work properly.

Thanks

jandy123, 2021-11-03:
@eloo 

> i hope the battery is not going to be drained too fast but we will see.
I hope so, please report here for comparison. I've just changed batteries (alkaline) in a modded device.

> i have also slightly improved your setup. so i can get the VCC from a pad like the others
Sure, but I'd be careful when physically pushing the button on the device, if the Attyni is powered/working.

I did enable BOD (1.8V), since the device stops working anyhow at 1.9V.
Also I used 1.2 MHz internal oscillator.

Let's see what happens!

jandy123, 2021-11-03:
Btw., I'm also considering powering the devices using eneloops. Has anyone tried this? Any figures regarding battery life? 

Thanks!

falmanna, 2021-11-03:
@jandy123 I got 30 of these sensors and the battery life was poor. They used to last for 3 months on energizer batteries.

jandy123, 2021-11-03:
@falmanna Hmm, that's not much. Curious about rechargeables even more...

dim6003, 2021-12-07:
Hello,
Having the same sensor, and of course same issue. Maybe some resistor or capacitor on the board manages the frequency of sending? but was unable to find any schematics... Also some components are missing from the board.

3615JMD, 2021-12-23:
Hello, @jandy123:  thank you for providing this solution that works fine. ATTiny adds 4µA of current drain. not so much. that is 35 mAH per year. 
I have improved the code so that the ATTiny output does not interfere with the push button. this is very simple: Program digital pin level as LOW (always) and then program the pin as INPUT. therefore the pin is Hi-Z except during the ISR where you program the PIN as OUTPUT for a while just to pull down the button signal, then INPUT = Hi-Z again for some minutes.

jandy123, 2021-12-23:
@3615JMD Great, thanks for sharing!

===

Btw., battery life with eneloops is by no means great. (I suspect the zigbee chip + lcd screen are the biggest consumers; not much to do here.)

I get about 3-4 weeks of a pair of new eneloops 750ma.  Well, at least I can recharge them quite a number of times ;).

eloo, 2021-12-23:
@jandy123 do you plan to provide an updated version of your code?

and maybe we should make a gist out of it

jandy123, 2021-12-23:
@eloo Please do so; sorry no time for it ;)  Would be great to include the latest contribution by @3615JMD.

Cheers and happy holidays!

eloo, 2021-12-27:
@jandy123 @3615JMD 
i've created a gist for the TS0201 with ATTiny13

I've also added the mod from @3615JMD, maybe you can give feedback if this is what you have explained above.

Feel free to give feedback

3615JMD, 2021-12-28:
@eloo, 
Yes, that was exactly my Idea. It works fine at home on 3 x TS0201 for the moment.
Sorry for not sharing my code, but I adapted & compiled it in Atmel Studio and it is quite different regarding I/O control. Showing my code would probably add confusion to arduino users.
Cheers to everybody here !
Jean Marc.

petitchti, 2022-01-02:
Hi everyone,

Just got some TS0201 and have the same problem. However the hack with Attyni13 seems promising.

But what about the battery life with this hack? I see 3-4weeks with 750mA batteries in the previous posts. Is it with the hack? Does anyone have a kind of comparison of battery life with and without this hack?

Thanks :)

[87 more replies]