Koenkk/zigbee2mqtt #4213: Reporting setups fails with Data request failed with error: 'undefined' (26)

sjorge, 2020-08-30:
# Bug Report
## What happened

Occasionally configuration of reporting fails with
```
Zigbee2MQTT:error 2020-08-30 12:37:47: Failed to setup reporting for '0xec1bbdfffe189fde' - Error: ConfigureReporting 0xec1bbdfffe189fde/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":300,"reportableChange":0}], {"timeout":10000,"disableRe
sponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'undefined' (26))
    at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:309:27)
    at Generator.next (<anonymous>)
    at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:24:58)
```

## What did you expect to happen
Reporting to work or fail with a clear error

## How to reproduce it (minimal and precise)
Power off a router device that requires reporting to be setup every time, e.g. trådfri stuff. Happens every so often with no real pattern, might be close to BUFF_FULL or MEM_ERROR as they usually show up for some of the subsequent bulbs or panels.

## Debug Info
```
[root@amethyst /opt/zigbee2mqtt]# mosquitto_sub -t zigbee2mqtt/bridge/info | jq .
{
  "commit": "6452500",
  "coordinator": {
    "meta": {
      "maintrel": 3,
      "majorrel": 2,
      "minorrel": 6,
      "product": 0,
      "revision": 20200728,
      "transportrev": 2
    },
    "type": "zStack12"
  },
  "log_level": "info",
  "network": {
    "channel": 11,
    "extended_pan_id": "0x31d4151f5fefd6c3",
    "pan_id": 6753
  },
  "permit_join": false,
  "version": "1.14.3-dev"
}
```

Koenkk, 2020-08-30:
Pushed a possible fix.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

sjorge, 2020-08-30:
Oh MAC_NO_RESOURCE would make sense yeah as I hit MEM_ERRORs sometimes shortly after.

(I really need to get the zzh driver working haha)

sjorge, 2020-08-31:
I've physically toggled about 14 bulbs without errors for now, closing this.
2 of them complete setup after about 3x as long as normal, so I assume those hit either this one or the buff_full error.

closed 2020-08-31 by sjorge